raspberrypi/picamera2 #1105: [BUG] Lens position must be set to 12 to get a non-blurry image 17.5 cm away from the RPicamera3 wide

essair, 2024-09-03:
I am using the Raspberry Pi Camera3 Wide to take pictures of a surface 17.5 cm away from the camera. The cameras are not tilted, a few of them produce sharp images (not blurry) with lens position 10, and others with lens position 12! 

I discovered this by outputting the metadata from test images, taken with autofocus, to a text file. Using the autofocus settings it is possible for the value of "Lens Position" to be 12, annoyingly this value doesn't work when I try to set the focus manually! Here the range is limited 0 - 10.

Is there something I can do to get the manual focus to accept values over 10, in order for me to get sharp images with the cameras that are behaving differently?

**The metadata from two images  - written to a .txt file using "picam.capture_metadata()" **
home/pi/pycam/lens_pos/images_/test_3_timesleep_2_2024-09-03T144209.jpeg
{'SensorTimestamp': 48306310000, 'ScalerCrop': (0, 0, 4608, 2592), 'FocusFoM': 9090, 'ExposureTime': 29553, 'ColourTemperature': 3978, 'AfState': 2, 'SensorTemperature': 32.0, 'AnalogueGain': 1.1228070259094238, 'AfPauseState': 0, 'FrameDuration': 30208, 'LensPosition': 12.0, 'DigitalGain': 1.0126278400421143, 'ColourCorrectionMatrix': (1.5586624145507812, -0.2736068367958069, -0.28505560755729675, -0.4208212196826935, 1.7169564962387085, -0.29613202810287476, -0.03456248715519905, -0.6009914875030518, 1.6355539560317993), 'AeLocked': True, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'Lux': 445.55718994140625, 'ColourGains': (1.763379693031311, 2.326901435852051)}

/home/pi/pycam/lens_pos/images_/test_3_timesleep_2_2024-09-03T144223.jpeg
{'SensorTimestamp': 62473699000, 'ScalerCrop': (0, 0, 4608, 2592), 'FocusFoM': 8611, 'ExposureTime': 29633, 'ColourTemperature': 3983, 'AfState': 2, 'SensorTemperature': 33.0, 'AnalogueGain': 1.1327433586120605, 'AfPauseState': 0, 'FrameDuration': 30288, 'LensPosition': 12.0, 'DigitalGain': 1.0128061771392822, 'ColourCorrectionMatrix': (1.5583807229995728, -0.2732362747192383, -0.2851444184780121, -0.4206940531730652, 1.7177613973617554, -0.2970641553401947, -0.034741196781396866, -0.6006830334663391, 1.635424256324768), 'AeLocked': True, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'Lux': 410.597900390625, 'ColourGains': (1.7648323774337769, 2.3244616985321045)}

My Hardware: RaspberryPi 3b+ and RaspberryPi Camera Module 3 wide

I'd be really grateful if someone can help me!

davidplowman, 2024-09-09:
I think that editing the camera tuning file may help. You should be able to find this here: `/usr/share/libcamera/ipa/rpi/vc4/imx708_wide.json`. @njhollinghurst should be able to give you advice on what to change.

njhollinghurst, 2024-09-09:
How are you setting the focus manually? I'm not sure why it's being limited to 10 (the default range should be 0-12, and a larger range 0-32 can be enabled using AfRangeFull).

essair, 2024-09-09:
Thank you for your replies!

I have misunderstood the picamera2 manual  "5.2.3. Setting the Lens Position Manually", the minimum and maximum values mentioned there are 0 and 10 - Hence I understood that these were **the** minimum and maximum values! It would be really useful if the range of values for lens position could be mentioned in the documentation, for other people like me! 

OR is there another place where Picamera2 is well documented that I should be looking at instead?

Thank you again for your help and advice :-)

njhollinghurst, 2024-09-09:
Yes I can see the documentation is a little confusing -- it seems to be talking about a notional / generic camera rather than this specific one(?)

I also share your surprise that some cameras require such a large focus setting, I suppose the variance for the wide camera is greater than the standard one because it has a smaller focal length, so small offsets make a relatively large difference. Are they accurate at 0.0 == infinity?