raspberrypi/picamera2 #1110: [BUG] Difference between preview and raw image

bfokin, 2024-09-10:
**Please only report one bug per issue!**

**Describe the bug**
I ran the app [`app_full.py`](apps/app_full.py) and was taking images in raw format. 
I noticed the preview was not replicated with the raw image output.

**To Reproduce**
- Running the above application. Set the contrast to 1.70.
- Disable AWB and AEC
- set exposure time to 29 (the lowest that the global shutter camera can do)
- set the file type to raw

**Expected behaviour**
The preview and captured raw image would look identical.

**Console Output, Screenshots**
[image]

**Hardware :**
- RPi 5
- R Pi Global shutter camera

Thank you everyone for your help!

davidplowman, 2024-09-11:
Hi, can you say how you have converted the raw image (a DNG file?) into something that you can view? Note that you can't expect a different raw converter to convert images in the same way as the hardware in the Pi. The Pi is tuned to produce fairly high contrast, high colour saturation images (though you can change this if you wish).

bfokin, 2024-09-11:
I was viewing raw images with rawtherapee.

Doesn't it show images in their raw form?

davidplowman, 2024-09-11:
I'm not quite sure what we're referring to by "raw" image here. I'm referring to the actual Bayer mosaic from the image sensor, and this is what's stored in a DNG file. I wasn't completely clear whether you're talking about DNG or PNG files, as they're very different things. There's lots of processing that has to be done to a DNG file to make it look like a regular image, and in generally, everyone will do this differently.

bfokin, 2024-09-11:
@davidplowman 

the image shown above, the right image is the DNG file produced by the full_app.py demo. the DNG doesn't look like the image shown within the preview. 

Is there a way for the preview of this demo to be the Bayer mosaic from the image sensor?

davidplowman, 2024-09-11:
We provide access to the raw (Bayer) image from the camera, and allow it to be saved to a DNG file, but we don't have any particular way of displaying it. The Picamera2 preview windows will only show fully processed (RGB or YUV, not Bayer) images.

I don't think I'm really understanding what you want to do. Do you want to view Bayer mosaic images? This would be fairly unusual, but obviously there can be reasons for doing this. Or are you wanting the signal processor on the Pi to perform the same raw conversion as Rawtherapee? Or maybe for Rawtherapee to perform its software conversion more like the Pi hardware?