raspberrypi/picamera2 #1113: [HOW-TO] Switch Mode and Capture - "NoneType" object has no attribute "Configuration"

schillingderek, 2024-09-12:
Similar to some other topics, I'm attempting to capture a full resolution image during both video capture and video streaming. As far as I can tell, the error I'm seeing represents a unique outcome from other questions I've seen asked (though I could just be misinterpreting the situation).

My current setup has two encoders running simultaneously, one used for streaming (lores) and one used for video capture (main). At intervals, I'd like to be able to capture a full resolution image as well for some ML inference processing.

Here is a simplified version of what I'm trying:
```
#!/usr/bin/env python3

import io
import picamera2
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput, CircularOutput
import time

from threading import Thread, Condition

width_main = 1920
height_main = 1080

width_lores = 720
height_lores = 405

class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

    def read(self):
        with self.condition:
            self.condition.wait()
            return self.frame

class Camera:
    def __init__(self):
        self.picamera = picamera2.Picamera2()
        self.video_config = self.picamera.create_video_configuration(
            {"size": (width_main, height_main), "format": "RGB888"}, lores={"size": (width_lores, height_lores)}
        )
        self.picamera.configure(self.video_config)
        self.video_capture_endoder = H264Encoder(10000000)
        self.video_capture_output = CircularOutput(buffersize = 1)
        self.streaming_encoder = H264Encoder()
        self.streaming_encoder.bitrate = 2500000
        self.streaming_encoder.profile = "baseline"
        self.stream_out = StreamingOutput()
        self.stream_out_2 = FileOutput(self.stream_out)
        self.streaming_output = [self.stream_out_2]

        self.picamera.start()

    def capture_image(self):
        still_config = self.picamera.create_still_configuration()
        job = self.picamera.switch_mode_and_capture_file(still_config, "test.jpg", wait=False)
        metadata = self.picamera.wait(job)

camera = Camera()

def stream():
    camera.picamera.start_encoder(
        camera.streaming_encoder, camera.streaming_output, name="lores"
    )
    camera.picamera.start_encoder(
        camera.video_capture_endoder, camera.video_capture_output, name="main"
    )

    time.sleep(3)
    camera.capture_image()

if __name__ == "__main__":
    stream()
```

When running this, the camera successfully switches to the still configuration and captures the full res image, but then when switching back to video capture(?) I get an error. Here is the output I see:

```
[1:59:33.849618134] [1294]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+50-69a894c4
[1:59:33.980721026] [1297]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:59:33.984925190] [1297]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media0
[1:59:33.985042789] [1297]  INFO RPI pipeline_base.cpp:1125 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:59:34.000570950] [1294]  INFO Camera camera.cpp:1191 configuring streams: (0) 1920x1080-RGB888 (1) 720x404-YUV420 (2) 2304x1296-SBGGR10_CSI2P
[1:59:34.001290090] [1297]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
[1:59:37.330651125] [1301]  INFO Camera camera.cpp:1191 configuring streams: (0) 4608x2592-BGR888 (1) 4608x2592-SBGGR10_CSI2P
[1:59:37.346724578] [1297]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 4608x2592-SBGGR10_1X10 - Selected unicam format: 4608x2592-pBAA
[1:59:38.253001069] [1301]  INFO Camera camera.cpp:1191 configuring streams: (0) 1920x1080-RGB888 (1) 720x404-YUV420 (2) 2304x1296-SBGGR10_CSI2P
[1:59:38.261289296] [1297]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
Exception during process_requests()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1293, in process_requests
    encoder.encode(encoder.name, req)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 213, in encode
    self._encode(stream, request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 257, in _encode
    cfg = stream.configuration
          ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'configuration'
Exception in thread Thread-2 (thread_func):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 29, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 85, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1293, in process_requests
    encoder.encode(encoder.name, req)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/encoder.py", line 213, in encode
    self._encode(stream, request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 257, in _encode
    cfg = stream.configuration
          ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'configuration'
```

If I then interrupt the process:
```
Traceback (most recent call last):
  File "/home/birdcam/birdCam/stream_test.py", line 78, in <module>
    stream()
  File "/home/birdcam/birdCam/stream_test.py", line 72, in stream
    camera.capture_image()
  File "/home/birdcam/birdCam/stream_test.py", line 58, in capture_image
    metadata = self.picamera.wait(job)
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1323, in wait
    return job.get_result(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 79, in get_result
    return self._future.result(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 451, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.11/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt
```

Any hints on what I'm missing here? Something to do with needing a specific `signal_function` being passed into `switch_mode_and_capture_file`? I just can't seem to figure it out!

**Additional context**
Raspberry Pi 3b+
Camera Module 3 [Arducam version](https://www.amazon.com/Arducam-Raspberry-Camera-Autofocus-15-22pin/dp/B0C9PYCV9S?ref_=ast_sto_dp)
Python 3.11.2

davidplowman, 2024-09-13:
Hi, thanks for the question.

Are you trying to do a `switch_mode_and_capture` while you have encoders running? I'm afraid this isn't supported. Trying to change the camera mode and image resolutions while encoders are running is not possible. (I'm a bit surprised it didn't complain explicitly about that, so maybe there's something to look into there.)

There are two obvious choices:

* Run the camera permanently at the full resolution. You can then ask for a full resolution "main" stream which you can use to capture stills, and a lower resolution "lores" stream which you can encode as video. But the sensor, in its full resolution, won't run at the full framerate, so you'd have to see if it's sufficient.
* Or run the camera in a mode that has a framerate suitable for video. Then that will determine the maximum resolution you can get for the still image capture. Again, you'll have to decide what you prefer.

Type `rpicam-hello --list-cameras` to check the resolutions and framerates of the camera modes that are available.

schillingderek, 2024-09-18:
@davidplowman  Thanks for the response! For now, I'm resolved to set the `main` stream at 1920x1080 (which seems to be the max that the RPI3b+ can hardware encode H264) and then use that stream to capture both still and video. I find that I can run that main stream at a higher resolution if I were to use the MJPEG encoder, but the file sizes that end up being generated for video is just WAY too big for my purposes.

closed 2024-09-18 by schillingderek