raspberrypi/picamera2 #1119: AttributeError: 'dict' object has no attribute 'AeExposureModeEnum'

Gordon999, 2024-09-18:
**Describe the bug**
Using https://github.com/raspberrypi/picamera2/blob/main/examples/hailo/detect.py and trying to send controls to the camera.
It fails as shown below

**To Reproduce**
picam2.set_controls({"AeEnable": True,"AeExposureMode": controls.AeExposureModeEnum.Normal})
AttributeError: 'dict' object has no attribute 'AeExposureModeEnum'

**Expected behaviour**
I expect the camera to switch Mode

**Console Output, Screenshots**
picam2.set_controls({"AeEnable": True,"AeExposureMode": controls.AeExposureModeEnum.Normal})
AttributeError: 'dict' object has no attribute 'AeExposureModeEnum'

**hardware:
Pi5, hailo, pi v3 camera

davidplowman, 2024-09-23:
I wonder if you've reused `controls` as a variable, pointing to a dictionary object? (Yes, I've done that!)

So maybe
```
import libcamera
...
picam2.set_controls({"AeEnable": True,"AeExposureMode": libcamera.controls.AeExposureModeEnum.Normal})
```

Gordon999, 2024-09-23:
Excellent, thankyou.

The picamera2 hailo example I based my code on uses..

controls = {'FrameRate': 30}
config = picam2.create_preview_configuration(main, lores=lores, controls=controls)

so I have renamed 'controls' and my code now works

Gordon999, 2024-09-23:
Solved

closed 2024-09-23 by Gordon999