raspberrypi/picamera2 #1121: [BUG] IMX500 examples : 'tuple' object has no attribute 'x'

camilaodsouza, 2024-09-19:
**Please only report one bug per issue!**

**Describe the bug**
Running examples **imx500_classification_demo.py** and **imx500_object_detection_demo.py** now return an error for some models.
 
 **File "/home/pi/git/picamera2/examples/imx500/imx500_classification_demo.py", line 62, in draw_classification_results
    cv2.putText(m.array, "ROI", (b.x + 5, b.y + 15), cv2.FONT_HERSHEY_SIMPLEX,
                                 ^^^
AttributeError: 'tuple' object has no attribute 'x'
Aborted**

This happens because imx500.get_roi_scaled now returns a tuple, so the code must be adjusted to reflect that change. 

**To Reproduce**
imx500_classification_demo.py --model /usr/share/imx500-models/imx500_network_efficientnetv2_b0.rpk --preserve-aspect-ratio

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
AI Camera, RPi5

**Additional context**
Add any other context about the problem here.

davidplowman, 2024-09-23:
Thanks for the report. We'll look into it!

naushir, 2024-09-23:
Should be fixed now.