raspberrypi/picamera2 #1122: Hailo Face Example

matt-desmarais, 2024-09-20:
I am playing around with the hailo examples, I created a [face example](https://github.com/matt-desmarais/picamera2/blob/main/examples/hailo/face.py) following the format of the pose and detect examples. I noticed on the [hailo-rpi5-examples](https://github.com/hailo-ai/hailo-rpi5-examples) that they resized and padded their examples so I decided to resize and pad the picamera2 examples so 1280x720 frames are resized/padded to 640x640 and the overlays can be displayed on a 1920x1080 frames. I also wanted to record everything as a video, I started using picamera2.encoders but that used a lot of the cpu so I switched to cv2.VideoWriter_fourcc which performs better. I have a version of this for pose, detect and face that I have been working on over the last couple days, I am still fine tuning the recording part. I [forked the repo](https://github.com/matt-desmarais/picamera2/tree/main/examples/hailo) and added my programs if they would be of any interest.

https://github.com/user-attachments/assets/e1540dfe-964d-48ea-89f1-ac68c08bee6e

davidplowman, 2024-09-23:
Hi, thanks for the information. Are you talking about the code in this example: https://github.com/matt-desmarais/picamera2/blob/main/examples/hailo/face_scaled_record.py ?

I'm actually a bit interested as to why `cv2.VideoWriter` would perform better. I was expecting it to use libav/FFmpeg on Linux (which is that `Picamera2.encoders` uses), but maybe I'm mistaken? So there's something to understand there.

Some other changes I might consider:
* Try `'YUV420'` for the main stream, instead of `'XRGB8888'`, it's much less bandwidth and should work fine (with the Picamera2 encoder, not so sure about cv2). Also saves libav/FFmpeg converting RGB to YUV.
* `create_video_configuration` is usually recommend over `create_preview_configuration` for video recording. It will default to 30fps and longer buffer queues, which usually works better.
* Can you choose 640x640 as the size for the `lores` stream? Resizing in software is usually very painful.

matt-desmarais, 2024-09-23:
cv2.VideoWriter actaully didnt perform better it was just recording at a slightly higher framerate which made it appear better. I appreciate and I will play around with the suggested changes. The reason I wanted to change lores and resize is the examples annotation doesn't line up as well without it.
This is the original pose example with recording
https://github.com/user-attachments/assets/ff3cd481-5b96-41e7-be6d-389b345bff46

this is scaled and padded version
https://github.com/user-attachments/assets/93b1c98d-46ec-4a20-8311-2564c9e39159