raspberrypi/picamera2 #1124: [BUG] is it a bug. Truncate (right shift) ExposureTime with long exposures. ArduCam 64Mp HawkEye

kwis2, 2024-09-27:
Hi
I need long ExposureTime for about 10 seconds. With libcamera-still there is no problem. I have written code in Python and I am not able to set longer time than about 3.2s. when I test libcamera it work OK.

my code is:
```
capture_config = picam2.create_still_configuration(
        main={"format": "YUV420", "size": (9152, 6944)},
        buffer_count=1,
        controls={
            "AeEnable": False,  
            "ExposureTime": 3200000,
            "AnalogueGain": 2.9, 
            "AfMode": False,
            "AwbEnable": False,
            "FrameDurationLimits": (500000,500000000)                     
            }
        )
picam2.configure(capture_config)
request = picam2.capture_request(flush=True)
metadata=request.get_metadata()
request.release()
```

I debug picam2.camera_config and metadata and when I set ExposureTime below 3200000, ExposureTime in metadata is ok, but for higher values there are problems.
for instance
set 'ExposureTime': 3200000, got 'ExposureTime': 3199951
set 'ExposureTime': 5000000, got 'ExposureTime': 2499908,
set 'ExposureTime': 8000000, got 'ExposureTime': 1999989,
set 'ExposureTime': 10000000, got 'ExposureTime': 2499908,

in my oppinion it looks like a bug which right shift ExposureTime to be lower than about 3200000. Maybe there is another explanation and I am doing something wrong?

picamera version 0.3.21, Raspberry pi 4, Bookworm Lite 64, 

How could I get longer times?.
Best regards
kwis2

log 3.2s
```
`{'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 1, 'queue': True, 'main': {'format': 'YUV420', 'size': (9152, 6944), 'stride': 9152, 'framesize': 95327232}, 'lores': None, 'raw': {'format': 'SRGGB10_CSI2P', 'size': (9152, 6944), 'stride': 11456, 'framesize': 79550464}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (500000, 500000000), 'AeEnable': False, 'ExposureTime': 3200000, 'AnalogueGain': 2.9, 'AfMode': False, 'AwbEnable': False}, 'sensor': {'bit_depth': 10, 'output_size': (9152, 6944)}, 'display': None, 'encode': None}`
`Metadata: {'AfState': 0, 'DigitalGain': 1.0025542974472046, 'AeLocked': False, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ColourCorrectionMatrix': (1.4354209899902344, -0.25621968507766724, -0.1792113482952118, -0.35534733533859253, 1.469967246055603, -0.11460991948843002, -0.12574085593223572, -0.6062259674072266, 1.7319668531417847), 'AnalogueGain': 2.892655372619629, 'FocusFoM': 7617, 'ExposureTime': 3199951, 'ColourTemperature': 4000, 'LensPosition': 1.0, 'Lux': 0.1130077913403511, 'ColourGains': (1.5943877696990967, 2.129925489425659), 'SensorTimestamp': 42287215197000, 'ScalerCrop': (0, 0, 9345, 6944), 'AfPauseState': 0, 'ScalerCrops': [(0, 0, 9345, 6944)], 'FrameDuration': 3202445}`

```
log 5s
```
{'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 1, 'queue': True, 'main': {'format': 'YUV420', 'size': (9152, 6944), 'stride': 9152, 'framesize': 95327232}, 'lores': None, 'raw': {'format': 'SRGGB10_CSI2P', 'size': (9152, 6944), 'stride': 11456, 'framesize': 79550464}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (500000, 500000000), 'AeEnable': False, 'ExposureTime': 5000000, 'AnalogueGain': 2.9, 'AfMode': False, 'AwbEnable': False}, 'sensor': {'bit_depth': 10, 'output_size': (9152, 6944)}, 'display': None, 'encode': None}
Metadata: {'AfState': 0, 'DigitalGain': 2.0051515102386475, 'AeLocked': False, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ColourCorrectionMatrix': (1.4354209899902344, -0.25621968507766724, -0.1792113482952118, -0.35534733533859253, 1.469967246055603, -0.11460991948843002, -0.12574085593223572, -0.6062259674072266, 1.7319668531417847), 'AnalogueGain': 2.892655372619629, 'FocusFoM': 8103, 'ExposureTime': 2499908, 'ColourTemperature': 4000, 'LensPosition': 1.0, 'Lux': 0.14569689333438873, 'ColourGains': (1.5943877696990967, 2.129925489425659), 'SensorTimestamp': 42123282845000, 'ScalerCrop': (0, 0, 9345, 6944), 'AfPauseState': 0, 'ScalerCrops': [(0, 0, 9345, 6944)], 'FrameDuration': 2501207}
```

log 8s
```
{'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 1, 'queue': True, 'main': {'format': 'YUV420', 'size': (9152, 6944), 'stride': 9152, 'framesize': 95327232}, 'lores': None, 'raw': {'format': 'SRGGB10_CSI2P', 'size': (9152, 6944), 'stride': 11456, 'framesize': 79550464}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (500000, 500000000), 'AeEnable': False, 'ExposureTime': 8000000, 'AnalogueGain': 2.9, 'AfMode': False, 'AwbEnable': False}, 'sensor': {'bit_depth': 10, 'output_size': (9152, 6944)}, 'display': None, 'encode': None}
Metadata: {'AfState': 0, 'DigitalGain': 4.0, 'AeLocked': False, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ColourCorrectionMatrix': (1.4354209899902344, -0.25621968507766724, -0.1792113482952118, -0.35534733533859253, 1.469967246055603, -0.11460991948843002, -0.12574085593223572, -0.6062259674072266, 1.7319668531417847), 'AnalogueGain': 2.892655372619629, 'FocusFoM': 10020, 'ExposureTime': 1999989, 'ColourTemperature': 4000, 'LensPosition': 1.0, 'Lux': 0.44593915343284607, 'ColourGains': (1.5943877696990967, 2.129925489425659), 'SensorTimestamp': 80175426825000, 'ScalerCrop': (0, 0, 9345, 6944), 'AfPauseState': 0, 'ScalerCrops': [(0, 0, 9345, 6944)], 'FrameDuration': 2000612}
```

log 10s
```
{'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 1, 'queue': True, 'main': {'format': 'YUV420', 'size': (9152, 6944), 'stride': 9152, 'framesize': 95327232}, 'lores': None, 'raw': {'format': 'SRGGB10_CSI2P', 'size': (9152, 6944), 'stride': 11456, 'framesize': 79550464}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (500000, 500000000), 'AeEnable': False, 'ExposureTime': 10000000, 'AnalogueGain': 2.9, 'AfMode': False, 'AwbEnable': False}, 'sensor': {'bit_depth': 10, 'output_size': (9152, 6944)}, 'display': None, 'encode': None}
Metadata: {'AfState': 0, 'DigitalGain': 4.0, 'AeLocked': False, 'SensorBlackLevels': (4096, 4096, 4096, 4096), 'ColourCorrectionMatrix': (1.4354209899902344, -0.25621968507766724, -0.1792113482952118, -0.35534733533859253, 1.469967246055603, -0.11460991948843002, -0.12574085593223572, -0.6062259674072266, 1.7319668531417847), 'AnalogueGain': 2.892655372619629, 'FocusFoM': 10319, 'ExposureTime': 2499908, 'ColourTemperature': 4000, 'LensPosition': 1.0, 'Lux': 0.45719826221466064, 'ColourGains': (1.5943877696990967, 2.129925489425659), 'SensorTimestamp': 79951085896000, 'ScalerCrop': (0, 0, 9345, 6944), 'AfPauseState': 0, 'ScalerCrops': [(0, 0, 9345, 6944)], 'FrameDuration': 2500584}
```

kwis2, 2024-09-28:
Update
Coleages have tested on another raspberry with libacmera-still and it looks it is problem of libcamera not picamera2.
command
`libcamera-still -o "test.jpg" --width 9152 --height 6944 --shutter 100000 --contrast 2.0 --gain 1.8 --metadata "test.json" --immediate --autofocus-mode manual --awbgains 1,1 --flush
`
results:
Shutter  [s]   ->     Exposure Time     [s]
1 ->                                       1    
2 ->                                        2  
3     ->                                    3  
3.5      ->                              1.75   
4            ->                            2    
5                ->                        2.5  
6                    ->                    3     
7                        ->                1.75      
8                            ->            2         

Are you able to test, maybe there is any configuration of system / picamera2 / libcamera version which work.

naushir, 2024-09-29:
Please speak to Arducam about this - they implemented the kernel driver and libcamera code to drive this sensor.  The Arducam Hawkeye device support has never been upstreamed into our libcamera tree.

closed 2024-09-29 by naushir