raspberrypi/picamera2 #1131: [BUG] lib logging info to standard error with logging level set to warn

tim-oe, 2024-10-13:
**Please only report one bug per issue!**

**Describe the bug**
A clear and concise description of what the bug is.
in setting the following
`Picamera2.set_logging(Picamera2.WARNING)`

getting info logs showing up in std error
[8:37:12.548662111] [25832]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[8:37:12.575426649] [25834]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[8:37:12.576952364] [25834]  WARN RPI vc4.cpp:393 Mismatch between Unicam and CamHelper for embedded data usage!
[8:37:12.577484892] [25834]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[8:37:12.577520873] [25834]  INFO RPI pipeline_base.cpp:1126 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[8:37:44.142096436] [25846]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[8:37:44.142986995] [25834]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
[8:37:46.271116344] [25847]  INFO Camera camera.cpp:1197 configuring streams: (0) 3280x2464-BGR888 (1) 3280x2464-SBGGR10_CSI2P
[8:37:46.274267348] [25834]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
[8:37:46.938085421] [25847]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-SBGGR10_CSI2P
[8:37:46.941555049] [25834]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 640x480-SBGGR10_1X10 - Selected unicam format: 640x480-pBAA
t

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.
following class is camera impl
https://github.com/tim-oe/WeatherWatch/blob/main/weatherwatch/camera/Camera.py
I'm running the app as a systemd service as sending std err to log file
https://github.com/tim-oe/WeatherWatch/blob/main/systemd/weatherwatch.service

**Expected behaviour**
A clear and concise description of what you expected to happen.
shouldn't be logging to STD ERR and should be using applcation log config
also shouldn't write logs below set level

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.
rpi 4 with pi camera noir v2

**Additional context**
Add any other context about the problem here.

tim-oe, 2024-10-13:
i looked deeper into the function
`def set_logging(level=logging.WARN, output=sys.stderr, msg=None)`

i changed my code to
    Picamera2.set_logging(level=logging.WARN, output=None)

but logs are still showing in std.err

and running 
python3-picamera2/stable,stable,now 0.3.22-2 all [installed]

davidplowman, 2024-10-14:
Hi, thanks for the question. The problem may be that the logging that you've shown above is _libcamera_ logging, not _Picamera2_ logging. To disable libcamera logging, you can set the environment variable `LIBCAMERA_LOG_LEVELS` to, for example, `*:3`. The `*` means "all modules" and the number is sets the amount of logging, with `0` producing very large amounts. So you could start Python from the command line like this
```
LIBCAMERA_LOG_LEVELS=*:3 python
```

tim-oe, 2024-10-14:
thanks!
this worked for me
LIBCAMERA_LOG_LEVELS=*:WARN

closed 2024-10-14 by tim-oe