raspberrypi/picamera2 #1139: [BUG] KeyError: 'ColourCorrectionMatrix' when capturing dng files

pbaetens, 2024-10-24:
version: picamera2 0.3.20
libcamera compiled from git, tag: v0.3.1+rpt20240829
OS: bookworm kernel 6.6.y on pi5

```
(venv) pi@raspberry5:~/ams_rpi_software/picamera2/examples $ python capture_dng.py 
[11:40:52.104576972] [3047]  INFO Camera camera_manager.cpp:316 libcamera v0.3.1+49-48fe316f-dirty (2024-10-23T21:41:59CEST)
[11:40:52.115072755] [3051]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 03-09-2024 (14:16:52)
[11:40:52.117891652] [3051]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[11:40:52.118595862] [3051]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/mira016@36 to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[11:40:52.118649788] [3051]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 03-09-2024 (14:16:52)
[11:40:52.120822716] [3051]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[11:40:52.121629075] [3051]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/mira050@36 to CFE device /dev/media4 and ISP device /dev/media1 using PiSP variant BCM2712_C0
QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
[11:40:53.488770636] [3047]  INFO Camera camera.cpp:1191 configuring streams: (0) 640x480-XBGR8888 (1) 400x400-GRBG_PISP_COMP1
[11:40:53.488914507] [3051]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/mira016@36 - Selected sensor format: 400x400-SGRBG12_1X12 - Selected CFE format: 400x400-PC1G
[11:40:53.492865766] [3072]  WARN IPARPI pisp.cpp:502 Could not set NOISE_REDUCTION_MODE - no Denoise algorithm
[11:40:56.212661363] [3055]  INFO Camera camera.cpp:1191 configuring streams: (0) 400x400-BGR888 (1) 400x400-GRBG_PISP_COMP1
[11:40:56.215983431] [3051]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/mira016@36 - Selected sensor format: 400x400-SGRBG12_1X12 - Selected CFE format: 400x400-PC1G
[11:40:56.217232331] [3088]  WARN IPARPI pisp.cpp:502 Could not set NOISE_REDUCTION_MODE - no Denoise algorithm
Traceback (most recent call last):
  File "/home/pi/ams_rpi_software/picamera2/examples/capture_dng.py", line 19, in <module>
    picam2.switch_mode_and_capture_file(capture_config, "full.dng", name="raw")
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/picamera2.py", line 1458, in switch_mode_and_capture_file
    return self.dispatch_functions(functions, wait, signal_function, immediate=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/picamera2.py", line 1354, in dispatch_functions
    return job.get_result(timeout=timeout) if wait else job
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/job.py", line 79, in get_result
    return self._future.result(timeout=timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
    return self.__get_result()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in __get_result
    raise self._exception
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/job.py", line 48, in execute
    done, result = self._functions[0]()
                   ^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/picamera2.py", line 1448, in capture_and_switch_back_
    done, result = self.capture_file_(file_output, name, format=format, exif_data=exif_data)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/picamera2.py", line 1392, in capture_file_
    request.save_dng(file_output)
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/request.py", line 178, in save_dng
    return self.picam2.helpers.save_dng(self.make_buffer(name), self.get_metadata(), self.config[name], file_output)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/picamera2/request.py", line 320, in save_dng
    camera = Picamera2Camera(config, metadata)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/pidng/camdefs.py", line 42, in __init__
    self.__settings__()
  File "/home/pi/ams_rpi_software/venv/lib/python3.11/site-packages/pidng/camdefs.py", line 69, in __settings__
    ccm = self.metadata["ColourCorrectionMatrix"]
          ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'ColourCorrectionMatrix'

```

davidplowman, 2024-10-24:
Hi, I've just tried this on an up to date image and am not seeing it.

However, looking at the log you posted, I see you're using a "mira" sensor (?) that I don't recognise. I'm guess the fault may be that the tuning file for this sensor may not be defining a proper colour matrix, which is then missing? Can you say anything more about this sensor, and where the support for it has come from? Thanks.

pbaetens, 2024-10-24:
@davidplowman 
thanks. 
the whole driver / software tuning was done by me.
indeed, the CCM was missing in the .json tuning file!
however - this mira sensor is monochrome - that's why there was no color matrix

closed 2024-10-24 by pbaetens