raspberrypi/picamera2 #1141: [HOW-TO] pip install picamera2 from source

pbaetens, 2024-10-25:
### situation:
raspberry pi5 bookworm, kernel 6.6
libcamera compiled from latest main.
picamera2 does not work when installing using `apt install python3-picamera2` because of binary incompatiblity.
I need to install the latest picamera2 from source.

So, I try this:
```
python3 -m venv venv --system-site-packages
source venv/bin/activate

git clone picamera2 
cd picamera2
cd picamera2 && pip install .
```

then i get complaints about not finding packages such as python3-jsonschema.
even when installing those using apt-get it fails.

### **solution**:

rename `python3-jsonschema` to `jsonschema` in `setup.py`
I believe all packages listed in setup.py must be coming from pypi and NOT apt..

davidplowman, 2024-10-25:
Yes, thanks for pointing this out. This is already fixed in our "next" branch, so should be available more generally quite soon.

pbaetens, 2024-10-31:
thanks! indeed - using the 'next' branch - all is fine.

closed 2024-10-31 by pbaetens