raspberrypi/picamera2 #1142: [BUG] `enable_sps_framerate` parameter only works for hardware encoder

mikkelam, 2024-10-25:
**Please only report one bug per issue!**

**Describe the bug**
Cannot include  `enable_sps_framerate`. As this interface is kinda hidden in how it switches, it's not easy for a user to understand this difference.

**To Reproduce**
```python
FPS = 30
enc = H264Encoder(enable_sps_framerate=True, framerate=FPS)
```

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
```
  encoder = H264Encoder(enable_sps_framerate=True, framerate=FPS)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: LibavH264Encoder.__init__() got an unexpected keyword argument 'enable_sps_framerate'
```
**Hardware :**
rpi4 -> rpi5 is what made me see the problem

**Additional context**
Add any other context about the problem here.

davidplowman, 2024-10-28:
Hi, I can put this on the list of little jobs to do. PRs also welcome!! :)

mikkelam, 2024-10-28:
> Hi, I can put this on the list of little jobs to do. PRs also welcome!! :)

I could try. How would you like me to handle it? I cannot see a similar configuration on the pyav side. Should we give a warning or silently accept the flag?

davidplowman, 2024-10-28:
Well, I think there is a "framerate" parameter in the codec context that one could set, and I think that does get reflected in the SPS. Though there might be some confusion about whether libav is talking about frames or fields, I'm not quite sure. But it might even be the right thing to do!!