raspberrypi/picamera2 #1145: [HOW-TO] install picamera2 on docker on ubuntu 24.04 on rapsberry pi 5

MahmoudMoumni, 2024-10-28:
**Any idea how to install picamera2 on docker on ubuntu 24.04 installed on raspberry pi 5**

**I have a raspberry pi 5 , I installed ubuntu 24.04 on it, then I created a docker image  with python 3.9.2 using for computer vision project (tensorflow, mediapipe,pyqt5...) everything is working fine, now I would like to read frames from picamera instead of usb camera ,but i cannot install picamera2 or even libcamera-dev , any idea how to do that please**

**I tried to install libcamera-dev but I get this : E: Unable to locate package libcamera-dev , you find attached the dockerfile used**

[image]

davidplowman, 2024-10-28:
Hi, I'm afraid I don't know anything about libcamera on Ubuntu. It might be worth asking on a Ubuntu forum?

If there is no libcamera package for Raspberry Pi on Ubuntu, you might want to consider building it for yourself? For example, although it was some time ago, there was a post about it [here](https://github.com/raspberrypi/picamera2/issues/563).

Sorry not to have better answers for you!

MahmoudMoumni, 2024-10-28:
Dear Mr David ,

Thank you for your reply.

I already did that in the docker container and managed to build the
libcamera library ( ../libcamera/build/src/libcamera/libcamera.so) .
[image: Screenshot from 2024-10-28 17-38-38.png]

the problem is that when I import picamera2 I got this error :
[image: Screenshot from 2024-10-28 17-29-57.png]

Do you think I need to add the libcamera.so path to LD_LIBRARY_PATH ?

Thank you in advance.

Sincerely,
Mahmoud

davidplowman, 2024-10-28:
I'm afraid your images aren't showing, so I can't tell exactly what errors you have. (Maybe just post text if they're text errors?)

I assume you installed libcamera as well, so that the .so files are on your system somewhere? Maybe check it's there and in the right place. Sometimes I find I need a `sudo ldconfig` when I rebuild and re-install stuff. Otherwise maybe try setting LD_LIBRARY_PATH - usually that shouldn't be necessary, but if something is a bit confused and it works, then it might be worth it.