raspberrypi/picamera2 #1147: [BUG] app_full limits the video vertical resolution to 1080

bemmens, 2024-10-29:
**Describe the bug**
app_full.py limits the video vertical resolution to 1080

**Expected behaviour**
Is it possible to enable the full resolution of the sensor?

[image]

**Hardware :**
Pi 5
HQ camera

Thanks!

davidplowman, 2024-10-29:
Hi, yes, I suppose it's a bit inconsistent that the height is limited to 1080 but the width isn't limited to 1920. I suppose I could see an argument for doing both or neither!

But really, this code here is simply an example for you to look at, understand, and copy for your own purposes. We're not providing it as an application that does something terribly useful exactly as it is. So please feel free just to grab the file and do whatever you want with it. The line in question that you'd want to change would be [this one](https://github.com/raspberrypi/picamera2/blob/main/apps/app_full.py#L879).

davidplowman, 2024-10-29:
Glad you're managing to figure it all out! Obviously you have the command line apps for (rpicam-still, rpicam-vid) for simple capture operations. There's also a nice little web UI to the camera that someone wrote [here](https://github.com/monkeymademe/picamera2-WebUI), that might make for a useful interface as well.