raspberrypi/picamera2 #1152: [BUG] "pip3 install picamera2" installs WITH PyQt5 bindings

neuron-whisperer, 2024-11-01:
**Please only report one bug per issue!**

**Describe the bug**

Installing picamera2 installs PyQt5, contrary to the instructions.

**To Reproduce**

python -m pip install picamera2 or some variant thereof.

**Expected behaviour**

The install instructions read as follows:

> This is no longer the recommended way to install Picamera2. However, if you want to do so you can use ...
> 
> `pip3 install picamera2[gui]`
> 
> ...which will install Picamera2 with all the GUI (Qt and OpenGL) dependencies. If you do not want these, please use ...
> 
> `pip3 install picamera2`

I am trying to install this package on a Raspberry Pi 5 running 64-bit Bookworm Lite without any windowed GUI. I installed Python 3.8.19 via pyenv (since I intend to run pygame), upgraded the OS and pip, and then found that picamera2 is not available. I followed the apt install instructions, which succeeded, but picamera2 was still not available in Python. So I tried following the above instructions to install picamera2 without GUI stuff, and it did this:

```
$ python -m pip install picamera2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picamera2
  Using cached https://www.piwheels.org/simple/picamera2/picamera2-0.1.1-py3-none-any.whl (42 kB)
Requirement already satisfied: numpy in ./.pyenv/versions/3.8.19/lib/python3.8/site-packages (from picamera2) (1.24.4)
Collecting PiDNG (from picamera2)
  Using cached pidng-4.0.9.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting piexif (from picamera2)
  Using cached https://www.piwheels.org/simple/piexif/piexif-1.1.3-py2.py3-none-any.whl (20 kB)
Collecting pillow (from picamera2)
  Using cached pillow-10.4.0-cp38-cp38-manylinux_2_28_aarch64.whl.metadata (9.2 kB)
Collecting pyopengl (from picamera2)
  Using cached https://www.piwheels.org/simple/pyopengl/PyOpenGL-3.1.7-py3-none-any.whl (2.4 MB)
Collecting PyQt5 (from picamera2)
  Using cached PyQt5-5.15.11.tar.gz (3.2 MB)
```

...which is clearly installing PyQt5 contrary to the install instructions.

**Console Output, Screenshots**

See above.

**Hardware :**

Raspberry Pi 5 8GB with Raspberry Pi Camera Module 3 connected.

**Additional context**

n/a

mgineer85, 2024-11-03:
Hey, I think best would be to use the venv as you were trying earlier but allow the venv access to the system packages:
https://docs.python.org/3/library/venv.html#creating-virtual-environments
Check out the `--system-site-packages` flag.

If you plan using numpy in your venv you might need to pin the last v1 numpy version. If you need numpy >2, also install simplejpeg in your venv as the system packages are quite outdated and saving images with picamera2 fails otherwise.  

Maybe this could be addressed by updating the system packages in the OS in the future?

neuron-whisperer, 2024-11-14:
I don't use venv in these circumstances because it's a Raspberry Pi running a single application in a kiosk context. My install script installs pyenv and a specific version of Python (3.8.19), and the device runs one script in that one Python environment... The End. The device does nothing else, not even run a windowed GUI (hence Raspberry Pi OS Lite), so a virtual environment for the script isn't required.

I tried again, fist installing simplejpeg and then again running `python -m pip install picamera2` to install picamera2 without PyQt. Same result: picamera2-0.1.1-py3-none-any.whl requires PyQT5, which is directly contrary to the install instructions.

There is a clear disconnect between _what the picamera2 install instructions state about the requirements of the picamera2 package_ and _the actual requirements of the picamera2 package at piwheels_. Please address this, either by fixing the dependencies in the install package or by correcting the install instructions.

neuron-whisperer, 2024-12-05:
Is this error going to be fixed or will the inconsistency between the documentation and dependencies be permitted to exist indefinitely?

neuron-whisperer, 2024-12-05:
Dumping output once again and bolding the error...

```
Glacialisaurus rpicam-apps $ sudo apt install -y python3-prctl libatlas-base-dev ffmpeg libopenjp2-7 python3-pip
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-prctl is already the newest version (1.8.1-1+b2).
python3-prctl set to manually installed.
ffmpeg is already the newest version (8:5.1.6-0+deb12u1+rpt1).
libopenjp2-7 is already the newest version (2.5.0-2).
libopenjp2-7 set to manually installed.
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
The following additional packages will be installed:
  libatlas3-base
Suggested packages:
  libatlas-doc liblapack-doc
The following NEW packages will be installed:
  libatlas-base-dev libatlas3-base
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,192 kB of archives.
After this operation, 30.4 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 libatlas3-base arm64 3.10.3-13 [2,537 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libatlas-base-dev arm64 3.10.3-13 [2,655 kB]
Fetched 5,192 kB in 0s (13.4 MB/s)       
Selecting previously unselected package libatlas3-base:arm64.
(Reading database ... 105665 files and directories currently installed.)
Preparing to unpack .../libatlas3-base_3.10.3-13_arm64.deb ...
Unpacking libatlas3-base:arm64 (3.10.3-13) ...
Selecting previously unselected package libatlas-base-dev:arm64.
Preparing to unpack .../libatlas-base-dev_3.10.3-13_arm64.deb ...
Unpacking libatlas-base-dev:arm64 (3.10.3-13) ...
Setting up libatlas3-base:arm64 (3.10.3-13) ...
update-alternatives: using /usr/lib/aarch64-linux-gnu/atlas/libblas.so.3 to provide /usr/lib/aarch64-linux-gnu/libblas.so.3 (libblas.so.3-aarch64-linux-gnu) in auto mode
update-alternatives: using /usr/lib/aarch64-linux-gnu/atlas/liblapack.so.3 to provide /usr/lib/aarch64-linux-gnu/liblapack.so.3 (liblapack.so.3-aarch64-linux-gnu) in auto mode
Setting up libatlas-base-dev:arm64 (3.10.3-13) ...
update-alternatives: using /usr/lib/aarch64-linux-gnu/atlas/libblas.so to provide /usr/lib/aarch64-linux-gnu/libblas.so (libblas.so-aarch64-linux-gnu) in auto mode
update-alternatives: using /usr/lib/aarch64-linux-gnu/atlas/liblapack.so to provide /usr/lib/aarch64-linux-gnu/liblapack.so (liblapack.so-aarch64-linux-gnu) in auto mode
Processing triggers for libc-bin (2.36-9+rpt2+deb12u9) ...
Glacialisaurus rpicam-apps $ python -m pip install --upgrade numpy
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting numpy
  Using cached numpy-1.24.4-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl.metadata (5.6 kB)
Downloading numpy-1.24.4-cp38-cp38-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (14.0 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.0/14.0 MB 13.6 MB/s eta 0:00:00
Installing collected packages: numpy
Successfully installed numpy-1.24.4
Glacialisaurus rpicam-apps $ **python -m pip install picamera2**
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting picamera2
  Using cached https://www.piwheels.org/simple/picamera2/picamera2-0.1.1-py3-none-any.whl (42 kB)
Requirement already satisfied: numpy in /home/pi/.pyenv/versions/3.8.19/lib/python3.8/site-packages (from picamera2) (1.24.4)
Collecting PiDNG (from picamera2)
  Using cached pidng-4.0.9.tar.gz (21 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting piexif (from picamera2)
  Using cached https://www.piwheels.org/simple/piexif/piexif-1.1.3-py2.py3-none-any.whl (20 kB)
Collecting pillow (from picamera2)
  Using cached pillow-10.4.0-cp38-cp38-manylinux_2_28_aarch64.whl.metadata (9.2 kB)
Collecting pyopengl (from picamera2)
  Using cached https://www.piwheels.org/simple/pyopengl/PyOpenGL-3.1.7-py3-none-any.whl (2.4 MB)
**Collecting PyQt5 (from picamera2)**
  Using cached PyQt5-5.15.11.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
[... 14 lines cut ...]
        File "/tmp/pip-build-env-gem4ibt8/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 608, in setup
          self.apply_user_defaults(tool)
        File "project.py", line 68, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/tmp/pip-build-env-gem4ibt8/overlay/lib/python3.8/site-packages/pyqtbuild/project.py", line 51, in apply_user_defaults
          super().apply_user_defaults(tool)
        File "/tmp/pip-build-env-gem4ibt8/overlay/lib/python3.8/site-packages/sipbuild/project.py", line 237, in apply_user_defaults
          self.builder.apply_user_defaults(tool)
        **File "/tmp/pip-build-env-gem4ibt8/overlay/lib/python3.8/site-packages/pyqtbuild/builder.py", line 49, in apply_user_defaults**
          raise PyProjectOptionException('qmake',
      sipbuild.pyproject.PyProjectOptionException
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Glacialisaurus rpicam-apps $ 
```

This is a serious problem. I just spent $80 on a Sony AI Camera, only to learn that it requires picamera2 for Python access, and because the instructions for installing picamera2 without GUI dependencies on my device are totally wrong, I cannot use this device. Why is this not being fixed?

naushir, 2024-12-09:
See https://github.com/raspberrypi/libcamera/issues/216.

closed 2024-12-09 by naushir