raspberrypi/picamera2 #1156: [BUG] The CnnOutputTensorInfo are zeros.

Andrewymd, 2024-11-06:
**Please only report one bug per issue!**

**Describe the bug**
A network with 11 output tensors encounters a situation where all fields in CnnOutputTensorInfo are zero when retrieving metadata.
example:
'CnnOutputTensorInfo': (0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0),
**To Reproduce**
Any model with greater than 8 output tensors.
metadata=picam2.capture_metadata()
print(metadata)

**Expected behaviour**
A clear and concise description of what you expected to happen.
Correct CnnOutputTensorInfo shoule be like below:
0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 7, 0, 0, 0, 0, 6, 0, 0, 3, 0, 0, 0, 2, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12, 0, 0, 3, 0, 0, 0, 4, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12, 0, 0, 3, 0, 0, 0, 4, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6, 0, 0, 3, 0, 0, 0, 2, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 9, 0, 0, 3, 0, 0, 0, 3, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 0, 0, 3, 0, 0, 0, 1, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 6, 0, 0, 3, 0, 0, 0, 2, 0, 24, 0, 32, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.
 Raspi4B + IMX500 sensor

**Additional context**
Add any other context about the problem here.

davidplowman, 2024-11-06:
Hi, thanks for the report.

Yes, there is in fact a hard-coded limit [here](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/controller/controller.h#L34). I don't think there's any reason we can't increase them, I think we've just been waiting to see what values people will want out in the wild. Any thoughts on that?

It'll take a little while for any change to make it into a release, so in the meantime you could check out and build libcamera for yourself.

Andrewymd, 2024-11-06:
Hi @davidplowman 
This is very strange. The CnnOutputTensor can actually retrieve 11 outputs normally, but all fields in CnnOutputTensorInfo are zero.
So if you want to use imx500.get_outputs(metadata, add_batch=True), the result returned here is empty.
However, you can directly get the CnnOutputTensor by metadata['CnnOutputTensor'], and then you need to manually set many parameters to ensure the parsing order is correct to achieve the same functionality.

Andrewymd, 2024-11-06:
> Hi, thanks for the report.
>
> Yes, there is in fact a hard-coded limit [here](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/controller/controller.h#L34). I don't think there's any [...]
I have tried modifying static constexpr unsigned int MaxNumTensors = 13; but there was no change—the CnnOutputTensorInfo is still all zeros, and its length remains fixed at 260.

naushir, 2024-11-06:
@Chizkiyahu, do you have any debug suggestions here?

davidplowman, 2024-11-06:
Ah, I think [these](https://github.com/raspberrypi/picamera2/blob/main/picamera2/devices/imx500/imx500.py#L23) definitions would have to match any new values in libcamera too.

Andrewymd, 2024-11-06:
Hi @davidplowman 
Thank you for your suggestions; they worked well. I updated the MAX_NUM_TENSORS to 15 in both libcamera and picamera, as 8 is too limited for multi-task networks. I’ll close the issue.

closed 2024-11-06 by Andrewymd

davidplowman, 2024-11-06:
Great, thanks for the update. Sounds like we should probably be changing the magic number to at least 16 in the official build.

Andrewymd, 2024-11-07:
yeah, I think 16 is better. i will submitting a PR on picamera2 and libcamera