raspberrypi/picamera2 #1160: [BUG]

julianstirling, 2024-11-11:
**Describe the bug**
Camera tuning can calculation can overflow causing alternating white and black lines when images are very red. This happens when using it as a microscope. We use a custom tuning file to do lens shading correction. A similar algorithm was developed for the old picamera stack and does not give this issue with the same hardware.

**To Reproduce**

*I apologise that this is significantly less self contained than you would like! But it gives you an idea of what we are doing.*

1. Take a PiCamera V2, [this 3D printed lens spacer](https://build.openflexure.org/openflexure-microscope/v7.0.0-beta2/models/lens_spacer_picamera_2_pilens.html), and this [3d printed platform](https://build.openflexure.org/openflexure-microscope/v7.0.0-beta2/models/camera_platform_picamera_2_pilens.html).  
2. Assemble the camera for microscopy as so:
[image]
3. Aim camera at white light (this may be easier if you build a whole [OpenFlexure Microscope!](https://build.openflexure.org/openflexure-microscope/v7.0.0-beta2/low_cost_microscope.html))
4. Load this customised [OS](https://build.openflexure.org/raspbian-openflexure/testing/) onto a pi.
5. Open the OpenFlexure software and run the autocalibration to generate a tuning file with alsc parameters which should correct for the pinkness around the edge caused by the change in chief ray angle.
6. Take image without a microscope slide and see it is pretty uniform grey. Image a blue stained microscope slide and see the colour is pretty consistent :+1: 
7. Aim camera at a glass slide covered in red ink from a red marker pen.
8. Get this weird cross with very strange artifacts in the grey area:

[image]

**Expected behaviour**
A uniform-ish red image. The alternating black and white lines in the grey area seem to imply there is some sort of overflow in the alsc calculation, in certain areas of the image if the pixels are too red.

**Console Output, Screenshots**

* [Example tuning filetuning.json](https://github.com/user-attachments/files/17704493/tuning.json)
Without the same lens set up this isn't going to be overly useful for reproducing the images.
* [Code that generates tuning file](https://github.com/labthings/labthings-picamera2/blob/main/src/labthings_picamera2/recalibrate_utils.py#L6)

**Hardware :**
Pi4, Pi Camera v2, Custom lens setup to make it a microscope as explained above.

**Additional context**
Tagging others who work on the OpenFlexure project.
@rwb27 @JohemianKnapsody

davidplowman, 2024-11-11:
Hi, thanks for the report. Can you provide a link to a DNG file that you've captured that gives rise to a bad LSC tuning? That would give me something I could investigate. Thanks!

julianstirling, 2024-11-11:
Thanks for the quick response @davidplowman. Would an `npz` that holds the a numpy array of the raw Bayer data work? Our software can spit that out. We will generate one tomorrow.

davidplowman, 2024-11-11:
Would prefer a DNG really. `cam.capture_file("test.dng", 'raw')` would do it. You could also run that through our `alsc_only.py` tool and see what comes out.

julianstirling, 2024-11-11:
I currently don't have hardware to hand so this was generated by someone else with:

```
with self.picamera() as cam:
    r = cam.capture_request()
    r.save_dng(path)
```

[Archive.zip](https://github.com/user-attachments/files/17707493/Archive.zip)

there is also a jpg. The DNG doesn't look raw? Do we need to redo with `cam.capture_file("test.dng", 'raw')`?

I have looked at `alsc_only.py` in the past. We didn't get good results. Our algorithm worked very well with the old picamera stack and works well for images that aren't red (see many stitched together here for a cervical smear scan with very uniform colour https://images.openflexure.org/cap_demo/viewer.html). 

Is there also a reason why the table is reduced to only 16x12 compared to the 52x39 that was possible in the old camera stack?

davidplowman, 2024-11-12:
I've taken a look, though I'm not quite sure what to make of it. A few comments:

* The LSC tables look plausible enough, though they're not super effective for the included DNG file.
* The white balance seems a bit off, everything is quite orange. Maybe try setting `"bayes"` to `0`?
* The colour matrix seems quite extreme. Maybe try disabling it to be sure nothing is overflowing? (Replace `"rpi.ccm"` by `"x.rpi.ccm"` to disable)

In case it's useful, here's what I wrote to process the DNG file into something viewable based on the tuning you have.
```
import cv2
import numpy as np
import json
import rawpy

dng = rawpy.imread("output.dng")
raw = np.maximum(dng.raw_image - 64, 0).astype(np.uint16)  # black level is 64 (out of 1023)
b = raw[0::2, 0::2]
gb = raw[0::2, 1::2]
gr = raw[1::2, 0::2]
g = ((gb.astype(np.uint32) + gr.astype(np.uint32)) / 2).astype(np.uint16)
r = raw[1::2, 0::2]
full_size = (r.shape[1], r.shape[0])

with open("red_cross_tuning.json", 'r') as f:
    tuning = json.load(f)

alsc = [algo for algo in tuning['algorithms'] if 'rpi.alsc' in algo][0]['rpi.alsc']
b_table = np.array(alsc['calibrations_Cb'][0]['table']).reshape(12, 16)
r_table = np.array(alsc['calibrations_Cr'][0]['table']).reshape(12, 16)
lum_table = np.array(alsc['luminance_lut']).reshape(12, 16)

b_gains = cv2.resize(b_table, dsize=full_size, interpolation=cv2.INTER_LINEAR)
r_gains = cv2.resize(r_table, dsize=full_size, interpolation=cv2.INTER_LINEAR)
lum_gains = cv2.resize(lum_table, dsize=full_size, interpolation=cv2.INTER_LINEAR)

b_gains *= lum_gains
r_gains *= lum_gains
b_gains /= np.min(b_gains)
r_gains /= np.min(r_gains)

b_corrected = b * b_gains
g_corrected = g * lum_gains
r_corrected = r * r_gains

wb_r, wb_g, wb_b, _ = dng.camera_whitebalance
wb_gain_r = 1 / wb_r
wb_gain_b = 1 / wb_b

rgb = np.empty((full_size[1], full_size[0], 3))
rgb[:, :, 0] = b_corrected * wb_gain_b
rgb[:, :, 1] = g_corrected
rgb[:, :, 2] = r_corrected * wb_gain_r
ccm = dng.color_matrix[:, :3][::-1, ::-1]  # rotate for BGR pixel order
rgb = np.maximum(rgb @ ccm, 0)
rgb *= 64  # make 10 bits into 16
rgb = np.power(rgb / 65536, 0.45) * 65536  # approximate gamma, so we can see something!
rgb = np.clip(rgb, 0, 65535).astype(np.uint16)

cv2.imshow("rgb", rgb)
cv2.waitKey(0)
```
Also, how are you capturing these images? They seem quite low resolution. Are you able to capture with `rpicam-still` as a comparison (using your tuning file)?

julianstirling, 2024-11-13:
Right. I have managed to reproduce this with `rpicam-still` and a custom tuning file.

So I made my custom tuning file with our software. And added an `x` in front of the "rpi.ccm", and also set bayes to zero.

I took an image of nothing with the white balance on. I got a nice image. I recorded the gains and using `--awbgains` option I was able to reproduce a nice white (grey) image.

I moved the microscope slide so I am looking directly at red pen

I then ran:

     rpicam-still --tuning-file tuning.json --awbgains .707,1.53 -o red-x.jpg -r 1
     
I still get the clear red-cross with the black and white artifacts:

[image]

The DNG has no cross and looks quite yellow. It has a bright centre and is dull around the edges. We expect this sort of ring shape due to the lenslet array when using at as a microscope. Hence the custom tuning.

When running your code above we get a uniform-ish image (screenshot):
[image]
Though it is more orange than red.

Tuning, DNG, and full-res jpg:
[redx.zip](https://github.com/user-attachments/files/17736225/redx.zip)

davidplowman, 2024-11-14:
Hi again. Still quite baffled by all this.

Looking more at it, I don't think LSC is the problem here. I think the clue is the black/white stripes down at the pixel level. They're mostly horizontal at the left/right edges, and vertical at the top/bottom. The colours tend to appear, or get worse, when the direction of these little edges is changing. This reminds me of extreme crosstalk problems, which can give rise to nasty aliasing-like artefacts, including strange colours. We could certainly imagine this kind of thing being exacerbated by the angle of the light rays.

I'm still not sure if it all adds up, but I might try changing the "rpi.geq" (green equalisation) settings in the tuning file. In the first instance, I'd try doubling the "offset" value that you find there. Keep doubling it until it has some effect (or not). It looks like 65535 is probably the largest possible value. If that actually achieves anything, we could consider how to find "good" tuning parameter later.

julianstirling, 2024-11-14:
@JohemianKnapsody telss me that setting rpi.geq to 5000 seems to solve the problem so we are certainly on the right track which is fantastic!

As we are building a microscope we are really keen on trying to make the image pipeline as deterministic and as understandable as possible so that two images can be directly compared. Is there a more detailed explanation of the green equalisation algorithm anywhere? (I have read the section in the tuning file)

davidplowman, 2024-11-14:
As I'm sure you know, the "green imbalance" problem is the green pixels on red rows (Gr pixels)can be slightly different to green pixels on blue rows (Gb).

The "green equalisation" algorithm basically averages Gr and Gb when they "don't look that different". The threshold below which things get averaged is given by
```
threshold = offset + pixel_value * slope
```
where `offset` and `slope` are the values in the tuning file, and pixel values are scaled to have 16 bits. So the trick is to find the lowest values that you can (therefore the "lowest" threshold line) but which fix the problems. Obviously there can be a bit of trade off between the `offset` and `slope` depending on the amount of imbalance you can get at different pixel brightness levels. Does that help a bit?

davidplowman, 2024-11-14:
Yes, good explanation. Really, one might want green imbalance correction to be variable across the image, and also to have spatially varying colour matrices. Unfortunately I don't know of any hardware that would support this!

julianstirling, 2024-11-14:
We have two different problems to solve. One is that the live preview doesn't look atrocious. And two is fixing images in post processing if we have the raw data. It's my understanding that if we jigger around with the `geq` value we can probably get something acceptable for live preview. Secondly, if we know we have this effect we can try to make a bespoke demosaicing algorithm for post processing the raw images and creating better saved images.

In terms of hardware we are also looking at the HQ camera which ups our bill of materials cost, but gets around the lenslet array issue!

Thank you for help, David! We now have a path forward which is fantastic. :heart_eyes_cat:

closed 2024-11-14 by julianstirling

julianstirling, 2024-11-14:
Sorry to keep bothering you @davidplowman, can I ask one final follow up question to check I have understood.

Am I right in my understanding that if we set the `geq` offest to 65535 then it will always average neighbouring green pixels?

This may actually be the way to go for a microscope, as the spacing between individual green pixels should be below or near the resolution limit of the microscope, but we are much more likely to get spurious effects.

davidplowman, 2024-11-14:
I think that's the case, though I can't 100% swear to the behaviour of both vc4 and PiSP based hardware off the top of my head. Certainly test it to check there's no weird corner case behaviour!

julianstirling, 2024-11-14:
Yeah, we can do some checks. A key one will be to check if doing this has and effect on our resolution. But assuming there are no corner cases this seems like a way forward for use.

Thank you again!