raspberrypi/picamera2 #1163: [BUG]  imx500_object_detection_demo.py: AttributeError: 'NetworkIntrinsics' object has no attribute 'bbox_order'

alexandre-perrin, 2024-11-15:
**Please only report one bug per issue!**

**Describe the bug**

Running [imx500_object_detection_demo.py](https://github.com/raspberrypi/picamera2/blob/main/examples/imx500/imx500_object_detection_demo.py) on Raspberry Pi 5 with IMX500 camera fails with: `AttributeError: 'NetworkIntrinsics' object has no attribute 'bbox_order'`

**To Reproduce**

On Raspberry Pi 5 with AI Camera:
execute the demo script with no parameters:

```
python imx500_object_detection_demo.py
```

**Expected behaviour**

Camera starts and application keeps running without crash.

**Console Output, Screenshots**

```
Traceback (most recent call last):
  File "/home/alexandre/aitrios-sdk-edge-app-dev/sample_apps/detection/python/imx500_object_detection_demo.py", line 246, in <module>
    last_results = parse_detections(picam2.capture_metadata())
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alexandre/aitrios-sdk-edge-app-dev/sample_apps/detection/python/imx500_object_detection_demo.py", line 30, in parse_detections
    bbox_order = intrinsics.bbox_order
                 ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NetworkIntrinsics' object has no attribute 'bbox_order'
```

**Hardware :**

Raspberry Pi 5 with AI Camera IMX500

**Additional context**

Current workaround: replace `intrinsics.bbox_order` with `args.bbox_order`

davidplowman, 2024-11-15:
Hi, thanks for the report.

I think the problem might just be that we're on the verge of rolling out a new camera software release, so all the code trees have been updated, but the underlying packages didn't quite make it before the weekend.

Perhaps try checking out out the v0.3.22 tag and see if that helps. Hopefully things will go back to normal at the start of next week. Sorry for the inconvenience.

ryota2425-murakami, 2025-02-06:
I can solve this problem by updating picamera2 with the following commands:

```
sudo apt update
sudo apt upgrade -y

sudo apt install -y python3-picamera2
```