raspberrypi/picamera2 #1170: [BUG] ValueError: numpy.dtype size changed

NMTHmations, 2024-11-22:
**Please only report one bug per issue!**

**Describe the bug**
I tried to run the following code in Thonny (under Python3), but I got Traceback, which indicates binary incompatibility:

from picamera2 import Picamera2
import Preview
import time
import cv2

picamera = Picamera2()
camera_config = picamera.create_still_configuration(main={"size":(1920, 1080)}, lores={"size": (640, 480)}, display="lores")
picamera.configure(camera_config)

picamera.start_preview(Preview.QTGL)
picamera.start()
#allow the camera to warm up
time.sleep(2.0)
#save image into file
picamera.capture_file("test.jpg")
picamera.stop_preview()
picamera.close()
#show image until a key is pressed
image = cv2.imread("test.jpg")
cv2.imshow("Image", image)
cv2.waitKey(0)
cv2.destroyAllWindows()

**To Reproduce**
Run with Python3, or in Thonny.

**Expected behaviour**
The program should start correctly.

**Console Output, Screenshots**
I got the following messages:

Traceback (most recent call last):
  File "/home/anemethhunor/Desktop/beagyazott/gyak8/7_access_camera.py", line 1, in <module>
    from picamera2 import Picamera2
  File "/usr/local/lib/python3.11/dist-packages/picamera2/__init__.py", line 11, in <module>
    from .picamera2 import Picamera2, Preview
  File "/usr/local/lib/python3.11/dist-packages/picamera2/picamera2.py", line 27, in <module>
    from picamera2.encoders import Encoder, H264Encoder, MJPEGEncoder, Quality
  File "/usr/local/lib/python3.11/dist-packages/picamera2/encoders/__init__.py", line 6, in <module>
    from .jpeg_encoder import JpegEncoder
  File "/usr/local/lib/python3.11/dist-packages/picamera2/encoders/jpeg_encoder.py", line 3, in <module>
    import simplejpeg
  File "/usr/lib/python3/dist-packages/simplejpeg/__init__.py", line 1, in <module>
    from ._jpeg import encode_jpeg
  File "simplejpeg/_jpeg.pyx", line 1, in init simplejpeg._jpeg
ValueError: numpy.dtype size changed, may indicate binary incompatibility. Expected 96 from C header, got 88 from PyObject
**Hardware :**
Raspberry Pi Zero 2W with Pi Camera (wide).

**Additional context**
Nope

NMTHmations, 2024-11-22:
I forgot to mention, the aforementioned numpy module's version is 2.1.3

andriilitvynchuk, 2025-10-03:
@NMTHmations hi, how did you solve this?