raspberrypi/picamera2 #1171: [BUG] After daily upgrade virtual picamera2 does not work

ZebrasHaveBigEars, 2024-12-01:
I am running picamera2 virtually on a raspberry 5.

This command works fine in system shell:
libcamera-vid --timeout 0000

This simple first code which has always worked now has an error:
from picamera2 import Picamera2, Preview
import time
picam2 = Picamera2()
camera_config = picam2.create_preview_configuration()
picam2.configure(camera_config)
picam2.start_preview(Preview.DRM)
picam2.start()
time.sleep(2)
picam2.capture_file("test.jpg")

[3:41:52.973046968] [3603]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[3:41:52.981003362] [3613]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[3:41:52.992696705] [3613]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[3:41:52.994471698] [3603]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+99-1230f78d
[3:41:53.003290089] [3616]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[3:41:53.016222260] [3616]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[3:41:53.020157522] [3603]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-BGGR_PISP_COMP1
[3:41:53.020364632] [3616]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected CFE format: 2028x1520-PC1B
Traceback (most recent call last):
  File "/home/pi/s1/cam1.py", line 7, in <module>
    picam2.start()
  File "/home/pi/s1/lib/python3.11/site-packages/picamera2/picamera2.py", line 1165, in start
    self.start_()
  File "/home/pi/s1/lib/python3.11/site-packages/picamera2/picamera2.py", line 1133, in start_
    self.camera.start(controls)
RuntimeError: Control type not implemented

davidplowman, 2024-12-02:
Hi, sometimes we need to do a Picamera2 update when the camera stack underneath has changed. So it may just be a case of updating the rest of the camera stack. I did just try this on a clean fresh system install and it seemed OK. Could you see if
```
sudo apt update
sudo apt full-upgrade
```
(and reboot) fixes this for you? (Always back up anything critical before a system update.) Thanks.

ZebrasHaveBigEars, 2024-12-06:
Thanks David for your excellent advice. I used these commands to update both the system and the virtual system and I get the same error.

davidplowman, 2024-12-06:
Can you say what you mean by "virtual picamera2"? If you're running in a venv, I'm wondering whether you might not see apt updates.

ZebrasHaveBigEars, 2024-12-07:
Yes - my picamera2 is in a venv. Sorry if I did not make this completely clear.  

By the way "libcamera-vid --timeout 0000" works fine both in the venv and when not running in a virtual environment.

ZebrasHaveBigEars, 2024-12-08:
Hi David

I used the following within my venv and it seems to have fixed the issue:

pip uninstall picamera2 rpi-libcamera
sudo apt update && sudo apt upgrade
pip cache purge
pip install picamera2 rpi-libcamera

Many thanks for your help which is very much appreciated.

mgineer85, 2024-12-13:
You want to create the venv in a way that system site packages are accessible, like this:
python -m venv --system-site-packages myenv

then you don’t install picamera2 in the venv because it’s available from the system.

this way works well for me, maybe for you too?