raspberrypi/picamera2 #1174: [BUG]Can't use pyside6 and picamera2 together

jd3096-mpy, 2024-12-05:
Here is the simple test code:
PART1:
```
import time, libcamera
from picamera2 import Picamera2, Preview

picam = Picamera2()

config = picam.create_preview_configuration(main={"size": (1600, 1200)})
config["transform"] = libcamera.Transform(hflip=1, vflip=1)
picam.configure(config)

picam.start_preview(Preview.QTGL)

picam.start()
time.sleep(2)
picam.capture_file("test-python.jpg")

picam.close()
```
PART2:
```
from PySide6.QtGui import QImage, QPixmap
from PySide6.QtWidgets import (
    QApplication,
    QLabel,
    QWidget,
    QHBoxLayout,
    QVBoxLayout,
    QPushButton,
)
from PySide6.QtMultimediaWidgets import QVideoWidget
from PySide6.QtGui import QFont, QFontDatabase
from PySide6.QtCore import QTimer, Qt, QDateTime, QUrl

class MyWindow(QWidget):
    def __init__(self, parent=None):
        super().__init__(parent)
        self.resize(800, 600)
        self.setupUI()

    def setupUI(self):
        self.btn_start = QPushButton("start")
        self.btn_stop = QPushButton("stop")
        h = QHBoxLayout()
        h.addWidget(self.btn_start)
        h.addWidget(self.btn_stop)
        v = QVBoxLayout()
        v.addLayout(h)
        self.setLayout(v)

import sys

app = QApplication(sys.argv)
window = MyWindow()
window.show()
sys.exit(app.exec())
```
I can run both parts separately without any issues, but when we put them together in a single Python file, it throws an error (regardless of which part runs first).

```
qt.qpa.plugin: From 6.5.0, xcb-cursor0 or libxcb-cursor0 is needed to load the Qt xcb platform plugin.
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/jd3096/Desktop/clock/lib/python3.11/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland, vkkhrdisplay, wayland-egl, linuxfb, vnc, minimalegl, eglfs, offscreen, xcb, minimal.

Aborted

```

In the new system of Raspberry Pi 5, it seems that the only way to use Python to access the camera is through the Picamera2 library. As far as I know, OpenCV doesn't work. So, are there any other ways to access the camera in PySide6?

davidplowman, 2024-12-05:
Hi, I'm afraid I don't know too much about using different GUI frameworks together, except that - in my experience, anyway - they often don't work together very nicely!

Maybe you could simply avoid using PyQt5 from Picamera2, thereby avoiding any conflicts. To do this, use `Preview.NULL` instead of `Preview.QTGL`. If you want to show camera images in your GUI, however, you'll have to forward them to your application. Obviously you'd have to figure out for yourself what the best way would be to render the images. Sorry that I can't be more helpful!

jd3096-mpy, 2024-12-05:
> Hi, I'm afraid I don't know too much about using different GUI frameworks together, except that - in my experience, anyway - they often don't work together very nicely!
>
> Maybe you could simply avoid [...]

Thank you for your suggestion; I think you're right. Perhaps I should wait for OpenCV to fix the Raspberry Pi 5 camera issue. Meanwhile, I'm trying alternative methods to solve the problem, such as running two Python processes simultaneously: one to write the captured images to a file and the other to handle the reading.

closed 2024-12-05 by jd3096-mpy

davidplowman, 2024-12-06:
Unfortunately I think you could be waiting a long time for OpenCV to add libcamera support - it's currently several years and counting with I think little sign of progress.

As I said, so long as you don't try and use the preview windows in Picamera2, I think you ought to be able to use the camera quite happily and pass images to your application - this would save the expense of writing and reading files which I would expect to be quite slow.