raspberrypi/picamera2 #1175: [BUG] memory allocation error

tim-oe, 2024-12-10:
**Please only report one bug per issue!**

**Describe the bug**
library failes to allocate memory while trying to take picture
the only way I can clear the error and get process working is to reboot 

**To Reproduce**
i cannot consistently reproduce this issue. it seems to happen if I interact with the system such as code/lib updates

**Expected behaviour**
the camera should allocate memory and take picture

**Console Output, Screenshots**
2024-12-08 17:53:02,670|CameraSvc.py:50|ERROR- failed to take picture
Traceback (most recent call last):
File "/home/tcronin/src/WeatherWatch/weatherwatch/camera/Camera.py", line 80, in process
picam2.switch_mode_and_capture_file(capture_config, imgFile, wait=True)
File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1467, in switch_mode_and_capture_file
return self.dispatch_functions(functions, wait, signal_function, immediate=True)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1363, in dispatch_functions
return job.get_result(timeout=timeout) if wait else job
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3/dist-packages/picamera2/job.py", line 79, in get_result
return self._future.result(timeout=timeout)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/concurrent/futures/_base.py", line 456, in result
return self.__get_result()
^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3.11/concurrent/futures/_base.py", line 401, in _get_result
raise self.exception
File "/usr/lib/python3/dist-packages/picamera2/job.py", line 48, in execute
done, result = self.functions[0](https://github.com/raspberrypi/picamera2/issues/887)
^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1430, in switch_mode
self.configure(camera_config)
File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1110, in configure
self.allocator.allocate(libcamera_config, camera_config.get("use_case"))
File "/usr/lib/python3/dist-packages/picamera2/allocators/persistent_allocator.py", line 29, in allocate
super().allocate(libcamera_config, use_case)
File "/usr/lib/python3/dist-packages/picamera2/allocators/dmaallocator.py", line 43, in allocate
fd = self.dmaHeap.alloc(f"picamera2-{i}", stream_config.frame_size)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/lib/python3/dist-packages/picamera2/dma_heap.py", line 98, in alloc
ret = fcntl.ioctl(self.__dmaHeapHandle.get(), DMA_HEAP_IOCTL_ALLOC, alloc)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [Errno 12] Cannot allocate memory

**Hardware :**
Raspberry Pi 4B rev 1.4 2g memory
camera: https://www.arducam.com/product/arducam-5mp-m12-picam/

**Additional context**
lib: python3-picamera2/stable,stable,now 0.3.23-1 all [installed]
OS: Debian GNU/Linux 12 (bookworm)

/boot/firmware/config.txt
`# _BT_DISABLE
dtoverlay=disable-bt
# _BT_DISABLE
enable_uart=1

# try to reduce temp...
arm_freq=1000
arm_freq_min=500
`

running code:
https://raw.githubusercontent.com/tim-oe/WeatherWatch/refs/heads/main/weatherwatch/camera/Camera.py
`import datetime
import time
from pathlib import Path

from conf.AppConfig import AppConfig
from conf.CameraConfig import CameraConfig
from picamera2 import Picamera2, Preview
from picamera2.allocators import PersistentAllocator
from py_singleton import singleton
from util.Logger import logger

@logger
@singleton
class Camera:
    """
    encapsulates camera functionality
    lib: https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf
    software: https://www.raspberrypi.com/documentation/computers/camera_software.html#rpicam-apps
    examples: https://github.com/raspberrypi/picamera2/tree/main/examples
    tuning: https://forums.raspberrypi.com/viewtopic.php?t=351189
    cli (for noir cam): libcamera-still --tuning-file /usr/share/libcamera/ipa/rpi/vc4/imx219_noir.json  --output preview.jpg
    TODO: https://github.com/raspberrypi/picamera2/issues/239
    work around: https://rockyshikoku.medium.com/use-h264-codec-with-cv2-videowriter-e00145ded181
    """  # noqa

    # second to micro second multiplier
    MICRO_SECOND: int = 1000000

    def __init__(self):
        """
        ctor
        :param self: this
        """

        # https://forums.raspberrypi.com/viewtopic.php?t=364677
        # default logging is warn
        Picamera2.set_logging()

        self._cameraConfig: CameraConfig = AppConfig().camera

        self._baseDir: Path = self._cameraConfig.folder

        self._baseDir.mkdir(parents=True, exist_ok=True)

    def process(self, lux: int) -> str:
        if self._cameraConfig.enable is False:
            self.logger.debug("camera not enabled")
            return

        try:
            tuning = None
            if self._cameraConfig.tuningEnable is True:
                tuning = Picamera2.load_tuning_file(self._cameraConfig.tuningFile)

            picam2 = Picamera2(tuning=tuning, allocator=PersistentAllocator())

            # TODO is this needed?
            time.sleep(1)

            picam2.start_preview(Preview.NULL)

            preview_config = picam2.create_preview_configuration()
            picam2.configure(preview_config)

            if lux <= self._cameraConfig.luxLimit:
                picam2.set_controls(
                    {
                        "ExposureTime": (self._cameraConfig.exposureTime * Camera.MICRO_SECOND),
                        "AnalogueGain": self._cameraConfig.analogueGain,
                    }
                )

            picam2.start()
            # TODO is this needed?
            time.sleep(1)

            imgFile: str = self.imageFile()
            capture_config = picam2.create_still_configuration()
            picam2.switch_mode_and_capture_file(capture_config, imgFile, wait=True)

            return imgFile
        except Exception as e:
            raise Exception("failed to take pic...") from e
        finally:
            picam2.stop_preview()
            picam2.stop()
            picam2.close()

    def imageFile(self) -> str:
        now = datetime.datetime.now()

        stamp = now.strftime("%Y-%m-%d-%H-%M-%S")
        imageName = f"{stamp}{self._cameraConfig.extension}"

        return str(self._baseDir / imageName)
`

davidplowman, 2024-12-10:
Hi, and thanks for the question.

So it sounds like you're running out of CMA memory. CMA is a somewhat limited resource and because there's no MMU that helps allocate it to applications, fragmentation is a risk, and this would cause the kind of failure you are seeing. It's possible, of course, that something is actually leaking, and if the behaviour was that we saw the free CMA gradually decreasing and failing reliably after N iterations, then I would certainly be suspicious.

I note that you're using the `PersistentAllocator`, which I think is probably the right thing to do. However, it looks like your `process` function is recreating the `Picamera2` object every time, which I think will throw away the benefits of the `PersistentAllocator`. If you're able to create the `Picamera2` object just once, maybe in your `__init__` method and hold on to it, then you could use `configure`/`start`/`stop` in your `process` method, and this should be much more reliable.

The other thing to try would be to increase the amount of CMA. Check out section 8.3 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf), though fixing the use of the `PersistentAllocator` in the first instance is certainly the thing to do. Note that you can monitor the amount of CMA using `grep -i cma /proc/meminfo`.

To answer a couple of your other questions:

* There's no need for a `sleep` after `configure`. The system isn't doing anything while you wait.
* You shouldn't need a `sleep` after `start` either if you've set the exposure and gain controls, as those take effect immediately on the first frame. You'll still get some AWB adaptation, but it should be quite minor. When you don't set those controls, there may still be some AGC/AEC adaptation going on in the first few frames, but it shouldn't be too great - a whole second is probably more than you need. But you'll have to experiment a bit.

tim-oe, 2024-12-10:
thank you so much for the quick reply and detailed critique.

i started looking into refactoring and realized that i moved the camera object creation because i couldn't find a good way of reseting the controls back to defaults:
`            if lux <= self._cameraConfig.luxLimit:
                self._picam2.set_controls(
                    {
                        "ExposureTime": (self._cameraConfig.exposureTime * Camera.MICRO_SECOND),
                        "AnalogueGain": self._cameraConfig.analogueGain,
                    }
                )
`
i tried setting picam2.controls = new Controls(picam2) but didn't clear values

looking the set_controls it appears to only add setting and does not clear any existing settings.

tim-oe, 2024-12-11:
in mucking around with it i got this to pass with my test, the pic it generates looks ok but not confident it's the right approach.

`
            controls = {}
            if lux <= self._cameraConfig.luxLimit:
                controls = {
                    "ExposureTime": (self._cameraConfig.exposureTime * Camera.MICRO_SECOND),
                    "AnalogueGain": self._cameraConfig.analogueGain,
                }
            else: # clear controls
                controls = {
                    "ExposureTime": 0,
                    "AnalogueGain": 0,
                }

            self._picam2.start()

            time.sleep(1)

            print(f"picamera contols {pprint(self._picam2.camera_configuration())}")

            imgFile: str = self.imageFile()
            capture_config = self._picam2.create_still_configuration(controls=controls)
`

davidplowman, 2024-12-11:
Thanks for the update.

As you've noticed, controls persist for the lifetime of the Picamera2 object, and are always incremental, in the sense that you can set new values but there's no way to return something to a default value.

In theory you can look at `picam2.camera_controls` to find out what the defaults are (the 3rd value of each tuple in the dictionary), but I'm not 100% confident it's always entirely correct. In your case, it's probably worth just knowing what controls you're interested in, and setting the values you feel are appropriate for you. So for exposure/gain, zero values are good for putting everything back into "auto" mode.

Putting default control values into the configuration object directly, as you've done, also feels like a good way to go. That way you can change them, but re-applying that configuration will always set them back.

If you have any questions about what particular controls do, and what kinds of values might suit your use case, we're always happy to try and help.

tim-oe, 2024-12-12:
I've followed all your recommendations and have the changes deployed.
will keep an eye on CMA.

thank you for your prompt and detailed responses been extremely helpful.

closed 2024-12-12 by tim-oe