raspberrypi/picamera2 #1183: [BUG] Memory Leak with capture_array

Vanlon, 2024-12-23:
**Please only report one bug per issue!**

I am using capture_array to capture an image to memory and then process it with opencv.  After the image is captured and passed to another function it does not release it in memory.  Every loop will add another variable to the memory that isn't released and eventually my Raspberry Pi 4 will run out of memory and kill the process.

Here is a dumbed down version of my code that reproduces the problem.  I've added tracemalloc to identify where the memory usage stems from.

`
  from picamera2 import Picamera2
  import tracemalloc
  import time
  import cv2
  
  def takepic():
      
      tracemalloc.start()
  
      picam2 = Picamera2()
      preview_config = picam2.create_still_configuration(main={"size": (4000, 3000)})
      picam2.configure(preview_config)
      picam2.start()
      frame = picam2.capture_array()
      snapshot = tracemalloc.take_snapshot()
      top_stats = snapshot.statistics('lineno')
  
      for stat in top_stats[:2]:
          print(stat)
      
      #picam2.stop()
      picam2.close()
      resizeit(frame)
  
  def resizeit(img):
      small = cv2.resize(img, (0,0), fx=0.5, fy=0.5)
      del img
      del small
      time.sleep(1)
      takepic()
      
  takepic()
`

I would expect that the variable gets released or overwritten every time the code loops instead of added to memory.  Using free -m I can see the amount of memory decreasing every loop

[image]

[image]

I am using a Raspberry Pi 4B with 4Gb of ram and an ELP 13Mp autofocus camera

davidplowman, 2025-01-03:
Hi, and thanks for the report.

I notice that your code calls `takepic()` recursively, meaning that the variables in the `takepic` stack frames might never be released. Does that sound plausible?

Perhaps the first thing to do is to rewrite the example slightly to avoid hanging on to those local variables. Is that something you could try, and then post back if you still have a problem? Thanks!

Vanlon, 2025-01-14:
I've tried for a week to write this so it releases the local variable.  In my program it's not recursive like this but still works similar with a trigger.  If you have any insights that would be helpful.

davidplowman, 2025-01-15:
Hi again, could you post a short example that reproduces the problem that you are experiencing? As I said, the memory problems in your previous example seem to be explainable, so it doesn't really help so much. Thanks!