raspberrypi/picamera2 #1184: [BUG] Huge difference in performance using picamera2 vs libcamera-vid on RPi 5

azsde, 2024-12-23:
Hello,

I am trying to achieve 1080p@50-60 FPS recording on my Raspberry Pi 5 using picamera2 but I am having issues.

When using libcamera-vid:

`libcamera-vid --level 4.2 --framerate 60 --width 1920 --height 1080 -o test-libcamera.mp4 -t 60s --denoise cdn_off -n`

And then when I count the number of frames:

`ffprobe -v error -select_streams v:0 -count_packets -show_entries stream=nb_read_packets -of csv=p=0 test-libcamera.mp4`

I get `3527`, which is quite close to the 3600 frames I am supposed to be having at 60FPS (60 frames per seconds * 60 seconds ), I have only a 2% loss of frames which is quite decent.

However, when using picamera2:

```
#!/usr/bin/python3
import time

from libcamera import controls

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (1920, 1080)}, controls={'FrameRate': 60})
picam2.configure(video_config)
picam2.set_controls({"NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Off})

encoder = H264Encoder()
output = FileOutput('test-picamera2.h264')

picam2.start_recording(encoder, output)
time.sleep(60)
picam2.stop_recording()
```

And then again, when I count the number of frames:

`ffprobe -v error -select_streams v:0 -count_packets -show_entries stream=nb_read_packets -of csv=p=0 test-picamera2.h264`

I get `2345` which is way further from the 3600 frames I am supposed to get, with a loss of 35% !

 I am very surprised by these results. Is this behavior expected ?
 
 Many thanks in advance for your help !

sandyol55, 2024-12-24:
You don 't mention which camera module is being used, but the most likely reason is the modes that libcamera-vid and picamera2 have chosen to run the camera in.

libcamera-vid has most likely chosen a fast but cropped mode to try and achieve the framerate you've requested, while picamera2 has chosen a higher resolution but slower mode to attempt to match the requested 1920x1080 format. 

A check of the INFO output message for the 'Selected sensor format' and 'Selected CFE format' each method has chosen may give a clue, or a check on the width of field of view of both recorded videos may also indicate the differing choices.

'The Sensor Configuration' guide in Section 4.2.2.3 of the Picamera2 manual gives some guidance on methods to 'force' picamera2 to select a specific mode. https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf

And https://www.raspberrypi.com/documentation/computers/camera_software.html#list-cameras indicates how to list the available modes.

azsde, 2024-12-24:
Hi @sandyol55 and thank you for your answer.

I am using the camera module v3, and indeed, it would seem that the format selected is different:

libcamera-vid:

```
[24:35:02.967150978] [77300]  INFO Camera camera.cpp:1197 configuring streams: (0) 1920x1080-YUV420 (1) 1536x864-BGGR_PISP_COMP1
    [24:35:02.967248700] [77305]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected CFE format: 1536x864-PC1B

```

picamera2: 

```
[24:36:29.561050846] [77422]  INFO Camera camera.cpp:1197 configuring streams: (0) 1920x1080-XBGR8888 (1) 2304x1296-BGGR_PISP_COMP1
[24:36:29.561170217] [77425]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected CFE format: 2304x1296-PC1B
```

Please excuse my ignorance, but I do not understand what is "CFE", is it the sensor "resolution" ?

If so, I am not sure I understand how I can achieve a 1920x1080 footage with a CFE format that is "less" that the requested dimensions ?

Would you be enable to enlighten me ? I thank you in advance for your help !

sandyol55, 2024-12-24:
The V3 camera module has 3 operating modes.
```
Modes: 'SRGGB10_CSI2P' : 1536x864 [120.13 fps - (768, 432)/3072x1728 crop]
                         2304x1296 [56.03 fps - (0, 0)/4608x2592 crop]
                         4608x2592 [14.35 fps - (0, 0)/4608x2592 crop]
```
By changing your libcamera-vid command you have now allowed both methods to choose mode 1, which has a maximum possible framerate of 56fps.  libcamera is getting closer than picamera2 to the maximum 56fps but neither can achieve 60 fps in that mode.
To get picamera2 to choose mode 0, (with its faster maximum rate),  you can add a sensor definition in the configuration, something like this:-
```
#!/usr/bin/python3
import time

from libcamera import controls

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

picam2 = Picamera2()

# Fastest framerate mode (0  for V3 Camera)
mode=picam2.sensor_modes[0]
# Pick out key parameters from mode
sensor={'output_size': mode['size'],'bit_depth':mode['bit_depth']}
# And apply them to the configuration
video_config = picam2.create_video_configuration(main={"size": (1920, 1080),"format":"YUV420"},
                                                 sensor=sensor,
                                                 controls={'FrameRate':60,
                                                           "NoiseReductionMode" : controls.draft.NoiseReductionModeEnum.Off})

picam2.configure(video_config)

encoder = H264Encoder()
output = FileOutput('test-picamera2.h264')

picam2.start_recording(encoder, output)
time.sleep(60)
picam2.stop_recording()
picam2.close()
```
(This code snippet also changes to the YUV420 format as that is likely to more computationally efficient than the default format)

With these changes picamera2 should now get closer to 60 fps.

Although mode 0 only has a resolution of 1536x864 , both libcamera and picamera2 will be upscaling that to the requested 1920x1080, before passing it to be encoded.

azsde, 2024-12-24:
@sandyol55 thank you so much, the whole sensor stuff it is still not 100% clear for me but at least I get things working just fine by specifying the sensor mode.

One last thing, if I'm ok reducing the FPS a little bit, do you agree that selecting mode 1 will provide a better image quality since there won't be any upscaling done ?

```
[{'bit_depth': 10,
  'crop_limits': (768, 432, 3072, 1728),
  'exposure_limits': (9, 77208384, None),
  'format': SRGGB10_CSI2P,
  'fps': 120.13,
  'size': (1536, 864),
  'unpacked': 'SRGGB10'},
 {'bit_depth': 10,
  'crop_limits': (0, 0, 4608, 2592),
  'exposure_limits': (13, 112015443, None),
  'format': SRGGB10_CSI2P,
  'fps': 56.03,
  'size': (2304, 1296),
  'unpacked': 'SRGGB10'},
 {'bit_depth': 10,
  'crop_limits': (0, 0, 4608, 2592),
  'exposure_limits': (26, 220417486, None),
  'format': SRGGB10_CSI2P,
  'fps': 14.35,
  'size': (4608, 2592),
  'unpacked': 'SRGGB10'}]
```

sandyol55, 2024-12-24:
Yes, assuming the ~56 fps suits your requirement, then I'd expect that  the downscaling from mode 1, instead of the upscaling from mode 0, would be perceptibly higher in quality - maybe not by much though?? .

azsde, 2024-12-30:
Closing as this is resolved thanks to @sandyol55

closed 2024-12-30 by azsde