raspberrypi/picamera2 #1193: [HOW-TO] Using PiCamera 2 with Flutter Without Flask/Web Server

Akshar062, 2025-01-22:
I am running Flutter on a Raspberry Pi 5 and trying to use PiCamera 2 (Python) to capture images and display a live preview in my Flutter app. I am currently using subprocess in Dart to run a Python script, but I haven’t been able to get the preview working.

Questions

- Has anyone successfully integrated PiCamera 2 with Flutter using a non-web-based approach?
- What’s the best way to pass image data from Python (PiCamera 2) to Flutter efficiently?
- Are there any existing libraries or examples for this type of integration?

davidplowman, 2025-01-22:
Hi, I'm afraid I know nothing at all about Flutter, so I probably can't help. However, it sounds like you're running the Picamera2 Python script in a separate process from your Flutter application, is that correct?

* Are you wanting the Python process to display a preview window? I can't think of a reason why that wouldn't work if the Python script works in isolation (does it?). If this is what's happening, are there any error messages?
* Or are you wanting to get the images back to your Flutter process, so that they can be displayed there? The image buffers are dmabufs so in principle you could send the file descriptors across and avoid copying all the pixel data. If this is what you want to do, there is an [example](https://github.com/raspberrypi/picamera2/blob/main/examples/picamera2_multiprocessing.py) of zero-copy buffer passing between Python processes - maybe that helps a bit?

Akshar062, 2025-01-22:
Hi, thanks for your response! Yes, I am running the Picamera2 Python script as a separate process from my Flutter app. I don’t want to open a built-in GUI window for the preview. Instead, I want to continuously receive frames or a video stream from Picamera2 so that I can display the preview inside my Flutter app wherever I want.

I’m looking for a way to get the frames without using the default preview window, and I’d prefer to receive them efficiently (e.g., via a socket, shared memory, or another optimized method). If you have any suggestions or examples, especially for zero-copy buffer passing between Python processes, that would be really helpful. Thanks!

davidplowman, 2025-01-22:
Picamera2 won't try to open a preview window unless you ask for one, so I don't foresee any problems there.

I think the most efficient way to get images across will be the zero-copy technique using file descriptors. You'll probably need a couple of fifos. The first will be for sending file descriptors across (you'll have to "acquire" the completed requests on the Python side to stop them being recycled to the camera), perhaps along with some image format information. The other fifo will be for a receiving thread on the Python side that gets notified by the Flutter app  that you're finished with the image, and it can then "release" the request back to the camera.

All this happens in the [Process](https://github.com/raspberrypi/picamera2/blob/main/examples/picamera2_multiprocessing.py#L18) class in that example. Though there, of course, it's Python to Python, which maybe makes some things easier, but it should give you some hints.

There are certainly other ways to do this. If your images aren't too big and you have a fast Pi (a Pi 5 definitely qualifies!) then simply copying the pixel data into a fifo/socket would be an option. Although less efficient, it would be much easier to get up and running. You could always start like that, and see if the performance is OK.

closed 2025-03-17 by Akshar062