raspberrypi/picamera2 #1199: [BUG] Running Picamera2@0.3.12 inside a docker container get Broken PIpe error.

DrChai, 2025-01-28:
```python
Python 3.11.11 (main, Jan  9 2025, 04:46:56) [GCC 14.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import libcamera
>>> libcamera.CameraManager.version
'v0.0.5+83-bde9b04f-dirty ()'
>>> from picamera2 import Picamera2, Preview
>>> picam2 = Picamera2()
[0:27:46.704043329] [17]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f-dirty ()
[0:27:46.808519116] [21]  WARN RPI vc4.cpp:383 Mismatch between Unicam and CamHelper for embedded data usage!
[0:27:46.809969693] [21]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media0 and ISP device /dev/media2
[0:27:46.810155840] [21]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/local/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
>>> preview_config = picam2.create_preview_configuration(main={"size": (800, 600)})
>>> picam2.configure(preview_config)
[0:27:53.333451265] [17]  INFO Camera camera.cpp:1033 configuring streams: (0) 800x600-XBGR8888 (1) 1640x1232-SBGGR10_CSI2P
[0:27:53.335407521] [21]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 1640x1232-SBGGR10_1X10 - Selected unicam format: 1640x1232-pBAA
[0:27:55.341574894] [21] ERROR IPCPipe ipc_pipe_unixsocket.cpp:134 Call timeout!
[0:27:55.341687446] [21] ERROR IPCPipe ipc_pipe_unixsocket.cpp:66 Failed to call sync
[0:27:55.341720936] [21] ERROR IPAProxy raspberrypi_ipa_proxy.cpp:370 Failed to call configure
[0:27:55.341759582] [21] ERROR RPI pipeline_base.cpp:1201 IPA configuration failed!
[0:27:55.341806874] [21] ERROR RPI pipeline_base.cpp:567 Failed to configure the IPA: -32
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.11/site-packages/picamera2/picamera2.py", line 1004, in configure
    self.configure_(camera_config)
  File "/usr/local/lib/python3.11/site-packages/picamera2/picamera2.py", line 943, in configure_
    if self.camera.configure(libcamera_config):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Failed to configure camera: Broken pipe

```
I've been trying to run `picamera2@0.3.12`  in a docker container for the past week. The reason for using this specific version is that our project requires only picture captures from the camera to be sent to a server for vision modeling. We do not need streaming or video features (e.g., ffmpeg) and want to make it less footprint.  Version `0.3.12` appears to be the latest stable version that does not depend on PyAV/ffmpeg. In addition, it is still the current version you can get when you `apt install python3-picmaera2` from [armhf package list](https://archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/Packages#:~:text=picamera2%0ASource%3A%20picamera2-,Version%3A%200.3.12%2D2,-Architecture%3A%20all%0AMaintainer)(used on RPi zero).  
To run this in an Alpine based container, I had to build nearly all dependencies from the source. The current error appears to be the final hurdle. From the error logs, the camera is detected, but the configuration fails. Upon looking at the line of the code, it seems to be a socket issue.

What I have done so far:
- I can run picamera2@0.3.12 successfully in the host environment by installing the library via `python3-picamera2`. This uses the same versions of picamera2(0.3.12) and libcamera(v0.0.5+83-bde9b04f).
- I tried to build it with Debian image and encountered the same result.
- I tested it on RPi Zero, RPi 5 and RPI Zero2, the same result.  
- I can successfully build and run in a container using the same Docker build steps, but with the source updated to the latest version of picamera2 and libcamera. 

 Here are all the [Dockerfiles](https://github.com/DrChai/docker-picamera2/blob/main/alpine/Dockerfile) and supporting files I used. Right now it works as expected with the latest source code hope it may help when this happens to other people.

DrChai, 2025-12-03:
@petekubiak I haven’t had any issues on Debian, I just tested by building a Debian container using `python:slim`, and picamera2 worked as expected. In my case, the problem only occurred with the latest version for armhf/v6.
I’ve uploaded my [Dockerfile](https://github.com/DrChai/docker-picamera2/blob/main/debian/Dockerfile) I used. You can have a try with your specific version, like: 
`docker build --build-arg LIBCAMERA_VER=v0.5.0+rpt20250429. --build-arg PICAMERA2_VER=0.3.31 -t picamera2 .
`

petekubiak, 2025-12-05:
I've fallen back to pure raspios to get the project unblocked for now and picked up another project, but I'll try to find some time to try those things and see what the result is, thanks!