raspberrypi/picamera2 #1200: [BUG] bizarre colors of the image

RaspiStack, 2025-02-01:
I am using a PiCam2 camera with a Raspberry Pi 3B+. Using the following code, I get the image below. The colors are not accurate: in reality, the sweater is dark red/burgundy, and the furniture is dark blue. In the image, the colors appear 'washed out'. The photo was taken in the evening, so with the room light on. I know that artificial lighting can affect the colors, but this much seems surprising. With daylight, the colors are better, but they still lack vibrancy. Is there a specific setting to adjust?

	#init Camera
	picam2 = Picamera2()
	camera_config = picam2.create_still_configuration(main={"size" :(1280, 960)}, lores={"size" : (640,480)})

	picam2.configure(camera_config)
	picam2.set_controls({"AfMode": 1 ,"AfTrigger": 0, "AwbMode":1})

	#Capture a picture
	picam2.start_preview(None)
	picam2.start()
	time.sleep(1)
	picam2.capture_file(Path)

[image]

RaspiStack, 2025-02-02:
I’m adding a photo taken with daylight. It’s better, but some colors are very faded. The cousin is yellow in reality, but here it looks gray.

If anyone has an idea of the settings to adjust, taht would be wonderful..!

[image]

davidplowman, 2025-02-03:
Hi, can you say exactly what camera you're using? It looks to me a bit like you might have a NoIR model (a camera with the IR filter removed). These cameras don't produce true colours because they're sensitive to IR, the amount of which is highly variable.  Holding an IR filter over the lens would improve things if that's the cause, otherwise do you have a regular camera module to try?

RaspiStack, 2025-02-03:
Hi Davidplowman !

Indeed, it's a NoIR webcam, more specifically a Raspberry Pi Camera Module 3 Wide NoIR - 12MP 120 Degree. I didn't see on the Adafruit or Raspberry Pi page that the colors wouldn't be completely accurate. It's still a bit surprising to release a camera on the market with incorrect colors. What’s the point of offering such a camera?

tvoverbeek, 2025-02-03:
Simplw. With no InfraRed filtered out you can take movies/pictures in the dark, especially if you light the scene with an IR lamp (not visible to humans).
This (= No IR filter) is what night surveillance cameras use.

RaspiStack, 2025-02-09:
Many thanks for all your comments. Next time, I'll be more cautious on the choice of the camera..

closed 2025-02-09 by RaspiStack