raspberrypi/picamera2 #1201: [HOW-TO] Any way to reliably detect dropped frames

zkelly1, 2025-02-05:
**Describe what it is that you want to accomplish**
I would like a reliable method to detect dropped frames (and the number thereof) that does not significantly further damage FPS. This is acceptable in either CPP or Python. 

**Describe alternatives you've considered**
I've tried using the metadata in Python, but that halves the FPS (200->100) and is also not correlated with frame drops. I've also tried timing it manually with `time.perf_counter_ns()`, but this induces dropped frames (200->194 in a second) and I am weary to delve into it further given that it is not directly attributed to the camera capture. I've tried the `sequence_number` in C++, which does catch when a frame is dropped, despite this being listed as its function in the libcamera documentation, and I've also tried the timestamps in C++ with libcamera. 

**Additional context**
I've tried implementing various approaches of synchronizing two cameras that are not both supported by picamera2. One is, the other is not. The reason I have to use these cameras is that this rolling shutter picamera-supported camera, IMX219, has the FPS that I need (200+), and the other is an eye-tracker from a specific brand that is compatible with their software. I've written code to capture from both cameras several times with CPP and Python. I've found that, ironically, Python stops frames from being dropped by the IMX219 when used on its own. However, I cannot effectively do synchronized multiprocessing in Python due to the GIL and also a need for it to be tightly synchronized.

davidplowman, 2025-02-05:
Hi, thanks for the questions. There's quite a lot there, so sorry if the answer may be a bit rambling!

Firstly, there is nothing that counts dropped frames, so far as I know. If libcamera itself is dropping frames then the Python layers will never know about it, so I suppose the lower you go down the stack the more chance you have of noticing. Though even then, if the frame interrupts are happening so fast that nothing can keep up, then things will get confused.

Having said that, if you're running at a fixed framerate, then I'd have thought the timestamps are quite a good way of detecting problems - and you can do that up in Python just as easily.

You don't say what platform you're using, so that might have an effect too. In my case, on a Pi 5, I can run a loop with `picam2.capture_metadata()` and (with an IMX219) that keeps up at 206 FPS. But it's true, things will become horribly sensitive to the amount of processing you're doing, the number of frame buffers you let the camera system have, and so on. Picamera2 has got better over time at allowing you to do stuff without copying pixel data, so that may be something to look at.

Multi-threading in Python is obviously limited, but multi-processing may not be out of the question. Our camera buffers are dmabufs, which means that you can pass them to other processes and mmap them with no pixel copying at all. You clearly have to manage the lifetime of the request that you capture while another process is using the buffer, but there is even an example showing one way you could do that kind of thing ([here](https://github.com/raspberrypi/picamera2/blob/main/examples/picamera2_multiprocessing.py)).

zkelly1, 2025-02-05:
No worries at all! Thanks for the fast reply. It's really interesting to me that you are able to get 206 FPS via metadata. I'm also running on a RPi5, with the following code, and I get a total of `World | Captured: 3006 frames` frames captured for a `30` second interval when I've set the `frame_duration` equal to 200 FPS (could this be a problem?). I also noticed this pattern if I change the frame duration to other numbers, if I use the call to metadata, it is always halved. I would be very open to using timestamps so long as I could capture at the correct FPS. I will take a look at the multiprocessing example too! Thanks for sharing that. I was naively trying to just put the 2 sensor recorders into their own processes and got deadlocks. 

Here's my code

```
# Record a video for a set duration and populate a buffer 
def record_world(world_cam: picamera2.Picamera2, buffer: np.ndarray, 
                 total_duration: int, chunk_duration: int, downtime_duration: int) -> None:
  
    # Calculate how many chunks we will record 
    assert(total_duration % chunk_duration == 0)
    num_chunks: int = total_duration // chunk_duration

    # Record for the desired number of chunks 
    for i in range(num_chunks):
        # Track which frame number we're on
        frame_num: int = 0

        # Start the capture 

        world_cam.start('video')

        # First metadata capture is always long, so do it outside of the timed loop
        world_cam.capture_metadata()

         # Define the start time of the capture 
        start_time: float = time.time()

        print(f"World | Starting chunk: {i} @ {start_time}", flush=True)

        while(True):
            # Retrieve the current time 
            current_time: float = time.time()

            if(current_time - start_time >= chunk_duration):
                break 

            # This causes a halve in frame rrate 
            metadata = world_cam.capture_metadata()

            world_frame: np.ndarray = world_cam.capture_array('raw')[:, 1::2]

            buffer[frame_num] = world_frame

            # Increment the frame number 
            frame_num += 1 

        # Stop capture while we output the buffer 
        world_cam.stop()

        # Output the result
        start_write_time: float = time.time()

        print(f'World | Captured: {frame_num} frames', flush=True)

        #np.save(os.path.join(os.path.dirname(__file__), "/media/rpiControl/FF5E-7541/test_folder", f"world_chunk_{i}.npy"), buffer[:frame_num])
        end_write_time: float = time.time() 

        time.sleep(downtime_duration - (end_write_time - start_write_time))

    return 

def main():
    # The duration of the recording in seconds
    total_duration: int = 300 # 5mins 

    # The duration of chunks in seconds 
    chunk_duration: int = 30 # 30 second chunks 

    # The standardized amount of downtime in seconds 
    downtime_duration: int = 15

    # Initialize the camera
    print("Initializing Cameras")
    
    world_cam: picamera2.Picamera2 = initialize_world_camera()
    #pupil_cam: uvc.Capture = initialize_pupil_camera()

    # Capture a frame from the pupil cam, it takes a second to start 
    #pupil_cam.get_frame_robust()

    # Allocate the buffer of memory
    print("Allocating Buffers") 
    world_buffer: np.ndarray = np.empty(((chunk_duration + 1) * WORLD_CAM_FPS, *WORLD_FRAME_SHAPE), dtype=np.uint8)
    #pupil_buffer: np.ndarray = np.empty(((chunk_duration + 1) * PUPIL_CAM_FPS, *PUPIL_FRAME_SHAPE), dtype=np.uint8)

    # Record for the desired duration and fill the buffer 
    print("Recording")
    record_world(world_cam, world_buffer, total_duration, chunk_duration, downtime_duration)

    print(f'Closing cameras')
    world_cam.close()
    #pupil_cam.close()
```

zkelly1, 2025-02-07:
Thanks so much! All of your help completely solved my issues!

closed 2025-02-07 by zkelly1