raspberrypi/picamera2 #1206: [HOW-TO]can't change the sensor bit_depth on pi5

HcosYu, 2025-02-07:
I am using a Pi5 and I want to change the configuration of the Picamera. I want to set the sensor's bit depth to 16. However, the result I get when running my program is 12. How can I change the bit depth to 16?

```python
from picamera2 import Picamera2

def set_raw_still_mode():
    picam2 = Picamera2()
    raw_mode_parameter = {
        'use_case': 'still',
        'buffer_count': 4,
        'main': {
            'size': (2028, 1520), 
            'format': 'BGR888', 
            'stride': 6144, 
            'framesize': 9338880
        },
        'raw': {
            'size': (2028, 1520), 
            'format': 'SBGGR12', 
            'stride': 3072, 
            'framesize': 4669440
        },
        'sensor':{'bit_depth': 16}  
    }
    
    picam2.configure(picam2.create_still_configuration(
        main={"size": raw_mode_parameter['main']['size'], "format": raw_mode_parameter['main']['format'], "stride": raw_mode_parameter['main']['stride']},
        raw={"size": raw_mode_parameter['raw']['size'], "format": raw_mode_parameter['raw']['format'], "stride": raw_mode_parameter['raw']['stride']}, buffer_count=4,
        sensor={"bit_depth":16},
    ))
    
    final_configuration = picam2.camera_configuration()
    print("final camera configuration", final_configuration)
    
    picam2.start_preview()
    picam2.start()
    
    picam2.capture_file("test_bit_depth16.jpg")
    picam2.stop_preview()

set_raw_still_mode()

```

and the output configuration:

```
final camera configuration {'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 
'main': {'format': 'BGR888', 'size': (2028, 1520), 'preserve_ar': True, 'stride': 6144, 'framesize': 9338880}, 
'lores': None, 
'raw': {'format': 'SBGGR16', 'size': (2028, 1520), 'stride': 4096, 'framesize': 6225920}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (100, 1000000000)}, 
'sensor': {'bit_depth': 12, 'output_size': (2028, 1520)}, 'display': None, 'encode': None}

```

HcosYu, 2025-02-07:
BTY, my raspberry pi5 version is :

> pi@raspberrypi:~ $ cat /etc/os-release
> PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
> NAME="Debian GNU/Linux"
> VERSION_ID="12"
> VERSION="12 (bookworm)"
> VERSION_CODENAME=bookworm
> ID=debian
> HOME_URL="https://www.debian.org/"
> SUPPORT_URL="https://www.debian.org/support"
> BUG_REPORT_URL="https://bugs.debian.org/"

davidplowman, 2025-02-07:
Hi, I'm afraid the IMX477 (HQ Cam) doesn't have a 16-bit sensor mode. If you run `rpicam-hello --list-cameras` in a terminal window, it will list all the available camera modes. For the IMX477, most of them are 12-bit, though there might also be 10-bit mode which gives you low resolution but faster framerates, IIRC.

HcosYu, 2025-02-07:
Thank you for your response! I would like to ask another question. In the response at [here](https://github.com/raspberrypi/picamera2/issues/1010#issuecomment-2065941209), mentioned that the Pi 5 provides 16-bit data for all raw formats. If the IMX477 camera is 12-bit, does that mean the raw data will be left-shifted by 4 bits?
Have a good day!

davidplowman, 2025-02-07:
Yes, that is correct. The bottom (least significant) 4 bits will all be zero (which is different from earlier Pis where the top bits would be zero). On a Pi 5 it's only the _sensor_ part of the camera configuration that tells you how many bits the sensor provided.

HcosYu, 2025-02-07:
I have another related question to ask. I've used the same code and tried to change the raw format and the sensor bit depth as below:
``` python
raw_mode_parameter = {
        'use_case': 'still',
        'buffer_count': 4,
        'main': {
            'size': (2028, 1520), 
            'format': 'BGR888', 
            'stride': 6144, 
            'framesize': 9338880
        },
        'raw': {
            'size': (2028, 1520), 
            'format': 'SBGGR10', 
            'stride': 3072, 
            'framesize': 4669440
        },
        'sensor':{'bit_depth': 10}  
    }
```
but the result of bit depth is still 12, and the output is below:
```
[25:56:25.883890261] [13285]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+50-83cb8101
[25:56:25.892936575] [13321]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[25:56:25.903857464] [13321]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx477@1a to CFE device /dev/media3 and ISP device /dev/media0 using PiSP variant BCM2712_C0
[25:56:25.915661446] [13285]  INFO Camera camera.cpp:1202 configuring streams: (0) 2028x1520-BGR888 (1) 2028x1520-SBGGR16
[25:56:25.916489927] [13321]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected CFE format: 2028x1520-BYR2
final camera configuration {'use_case': 'still', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'sYCC'>, 'buffer_count': 4, 'queue': True, 'main': {'format': 'BGR888', 'size': (2028, 1520), 'preserve_ar': True, 'stride': 6144, 'framesize': 9338880}, 'lores': None, 'raw': {'format': 'SBGGR16', 'size': (2028, 1520), 'stride': 4096, 'framesize': 6225920}, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.HighQuality: 2>, 'FrameDurationLimits': (100, 1000000000)}, 'sensor': {'bit_depth': 12, 'output_size': (2028, 1520)}, 'display': None, 'encode': None}
```

davidplowman, 2025-02-07:
There is no 2028x1520 10-bit mode, so it has to choose whether to prioritise the resolution you want or the bit depth (and you get the resolution). The only way to be 100% sure what camera mode will be chosen is to choose one that you know exists. You can check this with `rpicam-hello --list-cameras` or in Python like this:
```
from picamera2 Import Picamera2

picam2 = Picamera2()
picam2.sensor_modes
```
(Do this before doing anything else because it will reconfigure the camera many times to find out more about each mode. But after you've done it once, `picam2.sensor_modes` is cached.)

HcosYu, 2025-02-07:
Thank you David, really help me a lot!

closed 2025-02-07 by HcosYu