raspberrypi/picamera2 #1207: [SUGGESTION]I would like to get frames from the ffmeg.

CrabbyPete, 2025-02-08:
I wrote this code to try and get each frame
`
class StreamingOutput(io.BufferedIOBase):
    def __init__(self):
        self.frame = None
        self.condition = Condition()

    def write(self, buf):
        with self.condition:
            self.frame = buf
            self.condition.notify_all()

def frame_generator():
 with Picamera2() as picam2:
     video_config = picam2.create_video_configuration()
     picam2.configure(video_config)
     encoder = H264Encoder(bitrate=10000000)
     output = StreamingOutput()
     picam2.start_recording(encoder, FfmpegOutput(output))
     while True:
          with output.condition:
                output.condition.wait()
                yield output.frame`

but if I run it I get this error:
File "/usr/lib/python3/dist-packages/picamera2/outputs/ffmpegoutput.py", line 80, in start
    audio_codec + video_codec + self.output_filename.split()

Is there a way to allow it to just return the frame?

davidplowman, 2025-02-10:
Hi, and thanks for the question. I'm not sure I totally understand what you want to do, nor why you're using an `FfmpegOutput`. Do you want your `frame_generator` to yield H.264 encoded frames? What happens if you replace the line with `FfmpegOutput` by
```
     picam2.start_recording(encoder, output)
```
?

CrabbyPete, 2025-02-10:
Thanks for responding, and I may not have a complete understanding of the format, but I wanted to stream ffmpeg frames. If I do what you suggest, then it streams H264 format. I read this https://github.com/raspberrypi/picamera2/issues/182, but I am not sure it solved the issue, since the timestamps are are separate.

davidplowman, 2025-02-10:
Hi again, can you say what do you mean by "ffmpeg frames"? Do you mean frames that aren't encoded at all?

closed 2025-02-20 by CrabbyPete