raspberrypi/picamera2 #1213: [HOW-TO] Focus setting in Raspberry Pi 5 using Arducam IMX519

Akshar062, 2025-02-19:
**Description:**

I am trying to set the focus on an Arducam IMX519 camera module using Picamera2 on a Raspberry Pi 5, but I encounter the following error when calling `set_controls`:

```
File "/home/mindrontm/Desktop/AppAPEX1.0.0_editing/camera_preview.py", line 188, in set_single_autofocus
    self.picam2.set_controls({"AfMode": 1, "AfTrigger": 0})  # 0 to trigger autofocus
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1242, in set_controls
    self.controls.set_controls(controls)
  File "/usr/lib/python3/dist-packages/picamera2/controls.py", line 58, in set_controls
    self.__setattr__(k, v)
  File "/usr/lib/python3/dist-packages/picamera2/controls.py", line 33, in __setattr__
    raise RuntimeError(f"Control {name} is not advertised by libcamera")
RuntimeError: Control AfMode is not advertised by libcamera
```

Additionally, I see the following QML error:

```
file:///home/mindrontm/Desktop/AppAPEX1.0.0_editing/qml/pages/HomeleftPage.qml:460: Error: Control AfMode is not advertised by libcamera
```

**Expected Behavior:**
The autofocus should work as it does on the Raspberry Pi Camera Module 3 using the following function:

```python
@Slot()
def set_single_autofocus(self):
    # Set the camera to single autofocus mode and trigger autofocus.
    self.picam2.set_controls({"AfMode": 1, "AfTrigger": 0})  # 0 to trigger autofocus
    self.picam2.wait_for_af()
```

**Observations & Questions:**
- This works fine on the Pi Camera Module 3, but not on the Arducam IMX519.
- Is autofocus supported on the IMX519 through libcamera?
- Do I need a different control parameter for autofocus?
- If autofocus is not supported via `AfMode`, is there an alternative method to adjust focus?

**System Information:**
- **Raspberry Pi Model:** Pi 5
- **Camera Module:** Arducam IMX519
- **Picamera2 Version:** 0.3.25
- **Libcamera Version:**
```
mindrontm@raspberrypi:~ $ libcamera-hello --version
rpicam-apps build: v1.6.0 9825703b669a-dirty 05-02-2025 (06:44:27)
rpicam-apps capabilites: egl:1 qt:1 drm:1 libav:1
libcamera build: v0.4.0+78-818d45eb-dirty (2025-02-14T02:03:07GMT)
```

Would appreciate any guidance on enabling autofocus or manually setting focus for this module. Thanks!

davidplowman, 2025-02-19:
Hi, our libcamera implementation doesn't know about IMX519 modules with an autofocus mechanism, which probably explains the error. It is up to the module vendor to upstream support for their module to us, but they haven't done so. So I don't think there's any way for us to help you - you will have to reach out to the module vendor (Arducam) for support.

Akshar062, 2025-02-20:
i have change module to new arducam imx 519 , turns out that old one was faulty and was not able to move focus motor , new one is working fine, sorry for trouble

closed 2025-02-20 by Akshar062