raspberrypi/picamera2 #1214: [HOW-TO]

pitosalas, 2025-02-21:
Config:
Raspi 4b
Ubuntu 22.04
Raspi Camera v2

I have updated config.txt with 
```
[all]
start_x=1
gpu_mem=128

```
My camera is detected:

```
[dome:~] sudo vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0
```

Now I want to actually use the camera and I am going around in circles. There are so many posts that give ideas that don't work. I am not trying to do something fancy! 

Maybe you can help me with:

```
[dome:~] sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-picamera2
```

pitosalas, 2025-02-21:
Hi, thanks for the detailed answer! I appreciate it. So I will talk to Ubuntu folks because I am building a robot and using ROS2 which is at this point most reliable on Ubuntu. 

One question: is the new stack picamera2 tested on Ubuntu? That will be helpful to know because what I should then do is fully uninstall it, reasoning that having it partially installed might be causing problems. Agree?

davidplowman, 2025-02-21:
This whole Ubuntu question does come up from time to time and I think they are trying to support the Pi cameras. However, there are some difficulties and it's taking some time, for example see [here](https://github.com/raspberrypi/picamera2/issues/563#issuecomment-2637097612).

That suggests that support for the underlying libcamera library _might_ be available soon-ish, but the numpy major version change has complicated things in the Python world. So it sounds like Picamera2 simply doesn't work at the moment, but at the risk of repeating myself, you need to check with them.

pitosalas, 2025-02-21:
Thanks. (p.s. I accidentally emailed a response before that was gibberish. Feel free to iognore/delete it if it makes it onto the thread)

closed 2025-02-21 by pitosalas