raspberrypi/picamera2 #1219: [BUG] Jpeg Output of rpicam-still and basic Picamera2 script are different on RaspberryPi5

Esser50K, 2025-02-27:
**Describe the bug**
I get a good frame using:

```rpicam-still -o test.jpg```

I get a somewhat broken looking frame using:

```
import time
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start()
time.sleep(2)

picam2.capture_file("test.jpg")
picam2.stop()
```

**To Reproduce**
To reproduce one can use the command/script above.

This works fine on a raspberrypi3 but not on a pi5. Both used with fresh flahes of latest PiOS

**Expected behaviour**
Get same (or similar) image using Picamera2

**Console Output, Screenshots**
rpicam output:

```
rpicam-still -o test.jpg
[0:01:55.160987122] [1895]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:01:55.168496407] [1901]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:01:55.185874863] [1901]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
Preview window unavailable
Mode selection for 1296:972:12:P
    SGBRG10_CSI2P,640x480/0 - Score: 3296
    SGBRG10_CSI2P,1296x972/0 - Score: 1000
    SGBRG10_CSI2P,1920x1080/0 - Score: 1349.67
    SGBRG10_CSI2P,2592x1944/0 - Score: 1567
Stream configuration adjusted
[0:01:55.187611435] [1895]  INFO Camera camera.cpp:1197 configuring streams: (0) 1296x972-YUV420 (1) 1296x972-GBRG_PISP_COMP1
[0:01:55.187742120] [1901]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 - Selected sensor format: 1296x972-SGBRG10_1X10 - Selected CFE format: 1296x972-PC1g
#0 (0.00 fps) exp 33222.00 ag 8.00 dg 1.00
#1 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#2 (30.01 fps) exp 33222.00 ag 8.00 dg 1.00
#3 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#4 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#5 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#6 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#7 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#8 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#9 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#10 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#11 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#12 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#13 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#14 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#15 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#16 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#17 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#18 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#19 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#20 (30.01 fps) exp 33222.00 ag 8.00 dg 1.00
#21 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#22 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#23 (30.01 fps) exp 33222.00 ag 8.00 dg 1.00
#24 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#25 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
#26 (30.02 fps) exp 33222.00 ag 8.00 dg 1.00
[... 122 lines cut ...]
Still capture image received
```

python output:
```
Python 3.11.2 (main, Sep 14 2024, 03:00:30) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import time
>>> from picamera2 import Picamera2
>>> picam2 = Picamera2()
[0:08:36.979746960] [1924]  INFO Camera camera_manager.cpp:325 libcamera v0.3.2+27-7330f29b
[0:08:36.987163420] [1930]  INFO RPI pisp.cpp:695 libpisp version v1.0.7 28196ed6edcf 29-08-2024 (16:33:32)
[0:08:36.996568934] [1930]  INFO RPI pisp.cpp:1154 Registered camera /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 to CFE device /dev/media2 and ISP device /dev/media0 using PiSP variant BCM2712_C0
>>> picam2.start()
[0:08:36.999602044] [1924]  INFO Camera camera.cpp:1197 configuring streams: (0) 640x480-XBGR8888 (1) 640x480-GBRG_PISP_COMP1
[0:08:36.999719044] [1930]  INFO RPI pisp.cpp:1450 Sensor: /base/axi/pcie@120000/rp1/i2c@88000/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected CFE format: 640x480-PC1g
>>> time.sleep(2)
>>>
>>> picam2.capture_file("test.jpg")
{'SensorTimestamp': 519024563000, 'ExposureTime': 66638, 'FocusFoM': 112, 'AnalogueGain': 5.0625, 'AeLocked': True, 'ColourCorrectionMatrix': (1.9827418327331543, -0.45585930347442627, -0.5268846154212952, -0.5524343252182007, 2.0013644695281982, -0.4489300847053528, -0.11916769295930862, -0.7477512955665588, 1.8669110536575317), 'SensorBlackLevels': (1024, 1024, 1024, 1024), 'FrameDuration': 66773, 'Lux': 64.08871459960938, 'ColourTemperature': 3838, 'DigitalGain': 1.0008397102355957, 'ColourGains': (1.1946176290512085, 1.5613949298858643), 'ScalerCrops': [(16, 0, 164, 128)], 'ScalerCrop': (16, 0, 164, 128)}
>>> picam2.stop()
```

**Hardware :**
Issue is on raspberrypi5, also tested on raspberrypi3 on which it works fine.

Esser50K, 2025-02-27:
Ok it seems like one has to set the controls to capture the whole frame:
```
picam2.set_controls({
    "AeEnable": True,         # Enable Auto Exposure (ensures proper brightness)
    "AwbEnable": True,        # Enable Auto White Balance (ensures proper colors)
    "NoiseReductionMode": 3,  # High-quality noise reduction (matches ISP processing)
    "Sharpness": 1.0,         # Match rpicam-still sharpness
    "Contrast": 1.0,          # Prevent washed-out image
    "Saturation": 1.2,        # Slight color boost (adjust if needed)
    "ScalerCrop": (0, 0, 2592, 1944),  # Ensure full frame is used
    "ColourGains": (1.5, 1.5) # Try matching CFE color correction
})
```

it's weird because one has to do it after `picam2.start()` and one has to sleep for a bit for this to take effect before working

davidplowman, 2025-02-27:
Hi, I think some of this may be down to the selection of the sensor mode. `rpicam-still` starts in a full field-of-view (FoV) preview mode, and then switches automatically to the full resolution (also full FoV) mode for the capture.

In the Python world you have full control over the sensor, so you may have to give it some hints as to what you want. For example, sometimes people prefer reduced FoV modes because they often run at higher framerates, sometimes they prefer the full FoV.

There quite a lot about mode selection in section 4.2.23 of [the manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

A fairly typically example of how you'd do a full resolution capture can be found [here](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_full_res.py). In this example, a "still" configuration defaults to the full resolution, full FoV camera mode. Though note that there's nothing that forces the preview to be full FoV - best practice would probably be to specify the preview sensor mode to be sure you get what you want.

When setting controls, you're right that you have to wait "a few frames" for them to take effect if the camera is already running. If you set them just before calling `start()`, then they should take effect immediately.

Esser50K, 2025-02-27:
Cheers thanks for the reply :)

closed 2025-02-27 by Esser50K