raspberrypi/picamera2 #1222: [HOW-TO] Adapt code to use camera module rather than USB webcam

Tess314, 2025-03-01:
I am trying to alter my code to use the Raspberry Pi Camera Module 3 instead of a USB webcam. I know I can use picamera2 to do this, but cannot figure out how.

Here is my code: https://github.com/Tess314/Healthcare_AI-Support_System/blob/main/classify.py

Any help or advice would be appreciated.

davidplowman, 2025-03-03:
Hi, and thanks for the question. There are some OpenCV examples such as [this](https://github.com/raspberrypi/picamera2/blob/main/examples/opencv_face_detect.py), but the basic code that you need for a Camera Module 3 is:
```
import cv2
from picamera2 import Picamera2

cv2.startWindowThread()

picam2 = Picamera2()
main = {'format': 'RGB888', 'size': (640, 480)}  # "main" image output stream format and size
sensor = {'output_size': (2304, 1296)}  # use camera module 3 full field of view video mode
controls = {'FrameRate': 30, 'LensPosition': 1.0}
config = picam2.create_preview_configuration(main, sensor=sensor, controls=controls)
picam2.configure(config)
picam2.start()

while True:
    array = picam2.capture_array()
    cv2.imshow("Cam", array)
    cv2.waitKey(1)
```
A few things to point out:

* The format 'RGB888' actually gives you BGR pixels like OpenCV wants (don't ask...)
* You can request pretty much any output resolution you like.
* Here I've asked for the sensor's 2304 x 1296 readout mode. This is a good choice if you want the full field of view, reasonably high framerates, but don't need the full 12MP output.
* I've set the camera to a fixed 30fps, but higher or lower values can be achieved if you prefer.
* I've set the lens to a fixed 1m focal distance (the units are dioptres, or 1 / distance-in-metres). Obviously your application may require something different, or you might want to enable the continuous autofocus mechanism.

Please ask if you need to know more!

Tess314, 2025-03-06:
Thanks for that @davidplowman! Tested that example on my Raspberry Pi and it works great. I'm just not sure where this should go in my own code though, as currently the camera setup is in the middle, so doing:

```
while True:
    (all my code)
```

doesn't work.

davidplowman, 2025-03-06:
Hi again, assuming you're referring to from your link above, the `while True:` in my example probably corresponds to `while cap.isOpened():` in your `run` function. So the camera configuration and start stuff goes above there, and `success, image = cap.read()` is replaced by `image = picam2.capture_array()`. Does that help?

Tess314, 2025-03-06:
That worked great, thank you so much!

closed 2025-03-06 by Tess314