raspberrypi/picamera2 #1224: [HOW-TO] How can I capture a .dng file similar to the one generated by libcamera?

xuwimming, 2025-03-09:
I have tried using the capture_dng_and_jpeg.py and capture_dng_and_jpeg_helpers.py scripts, but I encountered an issue. The .dng files generated by these scripts are colorful, while the .dng files produced by libcamera-still --raw are grayscale images. Additionally, when using a for loop in the Python scripts to continuously capture 64 images, some of the images have darker top parts than others, and some are even completely dark. However, when I use the libcamera-still --raw command in a loop, I don’t encounter this issue.

xuwimming, 2025-03-10:
Hello davidplowman,

Thank you for the information. The issue with the dark section was resolved by adding a time.sleep(1) between each capture. I’m unsure of the exact cause of the dark top part, as it occurred randomly on both JPG and DNG files. I typically capture both JPG and DNG files simultaneously. The dark top part is determined by the specific pixel values in the image when viewed in VS Code. It is not a small region but affects the entire top rows. For example, if I have a 100 x 100 white image array, the 'dark top part' means that the top N rows have significantly lower values than the rest of the image. In my opinion, this could be due to insufficient illumination power and the capture speed being too fast, preventing the camera from acquiring enough photons.

Yes, the thumbnails are different. When I load the images into VS Code, the data types of the DNG files from both the rpicam-still and PiCamera2 are the same. However, the DNG files from PiCamera2 always have slightly lower values compared to those from rpicam-still under the exact same condition, the file size is also smaller, 15.4 MB vs 15.5 MB. Additionally, during post-processing with my algorithm, the time taken to process DNG files from PiCamera2 is significantly slower than for DNG files from rpicam-still, even the "dark top part" issue was resolved.

I am using a Raspberry Pi 5 with the Pi Camera Module 2.

davidplowman, 2025-03-11:
Hi again and thanks for the update. Quite a lot there to unpack, so perhaps I can try and answer inline.

> Thank you for the information. The issue with the dark section was resolved by adding a time.sleep(1) between each capture. I’m unsure of the exact cause of the dark top part, as it occurred [...]

I still find this quite puzzling. It's true that a short sleep is often necessary to let 3A algorithms settle fully, but this should not affect just a part of the image. It should always affect the whole image. The camera module 2 has quite strong vignetting problems, so maybe the time for the vignetting correction to "settle" could be a factor, but I'm not convinced - and you'd probably see an effect all round the edge of the image. So indeed, this remains a bit mysterious...

> Yes, the thumbnails are different. When I load the images into VS Code, the data types of the DNG files from both the rpicam-still and PiCamera2 are the same. However, the DNG files from PiCamera2 [...]

It's not obvious why you should see a difference in brightness. Have you compared the exposure time and ISO reported in the DNG file (e.g. with something like `exiftool`). If these are slightly different, then that would explain the slightly different values. Of course, there's still a question as to why they might be different. It can be that, if you're not careful, the camera mode used while letting the AEC/AGC settle can be different between Picamera2 and rpicam-still, and this could definitely cause a slight difference. rpicam-still will probably be using the 2x2 binned 1640x1232 mode, and you can force Picamera2 to use that as well.

Is the file size difference merely caused by the presence of a thumbnail in one of them?

Finally, I can't explain why your algorithm would take longer with one of the DNG files than the other. Once you've fetched the pixel values out of the DNG file, how can there be any difference? Sounds like maybe you need to find out what can cause your algorithm to take longer - sorry not to have any more suggestions on that!

> 
> I am using a Raspberry Pi 5 with the Pi Camera Module 2.

xuwimming, 2025-03-16:
Hello davidplowman,

Thank you again for your help. I’ve tried a few different settings, and at times the post-processing speed is similar, while at other times it’s not. I agree with you that I first need to identify the factors causing these differences.

closed 2025-03-16 by xuwimming