raspberrypi/picamera2 #1226: [HOW-TO] Use v4l2-ctl to change settings of running camera

ABelliqueux, 2025-03-14:
Hi, 

I've followed the tutorial here to use a Rpi Zero 2 W and a camera module V3 as a webcam /uvc_gadget :  

https://www.raspberrypi.com/news/how-to-build-your-own-raspberry-pi-webcam/

I was then able to enable serial console access and want to use `v4l2-ctl` to change the settings of the live feed.

The [picamera2 manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)  states on p.70 that I should be able to issue this command to alter a setting:  

```
v4l2-ctl --set-ctrl wide_dynamic_range=1 -d /dev/v4l-subdev0
```
but when trying to alter a setting in such a way, I get an error message stating :  

```
VIDIOC_S_EXT_CTRLS: failed: Device or resource busy
Error setting controls: Device or resource busy
```

likewise, trying to alter any of the advertised controls on `/dev/v4l-subdev0` results in the same error, e.g:

```
 v4l2-ctl --set-ctrl horizontal_flip=1 -d /dev/v4l-subdev0
VIDIOC_S_EXT_CTRLS: failed: Device or resource busy
Error setting controls: Device or resource busy
```

Is altering picture settings with the camera running not possible ? I know that it was possible with the old stack + camera v2 because I can to this with [showmewebcam](https://github.com/showmewebcam/showmewebcam) for example.

Here is a list of the controls that are advertised as available :   

```
$ v4l2-ctl -d /dev/v4l-subdev0 -l

User Controls

                       exposure 0x00980911 (int)    : min=1 max=2446 step=1 default=2002 value=1817
                horizontal_flip 0x00980914 (bool)   : default=0 value=1 flags=grabbed, modify-layout
                  vertical_flip 0x00980915 (bool)   : default=0 value=1 flags=grabbed, modify-layout

Camera Controls

             wide_dynamic_range 0x009a0915 (bool)   : default=0 value=0 flags=grabbed, modify-layout
             camera_orientation 0x009a0922 (menu)   : min=0 max=2 default=2 value=2 flags=read-only
         camera_sensor_rotation 0x009a0923 (int)    : min=180 max=180 step=1 default=180 value=180 flags=read-only

Image Source Controls

              vertical_blanking 0x009e0901 (int)    : min=40 max=8387184 step=1 default=1198 value=1198
            horizontal_blanking 0x009e0902 (int)    : min=5520 max=5520 step=1 default=5520 value=5520 flags=read-only
                  analogue_gain 0x009e0903 (int)    : min=112 max=960 step=1 default=112 value=112
                red_pixel_value 0x009e0904 (int)    : min=0 max=4095 step=1 default=4095 value=4095
          green_red_pixel_value 0x009e0905 (int)    : min=0 max=4095 step=1 default=4095 value=4095
               blue_pixel_value 0x009e0906 (int)    : min=0 max=4095 step=1 default=4095 value=4095
         green_blue_pixel_value 0x009e0907 (int)    : min=0 max=4095 step=1 default=4095 value=4095
error 28 getting ext_ctrl Notify Gains

Image Processing Controls

                 link_frequency 0x009f0901 (intmenu): min=0 max=0 default=0 value=0 flags=read-only
                     pixel_rate 0x009f0902 (int64)  : min=585600000 max=585600000 step=1 default=585600000 value=585600000 flags=read-y
                   test_pattern 0x009f0903 (menu)   : min=0 max=4 default=0 value=0
                   digital_gain 0x009f0905 (int)    : min=256 max=65535 step=1 default=256 value=256

```

# Versions

OS: Rpi OS legacy (bullseye) 64bits

```
$ dpkg -l | grep camera
ii  libcamera-apps-lite                  1.2.1-1                  
ii  libcamera-dev:arm64                  0~git20230720+bde9b04f-1 
ii  libcamera0:arm64                     0~git20230720+bde9b04f-1 

dpkg -l | grep v4l
ii  libv4l-0:arm64                       1.20.0-2                 
ii  libv4l2rds0:arm64                    1.20.0-2                 
ii  libv4lconvert0:arm64                 1.20.0-2                 
ii  v4l-utils                            1.20.0-2                 
```

davidplowman, 2025-03-14:
Hi, this isn't something I know a great deal about and probably doesn't really relate to Picamera2, but it might be something to ask on the [Camera Forum](https://forums.raspberrypi.com/viewforum.php?f=43) where you'll get a wider range of people who are able to reply.

From my knowledge of libcamera, it tends to take over complete control of the camera, so other users are locked out. I'm guessing that this means that the only way to change camera settings is for the program driving the camera (uvc_gadget?) to implement some form of communication that lets you do this. But as I said, it's not something I know much about. Sorry not to be more help, but do try the Camera Forum. Thanks!

ABelliqueux, 2025-03-15:
Hi, Thanks for your reply.
I've created a discussion as per your suggestion : https://forums.raspberrypi.com/viewtopic.php?t=385415
I'll update this issue with details if I get some new insights.