raspberrypi/picamera2 #1229: [BUG] Native Zoom on Camera

Rob-Nafous, 2025-03-18:
**Please only report one bug per issue!**

**Describe the bug**
The camera is super zoomed

**To Reproduce**
Hello, I wrote down a code in order to do facial_recognition with my raspberry pi 5. I use a picamera module 3.
I followed the tutorial a first time. And everything went well and perfect.

However, after reinitializating my raspberry pi. Setting my conda environment, downloading all my librairies. My code wasn't working anymore.
So I went back to my test.py as follow : 

```
from picamera2 import Picamera2, Preview
import time
picam2 = Picamera2()
camera_config = picam2.create_preview_configuration()
picam2.configure(camera_config)
picam2.start_preview(Preview.QTGL)
picam2.start()
time.sleep(2)
picam2.capture_file("test.jpg")
```

and saw that the camera was SUPER zoomed. Like x30. I don't understand why. Because here's the step : 

After reinitializating my pi : 

Installing conda with :
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-armv7l.sh
chmod +x Miniconda3-latest-Linux-armv7l.sh
./Miniconda3-latest-Linux-armv7l.sh

checking if the camera is working with : 
rpicam-hello
**All works well the camera is good**

Downloading picamera2 with : 
sudo apt install -y python3-picamera2

cd Desktop
python test.py 

and see that the camera is super zoomed. 

**Expected behaviour**
To not have a camera x30. 

**Console Output, Screenshots**
I can provide screenshot of whatever you want just ask me. I beg you help me please I'm turning crazy haha. 

**Hardware :**
Raspberry pi 5. Camera Module 3. 

**Additional context**
No additionnal context. But I could give you more if you need it. 
Thanks a lot for your time and help.

davidplowman, 2025-03-20:
Hi, can you just confirm you Picamera2 version (`apt list python3-picamera2`)? I know there was a crazy zoom-related bug a little while back.

I also don't know anything much about anaconda, but could that somehow mean you're not using the version of Picamera2 that you expect? I've tried your original script with a vanilla installation (i.e. no conda, no virtual environments), and it seems to work OK for me.

Rob-Nafous, 2025-03-20:
Hello david, 
```python3-picamera2/stable,stable,now 0.3.25-1 all [installed]```

I followed the instruction in first place after creating my conda environment. I'm not an expert either so couldn't help more I am sorry.

davidplowman, 2025-03-20:
Do you have a spare SD card lying around where you could install a clean version of Raspberry Pi OS. Don't add anything else to it (except for `sudo apt update` and `sudo apt full-upgrade`), so no conda environments, and then just try if the zoom problem is present or not?

Rob-Nafous, 2025-03-20:
I might have one at home, I'll try it tomorrow and let you know how it works :)

Rob-Nafous, 2025-03-21:
Hi David,

So yeah, it works with an empty SD. I don't understand how a conda environment can change the disposition ?

davidplowman, 2025-03-21:
Not sure either, I'm afraid! I'm wondering whether there's some kind of issue where the Picamera2 version gets coped/frozen when a conda environment is set up and then doesn't update (even though the system version does). Is that possible? Generally when you import a module you can look at `module.__file__`  which will give you a clue as to where it's being loaded from. Does that suggest anything?