raspberrypi/picamera2 #1230: [HOW-TO] Find minimal allowed bitrate for H264Encoder

nzottmann, 2025-03-18:
The H264Encoder does not work when the `bitrate` parameter is too low. This results in picamera2 not outputting any more frames, probably because all buffers are full.

I discovered that a bitrate is not too low in general, it depends on the dynamics of the picture (intra- and inter-frame). Thus having a still scene in front of the camera with a given bitrate working fine can stop working just by moving something in front of the camera.

If no bitrate is specified, picamera2 calculates a bitrate based on resolution, framerate and quality. If no quality is specified, the default is `Quality.MEDIUM`. This calculation unfortunately does not always give a "save" bitrate, especially for low resolutions, as I discovered in #1228 

- Why does the encoder just stop working if the bitrate is too low for the given picture complexity? Is this hardware-defined behavior?
- How can I determine the minimal required bitrate for any picture complexity?

davidplowman, 2025-03-21:
You could have a look at this PR https://github.com/raspberrypi/picamera2/pull/1236 and maybe try that branch. I believe it will fix the lock-ups and create valid mp4 files.

Of course you'll end up with pauses when the encoder runs out of bits, and I'm afraid I don't have a way to calculate at what bitrate (for given resolution/framerate) that is likely to start become an issue. It might be a thing you would have to experiment a bit with for your own use case. But do let me know if that branch improves things for you.

nzottmann, 2025-03-21:
> You could have a look at this PR [#1236](https://github.com/raspberrypi/picamera2/pull/1236) and maybe try that branch. I believe it will fix the lock-ups and create valid mp4 files.
>
> Of course [...]

Thank you, I will test and report.

I am absolutely fine with having to find the bitrate at which the encoder starts to skip frames by experiment if I continues to work. I was just worried about finding this limit by experiment if hitting it would cause the encoder to stop.

nzottmann, 2025-03-21:
#1236 fixes this bug, see [comment](https://github.com/raspberrypi/picamera2/pull/1236#issuecomment-2743754789) there