raspberrypi/picamera2 #1233: [HOW-TO] Utilize camera on-board downsampling instead of crop resolution for lower resolution modes

zkelly1, 2025-03-20:
**Describe what it is that you want to accomplish**
In my camera (Sony IMX219) [documentation](https://www.opensourceinstruments.com/Electronics/Data/IMX219PQ.pdf)'s documentation, on page 53, it describes a pixel binning mode to capture at a lower resolution but preserve the camera's FOV. Is there currently any way to do this with picamera2? By default, it seems to use a cropping method that takes the centermost 480x640 pixels, for instance.

davidplowman, 2025-03-21:
Yes there is. If you haven't done so already, please check out section 4.2.2.3 (Raw Streams and the Sensor Configuration) in [the manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

But the short summary for the IMX219 is that you probably want to add the parameter
```
    sensor={'output_size': (1640, 1232)}
```
when you create your camera configuration.

zkelly1, 2025-03-21:
Thanks for the fast reply! Sorry, should have also mentioned this. A requirement of my project is that the camera operate in the 206.65FPS mode, which has the output size 480x640. If I am correct in my understanding, by passing this output size key to the sensor dict, would it not select the sensor mode corresponding to (1640, 1232) @ ~41 FPS? Reading the section you linked me (thanks so much for that), I think what I would want here is to utilize the `crop_limits` parameter to double my FOV to `(960, 1280)`. Does that sound right to you? And if so, how should I go about that?

davidplowman, 2025-03-21:
You only have a number of fixed output modes from the camera. After that, you can crop from that image that you received (and resize it to match your output dimensions, for example to implement digital zoom), but that's all.

It might be worth typing `rpicam-hello --list-cameras` at the command line. This will list all the available modes, the maximum framerate for each, and also how they crop from the full sensor. You can't change any of that, you simply have to chose one of those modes and that will determine the maximum framerate and field of view.

zkelly1, 2025-03-21:
Ah, I see. I will have to make some compromise then. Thanks for your help!

closed 2025-03-21 by zkelly1