raspberrypi/picamera2 #1235: [BUG] mp4 videos variable bit rate, unreadable by some windows software

Gordon999, 2025-03-21:
**Please only report one bug per issue!**

**Describe the bug**
Using the attached code, on a pi5, imx500 camera, the resultant video is reported variable bit rate by kdenlive and unreadable by Vegas Pro on a Windows pc.

**To Reproduce**

#!/usr/bin/env python3

"""Based on imx500_object_detection_demo.py."""

import argparse
import sys
from functools import lru_cache
import cv2
import numpy as np
import time
import datetime
from picamera2 import MappedArray, Picamera2, Preview
from picamera2.devices import IMX500
from picamera2.devices.imx500 import (NetworkIntrinsics,postprocess_nanodet_detection)
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput
from libcamera import controls

# detection objects
objects      = ["cat","bear","clock","person"]
threshold    = 0.5   # set detection threshold   

# video settings
v_width      = 2028  # video width
v_height     = 1520  # video height
v_length     = 5     # seconds
show_detects = 0     # show detections on video

# initialise
last_detections = []
label    = " ( "
encoding = False

class Detection:
    def __init__(self, coords, category, conf, metadata):
        """Create a Detection object, recording the bounding box, category and confidence."""
        self.category = category
        self.conf = conf
        self.box = imx500.convert_inference_coords(coords, metadata, picam2)
    
def parse_detections(metadata: dict):
    """Parse the output tensor into a number of detected objects, scaled to the ISP output."""
    global last_detections
    bbox_normalization = intrinsics.bbox_normalization
    bbox_order = intrinsics.bbox_order
    threshold = args.threshold
    iou = args.iou
    max_detections = args.max_detections

    np_outputs = imx500.get_outputs(metadata, add_batch=True)
    input_w, input_h = imx500.get_input_size()
    if np_outputs is None:
        return last_detections
    if intrinsics.postprocess == "nanodet":
        boxes, scores, classes = \
            postprocess_nanodet_detection(outputs=np_outputs[0], conf=threshold, iou_thres=iou,
                                          max_out_dets=max_detections)[0]
        from picamera2.devices.imx500.postprocess import scale_boxes
        boxes = scale_boxes(boxes, 1, 1, input_h, input_w, False, False)
    else:
        boxes, scores, classes = np_outputs[0][0], np_outputs[1][0], np_outputs[2][0]
        if bbox_normalization:
            boxes = boxes / input_h
[... 152 lines cut ...]
                now = datetime.datetime.now()
                timestamp = now.strftime("%y%m%d_%H%M%S")
                print("New  Detection",timestamp,label)
                circular.open_output(PyavOutput("/run/shm/" + timestamp +".mp4"))
        # stop recording after v_length + 5 seconds to empty circular buffer
        if encoding and (time.monotonic() - startrec > v_length + 5):
            now = datetime.datetime.now()
            timestamp2 = now.strftime("%y%m%d_%H%M%S")
            print("Stopped Record", timestamp2)
            circular.close_output()
            encoding = False
        category = ""
        value = 0.0
        label = " ( "
**Expected behaviour**
Usable mp4 videos

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Pi5, imx500 camera

**Additional context**
Add any other context about the problem here.

Gordon999, 2025-03-21:
[imx500.txt](https://github.com/user-attachments/files/19389247/imx500.txt)

davidplowman, 2025-03-21:
Based on the previous discussion in the forum, should that be variable bit rate or variable framerate in the title?

Gordon999, 2025-03-24:
OK. I understand your point of view, but a bit disappointing if you can't write mp4s without use of other software.

 When l convert h264 to MP4 with ffmpeg l get a warning "Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly" which l haven't resolved yet.

nzottmann, 2025-03-25:
You can try pyav to write .mp4 directly from python, it has advantages with handling timestamps:
https://github.com/raspberrypi/picamera2/blob/fb031cb49cb2af4dab7439be7a90784c9d3db1ab/picamera2/outputs/pyavoutput.py#L15-L18
Example: https://github.com/raspberrypi/picamera2/blob/main/examples/pyav_capture.py

nzottmann, 2025-03-25:
Sorry, thought you were using ffmpeg explicit at some point.

So the PyavOutput is giving you the

> Timestamps are unset in a packet for stream 0. This is deprecated and will stop working in the future. Fix your code to set the timestamps properly

warning?
I got this warning when using an ffmpeg output, but not with pyav.

Gordon999, 2025-03-25:
Sorry for the confusion. In my original post l was using pyav. It was suggested l use ffmpeg to fix it but l pointed out l get this warning with ffmpeg when converting h264 to MP4. You responded toe 2nd part.

closed 2025-11-03 by Gordon999