raspberrypi/picamera2 #1240: [BUG] CircularOutput2 recording buffer separately.

reflexcameras, 2025-03-26:
**Describe the bug**
I am trying to implement CircularOutput 2 in some example code. It seems to be recording the buffer differently from how it is described in the docs

**To Reproduce**
I am running the following python script
```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

# 1) Configure the camera
picam2 = Picamera2()

config = picam2.create_video_configuration({
    "size": (1280, 720),
    "format": "YUV420"
})
picam2.configure(config)

# 2) Create an H.264 encoder at 10Mbps
encoder = H264Encoder(bitrate=10000000)

# 3) Create a CircularOutput2 with a 5s buffer
circ = CircularOutput2(buffer_duration_ms=5000)

# 4) Attach CircularOutput2 as the encoder’s output
encoder.output = circ

# 5) Start the camera, then start the encoder
# picam2.start()                   # sensor streaming
# picam2.start_encoder(encoder)    # frames encoded into circ’s buffer
picam2.start_recording(encoder, circ)

print("Camera + encoder started, capturing into a 5s circular buffer... wave hand")
time.sleep(5)

# 6) Open an MP4 file (PyavOutput) to store what’s in the buffer plus 5s of new frames
print("Opening output start.mp4 (this will contain the first 5s from the buffer!)")
mp4file = "/mnt/sdcard/start.mp4"
circ.open_output(PyavOutput(mp4file))

print("writing ~5s more of data - wave glasses case")
time.sleep(5)

# 7) Close start file
print("Closing start.mp4")
circ.close_output()

# 8) Open another MP4 file
print("Opening output end.mp4 (will capture final content until we stop).")
mp4file = "/mnt/sdcard/end.mp4"
circ.open_output(PyavOutput(mp4file))
print("wave banana")
time.sleep(5)

# 9) Stop everything and flush
print("Stopping encoder + camera, finalizing end.mp4.")
# picam2.stop_encoder()
# picam2.stop()
picam2.stop_recording()

print("Done! Check start.mp4 and end.mp4 for valid recordings.")
```

**Expected behaviour**
I expect the hand waving and waved glasses case to be in start.mp4. I expect the waved banana to be in end.mp4 
However I am seeing the hand waving in start.mp4. And the waved glasses case and banana are both saved in end.mp4.

**Console Output, Screenshots**
N/a

**Hardware :**
Compute Module 4 and camera module 3

**Additional context**
I was previously using start_encoder and start but have commented this out to further match the docs and this isn't causing the way the files are split in recording

davidplowman, 2025-03-27:
Hi, thanks for the report and the sample code.

First of all, I just wanted to make sure I was understanding the code properly. Where you "wave <something>" and then sleep for 5s, that means you're waving the <something> during those 5 seconds?

As far as I recall, the CircularOutput2 means you're always recording stuff from 5 seconds (or however large the buffer is) in the past. So you'd have to have a sleep of at least 5s _after_ you've been waving at it. So if, when you wave, but there's only 3 seconds of the sleep to go before you close the output, then you won't include it.

Does that agree with what you found, or is something else happening? Thanks!

reflexcameras, 2025-03-27:
Hi David,

Yes I am waving for 5 seconds at step 5 in the code I assumed this would be populating the buffer with me waving as I haven't started the circ.open_output yet as the code is sleeping after the camera has initiated.

In step 6 I then open the file writing with mp4file = "/mnt/sdcard/start.mp4" circ.open_output(PyavOutput(mp4file)) and wave a glasses case for 5 seconds during a second sleep.

In step 7 I then close the output with circ.close_output()

In steps 8 and 9 I then open and close another output in end.mp4.

I have set a 5 second buffer at the beginning of the code so I would have thought that the waving (in the buffer) and waving of glasses case (after open_output and before close_output) would both be in the same video start.mp4. I would then think that the waving of the banana would be by itself in the end.mp4. However in start.mp4 I only have the waving of the hand (i.e. the buffer before the open_output was called) and not the waving of the glasses case.

I hope that makes some sense,

Thanks,
Sam

davidplowman, 2025-03-28:
Thanks for the explanation. However, I'm still not 100% clear whether, after you have waved the glasses case, you are waiting for a _further_ 5 seconds before closing the file, or whether, once you've waved the glasses case, there's now less than 5 seconds before the file is closed. Sorry that I'm having trouble with this!

Would it help to edit the script so that it prints "Wave the thing now!" and then calls `input("Press ENTER once you've done that!")`, and then you can put the 5 second sleep. Maybe that would make it clearer?

reflexcameras, 2025-03-28:
Hi David, 

Sorry I'll clarify. I wave the hand during the whole 5 seconds during the first 5 second sleep before the circ.open_output call. After the circ.open_output call I then wave the glasses case for the whole five seconds whilst the second 5 second sleep occurs just and continue waving them until the circ.close_output() call.

I can implement the press enter timing but I am just switching waves as soon as I receive a print statement.

davidplowman, 2025-03-28:
It does sound to me a bit like you're waving during the 5 seconds that it's not yet recording to file. You do have to wait 5 seconds _after_ all the waving for it to be saved. It might be worth modifying it with some "press ENTER" prompts to see if we can make everything 100% explicit.

reflexcameras, 2025-03-29:
OK I have adjusted the code as suggested. It is now recording the hand wave and glasses case wave in the start.mp4 and the banana wave in the end.mp4.

Would it be possible to explain the buffer again in a different way. I just want to be 100% this is how you expect it to work 

1. When you call start_recording, the camera starts sending frames to the encoder, and the encoder immediately starts adding those encoded frames into the circular buffer (managed by CircularOutput2).
2. When you subsequently call circ.open_output(...), Picamera2 writes both the existing frames already in the circular buffer plus any new frames to that output until you close it or stop recording.

Sorry for all the questions I am trying to incorporate this method into my existing code and want to be 100% clear on how it works

```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

# 1) Configure the camera
picam2 = Picamera2()

config = picam2.create_video_configuration({
    "size": (1280, 720),
    "format": "YUV420"
})
picam2.configure(config)

# 2) Create an H.264 encoder at 10Mbps
encoder = H264Encoder(bitrate=10000000)

# 3) Create a CircularOutput2 with a 5s buffer
circ = CircularOutput2(buffer_duration_ms=5000)

# 4) Attach CircularOutput2 as the encoder’s output
encoder.output = circ

# 5) Start the camera and encoder recording
picam2.start_recording(encoder, circ)

print("Camera + encoder started, capturing into a 5s circular buffer... wave hand")
time.sleep(5)
input("Press ENTER once you've waved your hand for 5 seconds!")

# 6) Open an MP4 file (PyavOutput) to store what’s in the buffer plus 5s of new frames
print("Opening output start.mp4 (this will contain the first 5s from the buffer!)")
mp4file = "/mnt/sdcard/start.mp4"
circ.open_output(PyavOutput(mp4file))

print("writing ~5s more of data - wave glasses case")
time.sleep(5)
input("Press ENTER once you've waved the glasses case for 5 seconds!")

# 7) Close start file
print("Closing start.mp4")
circ.close_output()

# 8) Open another MP4 file
print("Opening output end.mp4 (will capture final content until we stop).")
mp4file = "/mnt/sdcard/end.mp4"
circ.open_output(PyavOutput(mp4file))

print("wave banana")
time.sleep(5)
input("Press ENTER once you've waved the banana for 5 seconds!")

# 9) Stop everything and flush
print("Stopping encoder + camera, finalizing end.mp4.")
picam2.stop_recording()

print("Done! Check start.mp4 and end.mp4 for valid recordings.")
```

reflexcameras, 2025-03-29:
Thankyou David. I have adjusted the code as suggested (see below). This code is now recording the hand wave and glasses case wave in the start.mp4 and the banana wave in the end.mp4.

Would it be possible to explain the buffer again in a different way. Does the description below match how you understand it?

When you call start_recording, the camera starts sending frames to the encoder, and the encoder immediately starts adding those encoded frames into the circular buffer (managed by CircularOutput2).
When you subsequently call circ.open_output(...), Picamera2 writes both the existing frames already in the circular buffer plus any new frames to that output until you close it or stop recording.
Sorry for all the questions I am trying to incorporate this method into my existing code and want to be 100% clear on how it works

```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

# 1) Configure the camera
picam2 = Picamera2()

config = picam2.create_video_configuration({
    "size": (1280, 720),
    "format": "YUV420"
})
picam2.configure(config)

# 2) Create an H.264 encoder at 10Mbps
encoder = H264Encoder(bitrate=10000000)

# 3) Create a CircularOutput2 with a 5s buffer
circ = CircularOutput2(buffer_duration_ms=5000)

# 4) Attach CircularOutput2 as the encoder’s output
encoder.output = circ

# 5) Start the camera and encoder recording
picam2.start_recording(encoder, circ)

print("Camera + encoder started, capturing into a 5s circular buffer... wave hand")
time.sleep(5)
input("Press ENTER once you've waved your hand for 5 seconds!")

# 6) Open an MP4 file (PyavOutput) to store what’s in the buffer plus 5s of new frames
print("Opening output start.mp4 (this will contain the first 5s from the buffer!)")
mp4file = "/mnt/sdcard/start.mp4"
circ.open_output(PyavOutput(mp4file))

print("writing ~5s more of data - wave glasses case")
time.sleep(5)
input("Press ENTER once you've waved the glasses case for 5 seconds!")

# 7) Close start file
print("Closing start.mp4")
circ.close_output()

# 8) Open another MP4 file
print("Opening output end.mp4 (will capture final content until we stop).")
mp4file = "/mnt/sdcard/end.mp4"
circ.open_output(PyavOutput(mp4file))

print("wave banana")
time.sleep(5)
input("Press ENTER once you've waved the banana for 5 seconds!")

# 9) Stop everything and flush
print("Stopping encoder + camera, finalizing end.mp4.")
picam2.stop_recording()

print("Done! Check start.mp4 and end.mp4 for valid recordings.")
```

reflexcameras, 2025-03-29:
Thankyou David. I have adjusted the code as suggested (I have included this version below... success!!,  this code is now recording the hand wave and glasses case wave in the start.mp4 and the banana wave in the end.mp4.

I am just trying to get my head around the CircularOutput2, does the description below sound right?

1. When you call start_recording, the camera starts sending frames to the encoder, and the encoder immediately starts adding those encoded frames into the circular buffer (managed by CircularOutput2).
2. When you subsequently call circ.open_output(...), Picamera2 writes both the existing frames already in the circular buffer plus any new frames to that output until you close it or stop recording.

Sorry for all the questions I am trying to incorporate this method into my existing code and want to be 100% clear on how it works.

```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

# 1) Configure the camera
picam2 = Picamera2()

config = picam2.create_video_configuration({
    "size": (1280, 720),
    "format": "YUV420"
})
picam2.configure(config)

# 2) Create an H.264 encoder at 10Mbps
encoder = H264Encoder(bitrate=10000000)

# 3) Create a CircularOutput2 with a 5s buffer
circ = CircularOutput2(buffer_duration_ms=5000)

# 4) Attach CircularOutput2 as the encoder’s output
encoder.output = circ

# 5) Start the camera and encoder recording
picam2.start_recording(encoder, circ)

print("Camera + encoder started, capturing into a 5s circular buffer... wave hand")
time.sleep(5)
input("Press ENTER once you've waved your hand for 5 seconds!")

# 6) Open an MP4 file (PyavOutput) to store what’s in the buffer plus 5s of new frames
print("Opening output start.mp4 (this will contain the first 5s from the buffer!)")
mp4file = "/mnt/sdcard/start.mp4"
circ.open_output(PyavOutput(mp4file))

print("writing ~5s more of data - wave glasses case")
time.sleep(5)
input("Press ENTER once you've waved the glasses case for 5 seconds!")

# 7) Close start file
print("Closing start.mp4")
circ.close_output()

# 8) Open another MP4 file
print("Opening output end.mp4 (will capture final content until we stop).")
mp4file = "/mnt/sdcard/end.mp4"
circ.open_output(PyavOutput(mp4file))

print("wave banana")
time.sleep(5)
input("Press ENTER once you've waved the banana for 5 seconds!")

# 9) Stop everything and flush
print("Stopping encoder + camera, finalizing end.mp4.")
picam2.stop_recording()

print("Done! Check start.mp4 and end.mp4 for valid recordings.")
```

reflexcameras, 2025-03-29:
Thankyou David. I have adjusted the code as suggested (I have included a new version below)... success!!,  this code is now recording the hand wave and glasses case wave in the start.mp4 and the banana wave in the end.mp4.

I am trying to wrap my head around the CircularOutput2 because I am getting some strange results incorporating it into my existing code. This is probably an issue with my code. Does the description below sound right?

1. When you call start_recording, the camera starts sending frames to the encoder, and the encoder immediately starts adding those encoded frames into the circular buffer (managed by CircularOutput2).
2. When you subsequently call circ.open_output(...), Picamera2 writes both the existing frames already in the circular buffer plus any new frames to that output until you close it or stop recording.

Sorry for all the questions

```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

# 1) Configure the camera
picam2 = Picamera2()

config = picam2.create_video_configuration({
    "size": (1280, 720),
    "format": "YUV420"
})
picam2.configure(config)

# 2) Create an H.264 encoder at 10Mbps
encoder = H264Encoder(bitrate=10000000)

# 3) Create a CircularOutput2 with a 5s buffer
circ = CircularOutput2(buffer_duration_ms=5000)

# 4) Attach CircularOutput2 as the encoder’s output
encoder.output = circ

# 5) Start the camera and encoder recording
picam2.start_recording(encoder, circ)

print("Camera + encoder started, capturing into a 5s circular buffer... wave hand")
time.sleep(5)
input("Press ENTER once you've waved your hand for 5 seconds!")

# 6) Open an MP4 file (PyavOutput) to store what’s in the buffer plus 5s of new frames
print("Opening output start.mp4 (this will contain the first 5s from the buffer!)")
mp4file = "/mnt/sdcard/start.mp4"
circ.open_output(PyavOutput(mp4file))

print("writing ~5s more of data - wave glasses case")
time.sleep(5)
input("Press ENTER once you've waved the glasses case for 5 seconds!")

# 7) Close start file
print("Closing start.mp4")
circ.close_output()

# 8) Open another MP4 file
print("Opening output end.mp4 (will capture final content until we stop).")
mp4file = "/mnt/sdcard/end.mp4"
circ.open_output(PyavOutput(mp4file))

print("wave banana")
time.sleep(5)
input("Press ENTER once you've waved the banana for 5 seconds!")

# 9) Stop everything and flush
print("Stopping encoder + camera, finalizing end.mp4.")
picam2.stop_recording()

print("Done! Check start.mp4 and end.mp4 for valid recordings.")
```

reflexcameras, 2025-03-31:
Thankyou that really clears things up for me. 

Yes, I'd be really happy to test some revised code 🙃

davidplowman, 2025-04-02:
I've had a bit of a fiddle with the code, you can find a branch with the revised version here: https://github.com/davidplowman/picamera2/tree/circular2-change.

Previously I was a bit bothered that there could be some unpredictable  behaviours, but I think this version should be better.

For reference, I'm pointing the camera at a stopwatch running on my phone and testing with the following:
```
import time
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

picam2 = Picamera2()
config = picam2.create_video_configuration()
picam2.configure(config)
encoder = H264Encoder(bitrate=5000000, iperiod=30)
circular = CircularOutput2(buffer_duration_ms=5000)

input("Start stopwatch and press ENTER")
picam2.start_recording(encoder, circular)

circular.open_output(PyavOutput("/dev/shm/test0.mp4"))
time.sleep(10)
circular.close_output()

circular.open_output(PyavOutput("/dev/shm/test1.mp4"))
time.sleep(5)
circular.close_output()

circular.open_output(PyavOutput("/dev/shm/test2.mp4"))
picam2.stop_recording()
```
At the end I check that the first 5 seconds (subject to the slight vagaries of things starting up and hitting ENTER) end up in `test0.mp4`, the next 5 in `test1.mp4`, and the final 5 in `test2.mp4`.

Note that there's always a small gap when it switches to a new file. That's because you can't start recording the new file until you run into an I frame, which I can't produce "on demand". You can mitigate this a bit by reducing the codec's `iperiod`, but not eliminate it completely.

nzottmann, 2025-04-02:
>  until you run into an I frame, which I can't produce "on demand". You can mitigate this a bit by reducing the codec's `iperiod`, but not eliminate it completely.

The h264 encoder has a v4l2 button control to output a keyframe on demand:
```
v4l2-ctl -d 11 -l

Codec Controls
[...]
                force_key_frame 0x009909e5 (button) : value=0 flags=write-only, execute-on-write
[...]
```

@davidplowman It would be possible to expose this feature through a method in `H264Encoder` or `V4L2Encoder`.

The code could look like that (based on V4L2Encoder):
```python
def force_keyframe(self):
        _controls = [(V4L2_CID_MPEG_VIDEO_FORCE_KEY_FRAME, 1)]
        controlarr = (v4l2_ext_control * len(_controls))()
        ext = v4l2_ext_controls()
        for i, tup in enumerate(_controls):
            idv, value = tup
            controlarr[i].id = idv
            controlarr[i].value = value
            controlarr[i].size = 0
        ext.controls = controlarr
        ext.count = len(_controls)
        ext.ctrl_class = V4L2_CTRL_CLASS_MPEG
        fcntl.ioctl(self.vd, VIDIOC_S_EXT_CTRLS, ext)
```

This actually works, I tried and verified the result using ffprobe. There comes an extra keyframe every time the method is called.

davidplowman, 2025-04-02:
Hi @nzottmann 

Yes that's a good point. Actually I was aware of this for V4L2 but I'm wanting to treat the two platforms the same. I expect `libav` has a similar feature, though PyAV often only exposes the things that "someone needed" at that precise moment, so it would be a thing to investigate.

The other option might be to allow you to open the new output whilst simultaneously continuing to write to the previous one until we get to an I frame, when it would switch. Some while back I wrote a prototype `SplittableOutput` for someone that did this, and one could do a similar thing here.

reflexcameras, 2025-04-03:
Thankyou, this is really useful. I had a play with the example code yesterday and it works for me.

The reason I was looking into circularoutput2 was so I didn't have to reencode a h264 or mjpeg file into mp4 and alter the framerate. It seems like the mp4 files that I am getting from circularoutput2/ pyavoutput are not working on QuickTime and iMovie and are jittery on VLC. Are there any settings I can change in the code to make the mp4 files compatible from the get go without the need for re-encoding?

Thanks again,
Sam

davidplowman, 2025-04-03:
I don't really see why the streams should be jittery at all. It might be worth extracting the timestamps using ffmpeg just to check that the deltas between them are steady.

The fact that these files aren't working in QuickTime is, IMHO, a QuickTime deficiency. It might be possible to bodge enough stuff to create a constant framerate (CFR) file, I'm not sure. I guess we'd have to invent the timestamps as we go, and probably set the timebase to 1/framerate. Or something. It would definitely be a case of caveat emptor, because it would presumably be prone to going horribly wrong if you had audio or were live-streaming the output (because we don't know the true framerate of the camera, plus there is the risk of dropped frames). Also the timestamps in consecutive segments probably wouldn't join up either, not sure if that matters. Maybe I'll sleep on it.

davidplowman, 2025-04-03:
@reflexcameras BTW, are you using audio, or is this purely a video stream? Would it matter if dropped frames are simply not flagged (so you'd get a slight jump forward by a frame's worth of time)?

reflexcameras, 2025-04-03:
Hi David,

Sorry, how do I install this modified branch of the repo? I've been installing picamera2 using pi-gen or apt but I assume I have to take another route for this.

Apologies, I was just running the code you send yesterday and not the updated repo!

Thanks again,
Sam

davidplowman, 2025-04-03:
You can install from a branch on a Github repo, something like
```
pip install git+https://github.com/davidplowman/picamera2.git@circular2-changes
```
might be right. Just remember that pip overrides the system package, so at some point you should uninstall that pip version or Picamera2 will never update ever again!

Alternatively you could just clone that branch, and point your `PYTHONPATH` environment variable at that folder.

reflexcameras, 2025-04-03:
Hi David,

I have tried the code below but unfortunately I am getting the same result, the videos play in VLC but not in QuickTime or iMovie.

```import time
from picamera2 import Picamera2
import picamera2
print(picamera2.__file__)
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput

picam2 = Picamera2()
config = picam2.create_video_configuration()
picam2.configure(config)

encoder = H264Encoder(bitrate=5000000, iperiod=30)
circular = CircularOutput2(buffer_duration_ms=5000)

input("Start stopwatch and press ENTER")
picam2.start_recording(encoder, circular)

# Force 30 fps in the file "test0.mp4"
circular.open_output(PyavOutput("/mnt/sdcard/test0.mp4", force_video_framerate=30))
time.sleep(10)
circular.close_output()

# Force 30 fps in the file "test1.mp4"
circular.open_output(PyavOutput("/mnt/sdcard/test1.mp4", force_video_framerate=30))
time.sleep(5)
circular.close_output()

# Force 30 fps in the file "test2.mp4"
circular.open_output(PyavOutput("/mnt/sdcard/test2.mp4", force_video_framerate=30))
picam2.stop_recording()```

davidplowman, 2025-04-04:
Hmm, that's disappointing. What does
```
ffprobe <filename.mp4> -show_frames |& grep pkt_duration_time | sort -u
```
report for those files? Also is
```
ffprobe <filename.mp4> -show_frames |& grep pts_time
```
showing completely even frame times?

I'm not sure what else you're supposed to do to get a "CFR" file. We're sure that's still what it's complaining about?

davidplowman, 2025-04-04:
OK, I think I've noticed something. Things are a bit different on Pi 4s (and earlier) where we need to advertise the constant framerate to PyAV. On Pi 5s this is already happening because we use PyAV to do the encoding as well. This is why the change seemed OK to me, because I spend most of my life using a Pi 5!!

Anyway, I've patched the code for the Pi 4 case, so try updating that branch again (`git pull --rebase` should do it) and then have another go.

reflexcameras, 2025-04-04:
Aah that might explain it I'm using cm4s, I'll try again and let you know how it goes,

Thanks again!

reflexcameras, 2025-04-04:
Hi David,

I updated the repo and re-ran the code but unfortunately I run into the same issue. The files would play in VLC but not Quicktime/ iMovie.

results from `ffprobe <filename.mp4> -show_frames |& grep pkt_duration_time | sort -u`
nothing
results from `ffprobe <filename.mp4> -show_frames |& grep pkt_duration_time | sort -u`
```pkt_dts_time=0.000000
pkt_dts_time=0.033333
pkt_dts_time=0.066667
pkt_dts_time=0.100000
pkt_dts_time=0.133333
pkt_dts_time=0.166667
pkt_dts_time=0.200000
pkt_dts_time=0.233333
pkt_dts_time=0.266667
pkt_dts_time=0.300000
pkt_dts_time=0.333333
pkt_dts_time=0.366667
pkt_dts_time=0.400000
pkt_dts_time=0.433333
pkt_dts_time=0.466667
pkt_dts_time=0.500000
pkt_dts_time=0.533333
pkt_dts_time=0.566667
pkt_dts_time=0.600000
pkt_dts_time=0.633333
pkt_dts_time=0.666667
pkt_dts_time=0.700000
pkt_dts_time=0.733333
pkt_dts_time=0.766667
pkt_dts_time=0.800000
pkt_dts_time=0.833333
pkt_dts_time=0.866667
pkt_dts_time=0.900000
pkt_dts_time=0.933333
pkt_dts_time=0.966667
pkt_dts_time=1.000000
pkt_dts_time=1.033333
pkt_dts_time=1.066667
pkt_dts_time=1.100000
pkt_dts_time=1.133333
pkt_dts_time=1.166667
pkt_dts_time=1.200000
pkt_dts_time=1.233333
pkt_dts_time=1.266667
pkt_dts_time=1.300000
pkt_dts_time=1.333333
pkt_dts_time=1.366667
pkt_dts_time=1.400000
pkt_dts_time=1.433333
pkt_dts_time=1.466667
pkt_dts_time=1.500000
pkt_dts_time=1.533333
pkt_dts_time=1.566667
pkt_dts_time=1.600000
pkt_dts_time=1.633333
pkt_dts_time=1.666667
pkt_dts_time=1.700000
pkt_dts_time=1.733333
pkt_dts_time=1.766667
pkt_dts_time=1.800000
pkt_dts_time=1.833333
pkt_dts_time=1.866667
pkt_dts_time=1.900000
pkt_dts_time=1.933333
pkt_dts_time=1.966667
pkt_dts_time=2.000000
pkt_dts_time=2.033333
pkt_dts_time=2.066667
[... 225 lines cut ...]
I'm no expert, relying on some AI here, but I've tested a lot of ffmpeg commands and it appears to be be a problem related to missing SPS/PPS in the headers of each frame. 

I can get the files to play by re-encoding the mp4 with headers by
```ffmpeg -i test0-3.mp4 \
       -c:v libx264 \
         -x264opts repeat-headers=1 \
       -video_track_timescale 90000 \
       -c:a copy \
       out.mp4
```
or I can use ffmpeg to take the file on a round trip by remuxing it into a ts (or mkv) file and then back into a mp4 file. This avoids re-encoding.
```# Step 1: MP4 -> TS (Annex B)
ffmpeg -i original.mp4 \
       -c:v copy -bsf:v h264_mp4toannexb \
       -c:a copy -f mpegts \
       intermediate.ts

# Step 2: TS -> MP4 (AVCC)
ffmpeg -i intermediate.ts \
       -c:v copy \
       -c:a copy \
       -movflags +faststart \
       fixed.mp4
```
Do you think adding headers to each frame is something that is possible in PyAvOutput?

davidplowman, 2025-04-04:
I was using `mediainfo` to tell me whether it was officially a constant or variable framerate file, but evidently that's not enough.

You could try adding the parameters `framerate=30, enable_sps_framerate=True` when you make the `H264Encoder`.

reflexcameras, 2025-04-05:
Thanks David, I've tried 
```encoder = H264Encoder(
    bitrate=5000000,
    iperiod=30,
    framerate=30,
    enable_sps_framerate=True,
)```
this morning and unfortunately that doesn't seem to help with the videos playing in QuickTime. I'll build the remux chain into my code if there isn't another option

davidplowman, 2025-04-07:
That's too bad. I'm kind of running out of ideas here - every method I know to confirm that these are constant framerate files says that they are. If I come across anything else, or if someone out there can tell me what is "wrong" with these files, then I'll happily return to this...

JooJooBee666, 2025-05-27:
What is the status of this?  I'm having trouble in general just trying to use this at all on the Pi 4b.  
`from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput2, PyavOutput
import time

# Initialize Picamera2
picam2 = Picamera2()
config = picam2.create_video_configuration()
picam2.configure(config)

circular_output = CircularOutput2(buffer_duration_ms=5000)
encoder = H264Encoder()

# Attach CircularOutput2 to the encoder
encoder.output = circular_output

# Start recording
picam2.start_recording(encoder, circular_output)

print("Recording video...")

# Let recording run for a few seconds
time.sleep(5)

pyav_output = PyavOutput("output.mp4")
circular_output.open_output(pyav_output)

time.sleep(5)  # Capture additional time
circular_output.close_output()

# Stop recording
picam2.stop_recording()

print("Video with audio saved as 'output.mp4'")`

But this fails like so:
`[6:26:19.854809628] [7306]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
[6:26:19.887983785] [7346]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[6:26:19.890557892] [7346]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media0
[6:26:19.890734854] [7346]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[6:26:19.952707131] [7306]  INFO Camera camera.cpp:1205 configuring streams: (0) 1280x720-XBGR8888 (1) 2028x1080-SBGGR12_CSI2P
[6:26:19.956304544] [7346]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
Recording video...
Exception in thread Thread-7 (thread_poll):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/encoders/v4l2_encoder.py", line 257, in thread_poll
    self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/encoders/encoder.py", line 326, in outputframe
    out.outputframe(frame, keyframe, timestamp, packet, audio)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/outputs/circularoutput2.py", line 102, in outputframe
    self._flush(timestamp, self._output)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/outputs/circularoutput2.py", line 92, in _flush
    output.outputframe(*front)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/outputs/pyavoutput.py", line 89, in outputframe
    packet.stream = orig_stream
    ^^^^^^^^^^^^^
TypeError: Argument 'stream' has incorrect type (expected av.stream.Stream, got NoneType)
Video with audio saved as 'output.mp4'`

The output file, is of course, garbage and only shows 1 frame of video.  I wish I could just use a Pi 5 here, but alas, I cannot;  they took away the hardware h264 encoding!  This should be fixed for the Pi 4 folks that don't want a downgraded GPU.  I'll do more digging myself but this is definitely better off in the hands of the guys who put all this together to begin with. 👍

davidplowman, 2025-05-28:
> What is the status of this for the Pi 4b?

Hi, I'm not aware of any specific differences between the Pi 4b and Pi 5 in terms of supported functionality or behaviour in this kind of area. Can you be more specific about the question or provide any test cases that could be investigated? Thanks!

closed 2025-05-28 by davidplowman