raspberrypi/picamera2 #1244: [BUG]red discolouring on right half of image

Rustimation, 2025-04-03:
**Describe the bug**
With the opd picamera software I could overcome an annoying pink-reddish tint in the center of the image by applying the parameter  camera.still_stats = 'true' 

With Picamera2 this problem reappeared, this time more on the right half of the image.

I am using the official IMX219 (V2) Camera, running o a headless Raspberry Pi

The commands are straight forward:
`    config = picam2.create_still_configuration()
    picam2.configure(config)
    picam2.start()    
    time.sleep(2)
    # Automatischen Weißabgleich korrekt setzen
    picam2.set_controls({"AwbMode": controls.AwbModeEnum.Auto})  # Alternativen: Auto, Daylight, Cloudy, Tungsten, Fluorescent
    
    # Farbtemperatur (optional) anpassen
    picam2.set_controls({"ColourTemperature": 5000})  # Werte von 2500 (warm) bis 8000 (kalt) testen
    
    # Bildhelligkeit anpassen
    picam2.set_controls({"Brightness": 0.1})  # make everything brighter

    # Erneut warten, damit sich die Änderungen auswirken
    time.sleep(5)
       
image_array = picam2.capture_array()
picam2.stop()`

The annoying effect is most dominant when sunlight comes from top left without hitting the lens directly.

**Expected behaviour**
Clear and natural colours throughout the image.

**Console Output, Screenshots**
see attached

**Hardware :**
Raspberry Pi B w/ IMX219 (V2) Camera -- Old warhorses but reliable

[image]

davidplowman, 2025-04-03:
Hi, thanks for the report.

My feeling is that the root cause of the problem is quite similar to what you saw previously, which is that the v2 camera module shows very pronounced colour shading which needs to be corrected. However, it seems to be manifesting itself in a different way. I guess we have measured different tables in the new software, and in this quite challenging lighting situation it doesn't seem to work as well.

The "still_stats" mechanism you mention doesn't really mean anything here, because you don't really have a "preview" phase in the same way - you're using just a single camera mode and not switching from a preview mode to a capture mode.

As regards what to do about it, I can't really think of anything other than trying to adjust the lens shading calibration tables. We could reduce the red gain a bit on the right, and perhaps increase it slightly on the left (bearing in mind the sensor is upside down in the mode, so the right of the picture may be the left of the calibration table...). Obviously you can't overdo it as you don't want the opposite problem when there's no sun, but we might be able to find a better compromise.

This would involve editing the JSON file where this table is defined, is that something you are able to do? I'm guessing performing a proper measurement might be difficult, but one could tweak and replace the current values.

Rustimation, 2025-04-16:
Hello David,
sorry for not coming back to you sooner. Although I think, that editing the calibration table is beyond my skills (I am more of a Node-Red guy), I can give it a try. Which file would that be and where is is located?
Anyhow, what would be today's equivalent to the old _camera.still_stats = 'true'_ parameter to get rid of the pinkish discolouring in the center of the image? Could I overcome the missing preview device (i.e. screen) by initiating a preview to a null-device like  _show_preview=False_?

davidplowman, 2025-04-16:
Firstly, to answer the _still_stats_ question, there's not really any such thing any more. Previously, statistics would be computed in the preview phase when the camera is running in its 2x2 binned mode, and then these would be used directly when processing the full resolution capture. As I said earlier, the imx219 is a bit nasty and we found that the preview/capture images don't match as well as they should. Hence the need to re-compute the statistics from the full resolution image. Does that make sense?

For the code you have above, there is no preview phase, so you're using statistics from the full resolution image already. So your problem is not the same, and there's nothing you can do like that to make it go away. Adding a preview phase (which you don't currently have) is not going to make anything any better! But again, as I said, the root cause of your problem is still that the lens/colour shading on the imx219 is pretty nasty, and any imperfections in the correction can give rise to visible artefacts. So that's what you're seeing again, but the nature is quite different - it's no longer a "ring" of discolouration, more of a cast just to one side.

Again, my feeling is that your best bet may be to adjust the lens shading correction tables to find a set of values that works better for you. It may not be possible to get ideal values that work in all cases, but maybe you can find a better compromise.

You should find the imx219 tuning file here: `/usr/share/libcamera/ipa/rpi/vc4/imx219.json`. You will need to be root to edit it, and make sure you back it up before touching anything - errors may stop the camera from starting at all.

The part of the file to look it is [here](https://github.com/raspberrypi/libcamera/blob/main/src/ipa/rpi/vc4/data/imx219.json#L484). This is the red ("calibratinos_Cr") correction for daylight (6000K). You should see a 16x12 array of numbers which are applied, scaled up, across the whole image to the red channel. Now, I think our sensors are normally mounted upside down, so in fact the _right_ half of your image will probably correspond to the _left_ half of this table. Anyway, I'd start with a rather large alteration to one of the corners, just to confirm you're having the effect you want, and also check which way round everything is!

Once you're feeling comfortable, you need to make some more careful adjustments so that, as you travel from the middle to the (probably) left of the table, the numbers don't get quite so large. If you're familiar with manipulating JSON/numpy arrays in Python that would make things easier. Good luck - and get back to us if you get stuck.