raspberrypi/picamera2 #1247: PyAv output stream YUVJ420 full range jpg

mgineer85, 2025-04-06:
Hey @davidplowman ,

just stumpled over this workaround in the pyav output:
https://github.com/raspberrypi/picamera2/blob/fb031cb49cb2af4dab7439be7a90784c9d3db1ab/picamera2/outputs/pyavoutput.py#L37

while the encoder is switching to the extended range for further processing:
https://github.com/raspberrypi/picamera2/blob/fb031cb49cb2af4dab7439be7a90784c9d3db1ab/picamera2/encoders/libav_mjpeg_encoder.py#L50

The PyAv distinguishes between YUV420 (not full range) and YUVJ420 (full range).
Don't understand the details, but maybe the encoder can set directly to YUVJ420 and the workaround is not needed at all?

```python 
self._stream.pix_fmt = "yuvj420p" 
```

davidplowman, 2025-04-07:
Hi, thanks for the info. I just gave this a try and it seems to work the same, so I guess it would be a bit tidier like this! I'll make up a PR.

davidplowman, 2025-04-07:
OK, now I'm really confused. Actually what seems to happen is that if I don't have the color_range workaround in pyavoutput.py, then I don't  get an output file at all! (which is why I thought nothing was changing)

But looking into the MJPEG files produced, it does seem to the case that "yuv420p" gives me limited range YUV, and "yuvj420p" gives me full range. So I think "yuvj420p" for the pix_fmt would actually be correct.

But it seems like the color_range thing needs to stay. Have you tried this for yourself at all? Thanks!

mgineer85, 2025-04-07:
Hi, I fiddled around and come up with following draft pull request: https://github.com/raspberrypi/picamera2/pull/1249

It seems, the encoder_stream codec_context needs to be copied over to the output stream (at least partially).
I chose the width/height and the pix_fmt.

This works for me when testing the example `capture_pyav.py`.

I thought, copying over the encoder stream settings would be a good idea so there is the function add_stream(template=encoder_stream) available but I did not have luck using that.
In newer PyAV add_stream(template=...) was replaced by add_stream_from_template() but it did not work either.

Here is an issue in PyAV repo, maybe it will be relevant in the future: https://github.com/PyAV-Org/PyAV/discussions/1673