raspberrypi/picamera2 #1252: [BUG]  🐛 `RuntimeError: Overlays not supported on this device` on Raspberry Pi 5

nerbivol, 2025-04-17:
### Description

When attempting to use the `set_overlay` function with `Picamera2` on a Raspberry Pi 5, I encounter a `RuntimeError` indicating that overlays are not supported on this device — even though previews without overlays work fine.

### Code to Reproduce

```python
import time
import numpy as np
from picamera2 import Picamera2, Preview

picam2 = Picamera2()

main_config = {
    "format": 'RGB888',
    "size": (2016, 1344)
}
lores_config = {
    "format": 'RGB888',
    "size": (400, 300)
}

picam2.start_preview(Preview.DRM, x=0, y=0, width=400, height=300)

preview_config = picam2.create_preview_configuration(
    main=main_config,
    lores=lores_config,
    display="lores"
)
picam2.configure(preview_config)

picam2.start()

overlay = np.zeros((300, 400, 4), dtype=np.uint8)
overlay[:150, 200:] = (255, 0, 0, 64)  # reddish
overlay[150:, :200] = (0, 255, 0, 64)  # greenish
overlay[150:, 200:] = (0, 0, 255, 64)  # blueish

picam2.set_overlay(overlay)  # <- This raises the error

time.sleep(10)
```

### Error Output

```
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1847, in set_overlay
    self._preview.set_overlay(overlay)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 108, in set_overlay
    raise RuntimeError("Overlays not supported on this device")
RuntimeError: Overlays not supported on this device
```

### Environment

- **Hardware**: Raspberry Pi 5  
- **Library Version**: picamera2 (latest from Raspberry Pi OS bookworm apt)  
- **OS**: Raspberry Pi OS (bookworm)  
- **Python**: 3.11.2  

### Notes

- The code works fine if I **remove the `set_overlay` line** — meaning previews display properly.
- This suggests a device capability check issue — or a limitation in the DRM preview backend on Raspberry Pi 5 that isn't documented.

Is overlay support intentionally not available on Raspberry Pi 5 DRM preview for composite output?

davidplowman, 2025-04-17:
> Is overlay support intentionally not available on Raspberry Pi 5 DRM preview for composite output?

@6by9 Could you maybe comment on this? Thanks!

6by9, 2025-04-17:
Overlays are only supported on the HDMI outputs of Pi5, as those run through the vc4 driver and HVS.

The RP1 display outputs (DSI, DPI, and VEC) only support a single RGB plane with no scaling.

nerbivol, 2025-04-17:
@6by9 Yeah, I saw it when double-checked with `kmsprint`, and here’s what I’m seeing on my Raspberry Pi 5:

**On HDMI:**
```
Connector 0 (33) HDMI-A-1 (connected)
  Encoder 0 (32) TMDS
    Crtc 2 (92) 1920x1080@60.00
      Plane 2 (81) fb-id: 677 ... (lots of formats listed)
        FB 677 1920x1080 RG16
```

**On Composite (VEC) output:**
```
Connector 0 (45) Composite-1 (connected)
  Encoder 0 (49) NONE
    Crtc 0 (48) 704x432i@59.98
      Plane 0 (46) fb-id: 50 ... 
        FB 50 704x432 XR24
```
Did you have any recommendation how it add this plane to composite?

nerbivol, 2025-04-17:
Or i have only option for overlay-like functionality is to **manipulate the image buffer directly**, similar to what’s done in [[opencv_face_detect_3.py](https://github.com/raspberrypi/picamera2/blob/main/examples/opencv_face_detect_3.py)](https://github.com/raspberrypi/picamera2/blob/main/examples/opencv_face_detect_3.py), using something like:

```python
def draw_overlay(request):
    with MappedArray(request, "main") as m:
        # Draw your shapes directly on m.array using OpenCV
        cv2.rectangle(m.array, (50, 50), (150, 150), (0, 255, 0), 2)
```

And then assign it like:
```python
picam2.post_callback = draw_overlay
```

davidplowman, 2025-04-17:
Hi again. Yes, drawing directly onto the image like that is certainly an option.

The advantage of the overlay is of course that it doesn't alter the original image, and this is often useful when drawing things like UI elements on top of the image, things that you don't want to be recorded.

You can often achieve a similar effect by asking for the same image twice. That is, you ask for it once as the "main" stream and then you also configure a "lores" (low resolution) stream, which can actually be the same resolution again if you want. Then you can draw on one of the images for the display, but save or store the other. Does that help at all?

nerbivol, 2025-04-17:
But this approach significantly reduces the number of frames per second.

6by9, 2025-04-17:
There are no simple options for adding a composition stage to the RP1 outputs (also applies to SPI displays, and many others. Even the big boys like amdgpu and i915 only have 2 planes as primary and cursor).

1. compose using Wayland or similar, which uses the GPU.
2. vc4 does have the writeback connector that writes the composed planes to a buffer, and that could then be rendered to the RP1 output. There are a number of threads from [dividuum](https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=73682) on the forums with regard this as he does this in his application.
3. draw on the base image as you've suggested (easiest if you don't ask for a YUV buffer).

picamera2 supports 1 already, but you generally end up with a desktop environment.
You have the access to do 3, but it is requires the application to do it.
2 is more involved, and isn't something that I'm expecting to be easily made into a generic solution.

nerbivol, 2025-04-17:
Okay, but still, I can just run the same code on a raspberry zero 2w and everything will work. What is the difference and complexity of implementation for raspberry pi5?

6by9, 2025-04-17:
On Pi5 the DSI, DPI, and VEC outputs are no longer part of the main SoC (BCM2712) and hence controlled by the vc4 driver. They are now part of the RP1 IC (along with most of the lower speed logic) and are totally different peripherals.

nerbivol, 2025-04-17:
So if I just use the J7 pad for analog output on the Pi 5, does that mean I'm accessing a more limited set of resources compared when I use HDMI?

closed 2025-04-19 by nerbivol