raspberrypi/picamera2 #1258: [HOW-TO] Embed the preview of IMX500 into a PyQt application

HelloCigar, 2025-05-01:
I already have a yolov8 face detection model and I've tested it using the imx500 object detection example. Now  I want to create a simple app in PyQt which has an rfid input from scanner and I want to embed the preview of the camera inside of it while the draw detection function is running. The workflow is like this: check till face detected is only 1 -> enable rfid input -> take the cropped img when rfid condition is satisfied -> do post request. Is it possible to do this when the inference is running and not interfere with the PyQT UI? All in the same windowK

davidplowman, 2025-05-01:
Hi, I wasn't clear from your message whether you're using the inferencing capabilities of the IMX500 sensor, or whether you have a software model that you tested with a version of the IMX500 demo.

I'm sure what you want sounds possible, though it will depend on what you want to do. If you're using the IMX500 inferencing, that doesn't run on the CPU so I'd have thought there shouldn't be too many problems about blocking the UI. If you're running models slowly in software, you might need to consider some kind of `QThread` so as to get the model inferencing time out of the Qt event loop. Unfortunately I can't claim any great expertise in writing Qt apps, does that help at all?

HelloCigar, 2025-05-01:
Hi. Thank you for your response. I just used my custom model instead of the default provided by the imx500_object_detection_example. What my main problem is that the imx500 examples spawns a Qt preview window running the model in real time but no import of PyQT is found in the source code. I know it is possible to embed the preview window of picamera2 inside a PyQT app, but I'm not sure how while the model is running and without interfering with the event loop. I guess I was just looking for more examples.

davidplowman, 2025-05-01:
So you're using the CPU for inferencing, and not the IMX500, have I understood that correctly? Presumably this model runs at just a few fps - in which case I guess you probably do want to get it out of the event loop.

I also assume you've found the [examples](https://github.com/raspberrypi/picamera2/tree/main/apps). There aren't any examples there that do anything quite like that, though I think they show you how you would call `capture_array()` (or whatever) and collect the result. You can certainly then dispatch that to another thread for processing - I don't know if using a `QThread` would make that easier.

HelloCigar, 2025-05-01:
No, the model definitely runs on the IMX500 chip in Raspeberry Pi AI Camera. [This](https://github.com/raspberrypi/picamera2/blob/main/examples/imx500/imx500_object_detection_demo.py) uses the pre_callback to draw detections. In the picamera2 pdf, page 65-67 show how to embed the preview but I'm confused how to combine these two examples.

davidplowman, 2025-05-01:
Ah, OK. Well I think that makes life easier if the model is running on the IMX500 and not the CPU. So I think what I'd do is something like this:

1. Firstly, I'd just make a Qt app that displays the camera preview.
2. Next I'd add code so that I'm continually calling `picam2.capture_metadata(wait=False, signal_function=qpicamera2.signal_done)`. There'll have to be an initial call, and then the callback (see next item) will have to launch the next `capture_metadata`.
3. I'd connect a callback, let's call it `metadata_acquired`, to the `qpicamera2`'s `done_signal`.
4. In `metadata_acquired`, I'd fetch the metadata (`metadata = job.get_result()` and parse it.
5. You can define a `pre_callback` just like the other examples to draw directly onto the  camera preview image. Or if you preferred, you could avoid the `pre_callback` and use `set_overlay` directly in the `metadata_acquired` function, once the parsing is finished.

Does that help a bit?

HelloCigar, 2025-05-01:
picam2 = Picamera2(imx500.camera_num)
    config = picam2.create_preview_configuration(controls={"FrameRate": intrinsics.inference_rate}, buffer_count=12)

    imx500.show_network_fw_progress_bar()
    #picam2.start(config, show_preview=True)

    if intrinsics.preserve_aspect_ratio:
        imx500.set_auto_aspect_ratio()

    from PyQt5.QtWidgets import QApplication
    from PyQt5.QtCore import QTimer
    from picamera2.previews.qt import QGlPicamera2
    
    app = QApplication([])
    qpicamera2 = QGlPicamera2(picam2, width=800, height=600, keep_ar=False)
    qpicamera2.setWindowTitle("UPTap Attendance System")

    
    last_results = None
    picam2.pre_callback = draw_detections
    import threading
    def detections_loop():
        while True:
            last_results = parse_detections(picam2.capture_metadata())
            print(last_results)
    thread = threading.Thread(target=detections_loop, daemon=True)
    thread.start()
    
    picam2.start()
    qpicamera2.show()
    app.exec()

hello, I did some dirty addition to your work sorry. Why does this not work? There's a Detection object detected but it is not drawn on top of the preview

davidplowman, 2025-05-01:
Hmm, not sure. I'd put a `print` of some sort into the `draw_detections` just to check it's being called (I can't see why it wouldn't be). Maybe also put some more `print` statements in there to check that it really is finding some objects that should get drawn?