raspberrypi/picamera2 #1261: [BUG] "Stream 'r' is not defined" when capturing raw buffer data from an unsupported stream type

anmolbyte, 2025-05-05:
**Please only report one bug per issue!**

**Describe the bug**
When calling `picam2.capture_buffers("raw")` on a Raspberry Pi 5 with an OV9281 camera (after configuring both "main" and "raw" streams), the call fails with `RuntimeError: Stream 'r' is not defined`. This occurs even though the camera configuration is applied successfully (resulting in a `MONO_PISP_COMP1` raw stream) and `picam2.capture_metadata()` works correctly just before the failing call. The traceback indicates the error originates in `request.make_buffer`, suggesting the stream name `"raw"` is being misinterpreted internally as `"r"`.

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

- Connect an OV9281 camera to a Raspberry Pi 5.
-  Run the following Python script:
```

import time
import traceback 
from picamera2 import Picamera2
# import numpy as np # Not needed to show the bug

# --- Configuration ---
CAMERA_INDEX = 0 # Assuming only one camera detected or using the first one
RESOLUTION = (1280, 800)
RAW_FORMAT = 'R10_CSI2P' # Requesting this, but expect override
# --- ---

picam2 = None 
try:
    print(f"Initializing Picamera2 for camera index {CAMERA_INDEX}...")
    picam2 = Picamera2(camera_num=CAMERA_INDEX)

    # --- Configuration (Standard Approach) ---
    raw_config_dict = {"size": RESOLUTION, "format": RAW_FORMAT}
    print("\nCreating standard configuration...")
    config = picam2.create_still_configuration(raw=raw_config_dict) 
    print("\nRequesting configuration:")
    print(config) 

    # Configure the camera
    picam2.configure(config)
    print("\nActual configuration applied:")
    cam_config = picam2.camera_configuration()
    print(cam_config) 
    actual_raw_format = cam_config.get('raw', {}).get('format', 'Unknown')
    print(f"--> Actual Raw stream format: {actual_raw_format}") # Expect MONO_PISP_COMP1

    # Start the camera
    print("\nStarting camera...")
    picam2.start()
    time.sleep(1) 

    # --- Capture Metadata (Works) ---
    print("\nCapturing metadata...")
    metadata = picam2.capture_metadata() 
    print("Capture Metadata Captured (omitted for brevity)")
    # print(metadata) 

    # --- Attempt to Capture Raw Buffer (Fails) ---
    print(f"\nAttempting capture_buffers for stream 'raw'... (EXPECTED TO FAIL)")
    # This is where the error occurs
    buffers = picam2.capture_buffers("raw") 

    # Code below this point is not reached due to the error
    if buffers:
        print(f"\n*** Successfully captured raw buffer! (BUG NOT PRESENT) ***")
        # ... processing ...
    else:
        print("\ncapture_buffers did not return any buffers.")

except Exception as e:
    print(f"\n--- ERROR ---")
    print(f"An error occurred:")
    traceback.print_exc() 
    print(f"-------------")

finally:
    # --- Cleanup ---
    if picam2 is not None and picam2.is_open:
         if picam2.started:
              print("\nStopping camera...")
              picam2.stop()
         print("Closing camera...")
         picam2.close()
    print("\nScript finished.")
```

**Console Output, Screenshots**
```
Attempting capture_buffers for stream 'raw'... (EXPECTED TO FAIL)

--- ERROR ---
An error occurred:
Traceback (most recent call last):
  File "/home/anmol/untitled2.py", line 49, in <module>
    buffers = picam2.capture_buffers("raw")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1914, in capture_buffers
    return self.dispatch_functions([partial(self.capture_buffers_and_metadata_, names)], wait, signal_function)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[...]
  File "/usr/lib/python3/dist-packages/picamera2/request.py", line 150, in make_buffer
    raise RuntimeError(f'Stream {name!r} is not defined')
RuntimeError: Stream 'r' is not defined
```
**Hardware :**
Raspberry Pi 5 Model B
OV9281 Monochrome Camera Module (connected via CSI interface)

davidplowman, 2025-05-06:
Hi, thanks for the question. I think `capture_buffers` is for capturing buffers from multiple different streams, so you would need to pass it a list of stream names. Maybe try `buffers = picam2.capture_buffers(["raw"])` and get back to us if there's still a problem. Thanks!

anmolbyte, 2025-05-06:
yep, that was the issue- this is what I get for vibe coding lol. Would you happen to know a way to process MONO_PISP_COMP1 raw buffer? thanks so much

davidplowman, 2025-05-07:
It's defaulting to a compressed raw format. You probably need to specify an uncompressed raw stream format `raw={'format': 'R10'}` when you create the configuration. The `CompletedRequest` does have a `decompress` method for buffers like that, but it's going to be much quicker to ask for it uncompressed in the first place.

In my experience, AI coding tools are pretty rubbish with Picamera2. There's not enough code around for them to get really good, meanwhile they get led astray by all the legacy PiCamera code that is now completely broken and which will lurk on the internet forever.