raspberrypi/picamera2 #1263: [BUG] exposure time not being honored after upgrade to latest version

tim-oe, 2025-05-07:
**Please only report one bug per issue!**

**Describe the bug**
ExposureTime not being applied when set via controls.

the picture is dark  and the exif tags showing 
Exposure Time           : 1/15
ISO                             : 800
Shutter Speed            : 1/15

**To Reproduce**
take picture at night and set the following controls:

                controls = {
                    "ExposureTime": (3 * 1000000),
                    "AnalogueGain": 8
                }

take picture
**Expected behaviour**
lighter pic with long exposure
exif tags showing:
Exposure Time           : 3
ISO                             : 800
Shutter Speed             : 3

**Console Output, Screenshots**

**Hardware :**
Raspberry Pi 4B rev 1.4 2g memory
camera: https://www.arducam.com/product/arducam-5mp-m12-picam/

**Additional context**
python3-picamera2/stable,stable,now 0.3.26-1 all

class Camera:

    # second to micro second multiplier
    MICRO_SECOND: int = 1000000

    def __init__(self):
        """
        ctor
        :param self: this
        """

        # https://forums.raspberrypi.com/viewtopic.php?t=364677
        # default logging is warn
        Picamera2.set_logging()

        self._camera_config: CameraConfig = AppConfig().camera

        self._base_dir: Path = self._camera_config.folder

        self._base_dir.mkdir(parents=True, exist_ok=True)

        tuning = None
        if self._camera_config.tuning_enable is True:
            tuning = Picamera2.load_tuning_file(self._camera_config.tuning_file)

        self._picam2 = Picamera2(tuning=tuning, allocator=PersistentAllocator())

    def process(self, lux: int) -> str | None:
        """
        camera processing entry point
        :param self: this
        :param lux: the ambient light level to contol exposure/iso settings
        :returns image path
        """
        if self._camera_config.enable is False:
            self.logger.debug("camera not enabled")
            return None

        try:
            self._picam2.start_preview(Preview.NULL)

            # TODO is this needed?
            time.sleep(1)

            preview_config = self._picam2.create_preview_configuration()
            self._picam2.configure(preview_config)

            controls = {}
            if lux <= self._camera_config.lux_limit:
                controls = {
                    "ExposureTime": (self._camera_config.exposure_time * Camera.MICRO_SECOND),
                    "AnalogueGain": self._camera_config.analogue_gain,
                }
            else:  # clear controls https://github.com/raspberrypi/picamera2/issues/1175
                controls = {
                    "ExposureTime": 0,
                    "AnalogueGain": 0,
                }

            self._picam2.start()

            time.sleep(1)

            img_file: str = self.image_file()
            capture_config = self._picam2.create_still_configuration(controls=controls)
            # TODO inject exif data here
            self._picam2.switch_mode_and_capture_file(capture_config, img_file, wait=True)

            return img_file
        except Exception as e:
            self.logger.exception("failed to take pic...")
            raise Exception("failed to take pic...") from e
        finally:
            self._picam2.stop_preview()
            self._picam2.stop()

LeifSec, 2025-05-07:
Since yesterday i have got a similar problem:

```
from picamera2.picamera2 import Picamera2
cam = Picamera2()

def f(request):
   metadata = request.get_metadata()
   print(metadata['ExposureTime'], "/", metadata['AnalogueGain'])

controls = {"ExposureTime": 100, "AnalogueGain": 5}
capture_config = cam.create_still_configuration(controls=controls)
cam.configure(capture_config)
cam.start()
```

output
```
89938 / 4.0
119989 / 5.595628261566162
119989 / 5.3612565994262695
119989 / 5.3612565994262695
119989 / 5.389473915100098
119989 / 5.389473915100098
119989 / 5.389473915100098
119989 / 5.389473915100098
```

That looks like that auto-exposure time is activated ("ExposureTime": 0) although it is not activated.

I using current version of Raspberian Stable (32 bit):
```
python3-picamera2 is already the newest version (0.3.26-1)
```

davidplowman, 2025-05-07:
Hi, thanks for this. Indeed, something got overlooked. I've patched it up now, and will make another release up.

davidplowman, 2025-05-07:
You should be able to pick up the fixed version from apt now (`sudo apt update` and `sudo apt install python3-picamera2`).

tim-oe, 2025-05-08:
it's working! 
thanks for the quick response

closed 2025-05-08 by tim-oe