raspberrypi/picamera2 #1271: [BUG] PyavOutput dcoes not work on Pi 4b

JooJooBee666, 2025-05-27:
The problem is that no matter how I initialize the camera, the PyAVOutput screams about having an invalid stream.  I used the sample code right from this repo.

To reproduce, simple install Bullseye on a Pi 4b, apt install python3-picamera2, then try to run this:
```
#!/usr/bin/python3

# Example using PyavOutput to record to an mp4 file.

import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import PyavOutput

picam2 = Picamera2()
config = picam2.create_video_configuration({'size': (1280, 720), 'format': 'YUV420'})
picam2.configure(config)

encoder = H264Encoder(bitrate=10000000)
output = PyavOutput("test.mp4")
picam2.start_recording(encoder, output)

time.sleep(5)

picam2.stop_recording()
```

I would expect this to create the test.mp4 file in the folder of the script....which it does but not until it crashes with the following error:
```
[8:39:31.806185009] [7913]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+59-d83ff0a4
[8:39:31.864577681] [7959]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[8:39:31.872419308] [7959]  INFO RPI vc4.cpp:447 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media0
[8:39:31.872679471] [7959]  INFO RPI pipeline_base.cpp:1121 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[8:39:31.943010192] [7913]  INFO Camera camera.cpp:1205 configuring streams: (0) 1280x720-YUV420 (1) 2028x1080-SBGGR12_CSI2P
[8:39:31.950823245] [7959]  INFO RPI vc4.cpp:622 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
Exception in thread Thread-7 (thread_poll):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/encoders/v4l2_encoder.py", line 257, in thread_poll
    self.outputframe(b, keyframe, (buf.timestamp.secs * 1000000) + buf.timestamp.usecs)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/encoders/encoder.py", line 326, in outputframe
    out.outputframe(frame, keyframe, timestamp, packet, audio)
  File "/home/pi/penvs/ps/lib/python3.11/site-packages/picamera2/outputs/pyavoutput.py", line 89, in outputframe
    packet.stream = orig_stream
    ^^^^^^^^^^^^^
TypeError: Argument 'stream' has incorrect type (expected av.stream.Stream, got NoneType)
```

**Hardware :**
Pi 4b and an Arducam Camera model UC-517

**Additional context**
I tried the normal ffmpegencodeer but apparently, the later version of this are bugged as well as I can start the recording and finish it, and the file saves, but the encoder thread never exits and thanks to stupid python, there is absolutely NO WAY to kill a child thread from the parent.  And I mean KILL.  the Picamera2 ffmpegencoder working prior to an apt update I did a few months back that broke this on all my currently deployed units so now I have to run them out of date. But, I digress as it's not fully related.  I was hoping to use a new method to encode, but sadly not even this works.

davidplowman, 2025-05-28:
Hi, and thanks for the report.

Can we start by checking software versions. You mention that you're using Bullseye, is that correct? Bullseye has been the "legacy OS" for some time now, so it's not possible to fix anything there now. And also not surprising if things are not working there that have been fixed since.

I tried your script on our latest Raspberry Pi OS release (which is Bookworm), where it ran without any problems. Is this something you're able to try? Note that I was using one of the official Raspberry Pi camera as I don't have access to any 3rd party modules. Thanks!

I'm afraid I didn't fully understand what you wrote about the "ffmpegencoder". There is no ffmpegencoder, so I think you may be referring to the FfmpegOutput? Again, there's probably nothing I can do about Bullseye, but if you have (simple) test cases that are failing on the latest Raspberry Pi OS, please do post them and we can investigate. Thanks.

JooJooBee666, 2025-05-29:
Ok. Done and working.  

However, I am now seeing something similar to this [issue.](https://github.com/raspberrypi/picamera2/issues/1240).  However, I do not see that it was resolved.   When I run the test circular recording with pyavOutput, I get a video that jumps over the first sleep time after starting the video.  All I get are the last 5 seconds (or whatever I set that last sleep to).  The video key frame seem to be all munged up.  If I change the ending sleep timer, I get a video that is over 485701 hours long and crashes VLC.   

So even after all of this, it appears I still cannot use this on the Pi 4b.  Any way you can tell me how to implement [this](https://github.com/raspberrypi/picamera2/issues/1240#issuecomment-2772113380)?

davidplowman, 2025-05-29:
Thanks for the update. Would you be able to include a simple script (like the original one you posted) that demonstrates the problem. That would make it easier for me to understand what's happening and to be sure I'm investigating the right thing. Thanks.

JooJooBee666, 2025-05-29:
Actually, I was wrong and in the process of updating my previous post.  It is NOT working correctly, though my initial problem is gone.

JooJooBee666, 2025-05-29:
This is a clean install as you requested.  I literally spent all of yesterday evening getting it set back up on a clean 256GB SD.  Granted, I had to make some minor changes for my environment such enabling X11 instead of wayland, enabling RealVNC service and modifying the config.txt so my camera would work and my HDMI resolution is configured properly.  I'm happy to leave copy of the config.txt if you think this may be related in any way.

davidplowman, 2025-05-29:
OK, well this is all very mysterious. I need to do some other stuff but I'll take a Pi 4B home tomorrow, redo my clean install and try it all again. I don't have that Arducam camera, but IIRC it's basically an IMX477, is that right? Maybe also post a link to the output of the first of those 3 scripts, maybe just 10 seconds if poss, that seems like the easiest failure case to start with, and I can compare it with what I get. Thanks.

JooJooBee666, 2025-05-29:
Camera is an imx477 and the following edit is REQUIRED to geet it working on the config.txt:
```
# Automatically load overlays for detected cameras
camera_auto_detect=0
dtoverlay=imx477
```

JooJooBee666, 2025-05-29:
> Thanks. Actually you only need the config.txt edit when using a 3rd party camera so I'll be OK on that!

Ok great.  Let me know what you find as I'd love to use this in my security camera project.  It's currently using the old CircularBuffer with some overrides to input the frames manually and handle start/stop of recording but I have no audio for the first 4 seconds in any saved/recorded video then using ffmpegOutput to mux it with mic audio (with a 4 second delay).  Hopefully you are able to reproduce! 🤞

davidplowman, 2025-05-29:
Yes, will let you know tomorrow. As I think I said, maybe a link to a 10-second "garbage" file might be handy, in case I want to compare it to mine.

davidplowman, 2025-05-30:
OK, I think I've made some progress here.

The first discovery was that your simple "pyav with audio" file was just like mine. It plays using ffplay. But, like mine, I found I couldn't get vlc to play the audio stream, ever after fiddling around and re-muxing, it just didn't see it. So there's obviously something "fishy" with these files.

Diving into it, I saw that the video stream always starts at time zero, but the audio seems to have system timestamps on it. Evidently some players see that, think "that's wrong, I'll use zero instead", and work - but others don't.

Anyway, I've patched it up so that audio streams start at zero too, and vlc is happy with the files now. (Actually I think my patch is a bit too simple at the moment - the video and audio streams need to agree what "zero" is jointly, but that will require a bit more work. I don't think the difference will be critical.)

In the CircularOutput2 there was a clear bug, leaving me in that "how did this ever work?" state. So I don't know what was going on there. Anyway, I fixed something, and I think it's better.

If you want to give these fixes a try, check out the pyav-fixes branch in this repo, point your PYTHONPATH at it (or whatever), and let me know how you get on.

JooJooBee666, 2025-05-31:
Working on this over the weekend.  I will report my results when completed. 🍻

JooJooBee666, 2025-05-31:
Well, we made some progress here!  The direct pyavOutput works with Audio now, but the circularoutput2 is still a train-wreck.  It's not much to look at this time, but [here's the current state of that](https://1drv.ms/v/s!AvOmBRN37ygOijRiOkwsuw5ZlWs8?e=VQlgJi) if you would like to further analyze.

If you would like to still review the outputs from pyavoutput direct, let me know and I will upload as well.

davidplowman, 2025-06-02:
Unfortunately I don't seem to be able to download those files any more, not sure what's going on. It made me log in with a Microsoft account and then told me it didn't exist or I didn't have permissions.

Anyway, I've run that third script, once as written without audio, and once with, and uploaded the files [here](https://drive.google.com/drive/folders/1EYs68Cyuov_BUQHzb75bAwi0R3IaZutg?usp=sharing). I start the stopwatch (as near as I can) when the camera starts, so the videos should be showing approximately seconds 5 to 10. In practice you get a bit less than that (for "reasons"), but they look OK to me (and to both ffplay and vlc).

Can you try and play them and check if they're good for you or if you see a problem with them? Thanks!

davidplowman, 2025-06-04:
OK, well I think that may explain what's going on. A bit weird that the files don't play properly for you with VLC (I assume you meant VLC and not VNC?) while it does for me, but maybe there's a version thing going on there.

The mp4 files that get recorded have the timestamps from when the CircularOutput2 is started. So in the files I provided, the timestamps of both streams start shortly after 5s and end just before 10s. So far as I know, this is perfectly legal, so I'm quite surprised that some software seems not to like it.

In my folder I've put versions of my 2 files and your circularoutput2 file which I've remuxed just to bring the initial timestamp down to zero. Perhaps you could test those and check they are now OK for you, as that would verify that we've diagnosed the problem correctly.

(If you want to remux any files of your own as further confirmation, the command I used was `ffmpeg input.mp4 -map 0 -c copy -avoid_negative_ts make_zero remux.mp4`)

JooJooBee666, 2025-06-08:
Remuxing is not an option.  This needs to come out of the Pi4b ready to go.  I'm just not sure what I'm supposed to do with this at this point as this is still basically broken and unusable here. It's obviously not a solution to the root of the problem.  Those files **are not ok** and still do not play back properly.  It's still the same problems in multiple applications...which should clearly say there is still a problem with the how the files are created.  Is there some other possible code changes you would like me to test to try and get this resolved?

davidplowman, 2025-06-09:
OK, well I'm trying very hard to help fix the problem here. If we can establish what the problem is, which is why I sent those files, then I can probably fix it. I suggested remuxing other files only for further confirmation. 

But from what you say, if those remuxed files are not working for you, it may not be the initial non-zero timestamps in the file, in which case I have no idea what we're looking for. 

Anyway, I believe I've fixed the non-zero starting timestamp problem in that pyav-fixes branch, so you could give that a try. If that doesn't do the trick, then we need to find some way to determine what is "bad" about these files - is there anything I can run on the Pi that will tell me? Anyway error messages being produced that might give us a clue? What are these "multiple applications", and is there one I can easily try?

JooJooBee666, 2025-06-10:
I will give the updated branch another go here today or tomorrow.  I need to set up my bench test again as I rolled out the last update I was working on.

Regarding the applications I checked with, they are:

- VLC (Windows and Linux)
- Dragon Player (Linux)
- Media Player in Windows 11.

davidplowman, 2025-06-10:
I've tried an output file from the latest branch on both VLC and Dragon Player on the Pi 4 and my Ubuntu 22.04 PC (I have no ability to run Windows 11). As far as I can tell, everything is playing perfectly, though Dragon Player complains a lot about "chain filter error: Too high level of recursion (3)" while starting up (but is then fine).

I'm aware that versions on Pi OS and Ubuntu 22.04 are probably all a bit old, so maybe we have some kind of version issue going on? Here's what I have:

My Pi:
vlc 3.0.21
dragon player 22.12.3

My PC:
vlc 3.0.16
dragon player 21.12.3

Both Dragon Player versions spit out that error. Looking it up, it's not uncommon - but mostly reported against vlc! So I guess there are some shared libraries underneath? Who knows.

Anyway, let me know what versions you have and what doesn't work.

JooJooBee666, 2025-06-12:
I am seeing video and hearing audio, but not what I am expecting.  So, maybe I'm the one that is confused?

From the script, I would expect an output.mp4 of 10s in length; 5s for the buffer, and the additional 5s sleep.  Basically, I would expect 5s (or whatever it's set to) of buffer prepended to the additional recording of all the second sleep timer.  

Countering this to your explanation where you say "...the data written to output.mp4 will be from t=5s to t=10s" makes it sound like you only expect a 5s output. Where did the other 5s of video go from the second sleep timer?

Anyhow, I look forward to the answer to this.  As far as testing, I have not found the time just yet but I plan on doing it today.  I'll let you know my results when I have them.

JooJooBee666, 2025-06-12:
Ok, confirmed it is working much better now after this last round of fixes.  Therefor, I  would say this bug appears mostly fixed from this perspective.

The only remaining question is how do we flush out the rest of the buffer to the same file?  i.e. if I have a timer in front of it (which I did to test), and it stops on 20 seconds, is there any way to get those remaining frames of AV from the buffer written out immediately without destroying what's in there?  Right now, and in the updated demo, its configured to dump those to a separate file, but that won't work for my needs.  I know this is a bit out of scope of the bug, but just wondering.  If not, I'll look at doing some overrides to see if I can make it do what is needed for my devices.

If this is just not possible, that's fine.  Let me know and I'll submit a feature request and you can close this if you are satisfied as well.

Thanks again! 🍻

davidplowman, 2025-06-16:
Glad that things are looking better, With the code on the "next" branch, I'm not aware of any actual problems, so if you believe there are any issues left please do let me know (with a simple test case, of course!). Thanks.

JooJooBee666, 2025-06-19:
The only issues left were the key frame issues (need more of them) and the fact that some of the mp4's are so far out of spec that ffprobe won't even read them, though they seem to generally play.

davidplowman, 2025-06-19:
So the I frame period can simply be decreased in the codec, I believe.

Can you explain what you mean by "out of spec"? Perhaps send me an example file. I haven't seen this problem - and I can't fix anything until I do. Thanks!

JooJooBee666, 2025-06-23:
I'm not sure exactly what is out of spec but I assumed ffprobe's outright **refusal** to show any information about some of the MP4 files created was a dead giveaway to some issues with the encoding.  I do not know how to check on anything else with that as that is my go to tool.....which doesn't work so I'm not sure where to begin and would have to learn the whole kit and caboodle regarding encoding videos.  Of which, I am a long ways away from being to that point as I haven't even started (my current understanding is what I would consider rudimentary, but better than the average street Joe by a longshot in comparison 😛 ).

I overwrote the broken videos when I was doing some follow up testing.  If I create another one, I will upload and link it here for you to review.  I tried again today but both videos I made were fine and ffprobe had no difficulty processing them.

JooJooBee666, 2025-07-26:
I agree that this clearly fixed some issue.  I haven't had time to return to this recently but I know this helped move things forward.  I can see that for myself with the outputs. 😄

davidplowman, 2025-07-28:
Great, thanks for the confirmation that things are working better. If we can ever pinpoint any other problems/improvements, I'm always keen to implement them!!