raspberrypi/picamera2 #1278: [HOW-TO]

Akshar062, 2025-06-13:
Hi team,

I'm working on a Raspberry Pi project using the picamera2 library with GPIO-based LED and PWM light control to simulate a multi-light, multi-exposure imaging system.

The loop captures 4 frames in sequence:

Each frame activates a different light (3 regular GPIO LEDs + 1 PWM-based trigger lamp).

Each frame is configured with a unique exposure and gain setting.

The lights are synchronized manually using gpiozero before frame capture.

Here’s what I’m doing:

Set FrameDurationLimits, ExposureTime, and AnalogueGain for the current frame.

Turn on the corresponding light for that frame (using LED.on() or PWMLED.value = x).

Wait briefly, then capture the frame using capture_array(wait=True) or capture_request().

Turn off the light.

Repeat for next frame.

Problem:
I'm facing synchronization issues between the camera settings, the light trigger, and the captured frame:

Sometimes the camera captures with exposure from the previous frame.

Occasionally, the high-exposure frame blinks or overwrites another frame's settings, likely due to delay or request mismatch.

Despite setting FrameDurationLimits and adding delays (time.sleep()), this inconsistency continues.

`    
def capture_loop(self):
        exposure_map = {
            0: 2200000,    # High exposure frame
            1: 50000,
            2: 300000,
            3: 30000
        }

        gain_map = {
            0: 23.0,
            1: 25.0,
            2: 32.0,
            3: 20.0
        }

        while self.running:
            try:
                index = self.frame_count % 4
                exposure = exposure_map[index]
                gain = gain_map[index]
                frame_duration = exposure + 5000  # Enough slack

                # Apply frame duration limits
                self.picam2.set_controls({
                    "FrameDurationLimits": (frame_duration, frame_duration)
                })

                # Apply exposure and gain
                self.picam2.set_controls({
                    "ExposureTime": exposure,
                    "AnalogueGain": gain
                })

                # Wait for settings to take effect
                time.sleep(frame_duration / 1_000_000.0 + 0.05)

                # Capture frame with synchronized request
                request = self.picam2.capture_request()
                frame = request.make_array("main")
                metadata = request.get_metadata()
                request.release()

                actual_exp = metadata.get("ExposureTime")
                actual_dur = metadata.get("FrameDuration")
                print(f"Frame {self.frame_count}: Exposure {actual_exp}μs | Duration {actual_dur} | Gain {gain}")

                # Store and show
                if not hasattr(self, "frame_buffer"):
                    self.frame_buffer = [None] * 4
                self.frame_buffer[index] = frame

                self.frame_count += 1

                if self.frame_count % 4 == 0:
                    for i in range(4):
                        if self.frame_buffer[i] is not None:
                            self.display_frame(i, self.frame_buffer[i])

            except Exception as e:
                print("[!] Frame capture error:", e)
`

Questions:
How can I ensure that camera exposure and gain settings are committed before capture starts?

Is there a way to force synchronization between light trigger and actual frame exposure?

Would a manual flush_requests() or a delay pattern (e.g., skipping 1 frame after setting controls) help with stability?

Is capture_request() safer for this use-case than capture_array(wait=True) in terms of sync?

Any suggestions on best practices to ensure stable camera-light synchronization for every frame would be really helpful.

davidplowman, 2025-06-13:
Hi, that looks like an interesting (and tricky) problem!

Here's the advice I would give. I apologise in advance if it's a bit rambling, but if any of it looks relevant we can always pursue it subsequently.

1. The time it takes for exposure and gain settings to take effect is (normally) a fixed number of frames. It should be something like `num_buffers + 3` or maybe `num_buffers + 4`, where `num_buffers` is the value that was given to the camera configuration. However, I say _normally_ because it's not 100% guaranteed - there can be internal frame drops, though these are rare unless the camera is running at a high framerate.

The only way to be 100% sure that your settings have taken effect, is to check what's in the image metadata - this is guaranteed to be correct. But note that the values are _what the camera is using_, not _what you asked for_. The camera can't produce arbitrary exposure times and gain, so your value will not be exact, but "close".

2. Synchronising the light and camera frame is certainly tricky, but there might be some things we can do.

If you know the time when the light is triggered, you can wait for the first camera frame after the trigger moment (using the "SensorTimestamp" in the metadata). But note that most cameras start exposing a frame _before_ the sensor timestamp is recorded, so you'd have wait for the first frame where `sensor_timestamp - exposure_time > trigger_time`. I think many of the capture methods have a `flush` parameter which helps with this (I think just set `flush=sensor_timestamp`).

Another thought is that once your camera is set to a fixed framerate (as you are doing, but remember this has the same delay as setting the exposure!), then you can sample two frames and predict very accurately when subsequent frames will be. Maybe you can now arrange for the light to be triggered at the right moment to match a camera frame? You'd still have to allow for the exposure time, but more significantly, Python can be super-non-real-time, so you might have trouble triggering reliably at the right moment.

3. I doubt that sneaky delay patterns, or using capture_requiest() instead of capture_array() would make much difference.

But what I might also add, is that gaining more of a feel for how things change in the camera system. Maybe replace the `time.sleep(delay)` by
```
start_time = time.monotonic()
while time.monotonic() - start_time < delay:
    metadata = picam2.capture_metadata()
    print(metadata['SensorTimestamp'], ":", metadata['ExposureTime'], metadata['AnalogueGain'])
```
and you'll understand better what's going on.

I hope some of that might be helpful, do get back to me if you want to know more about any of it. Thanks!

Akshar062, 2025-06-17:
Hi, thanks for the suggestion!

I've implemented the approach using set_controls with FrameDurationLimits and added a custom method to check if the settings have stabilized:

```
def apply_settings(self, exposure_time, gain):
    frame_duration = max(exposure_time + 1000, 33333)
    self.picam2.set_controls({
        "AeEnable": False,
        "ExposureTime": exposure_time,
        "AnalogueGain": gain,
        "FrameDurationLimits": (frame_duration, frame_duration)
    })

def settings_stabilized(self, target_exposure, tolerance=0.05):
    metadata = self.picam2.capture_metadata()
    actual_exp = metadata.get("ExposureTime", 0)
    print(f"[Metadata] Exposure: {actual_exp}")
    return abs(actual_exp - target_exposure) < target_exposure * tolerance

```

This works well for most exposure values, but I noticed that for higher exposures (like 2.5 seconds), the delay becomes quite significant. It takes a few seconds before the settings stabilize and the frame is ready to be captured.

Let me know if there's a way to speed up stabilization for long exposures — or if this delay is expected due to sensor limits.

Thanks again!

davidplowman, 2025-06-17:
Hi,  that's great that you've made some progress.

Just one suggestion on long exposures.

Usually, when exposures are short, it's quickest to set the exposure time and wait a few frames.

The alternative would be to stop the camera, and then to restart with the new exposure time. If you apply this before restarting the camera, it should take effect straight away. Normally we avoid this method because it typically takes up to ~1s to restart the camera, so the previous method (set it and wait a few frames) is quicker. But in your case, you may well find it preferable.

Akshar062, 2025-06-17:
Hi, thank you for the great suggestion!

I've implemented the long-exposure stabilization logic based on your advice.

For shorter exposures (e.g., under 200–300ms), I continue using the live metadata loop to wait for the exposure to stabilize. However, when the requested exposure time is long (e.g., ≥ 1 second), I now stop the camera, apply the new exposure and gain settings during reconfiguration, and restart the camera. This ensures the first captured frame is already correctly exposed, avoiding the need to discard several initial frames.

This hybrid approach seems to work well so far — it significantly reduces the delay when switching to a long exposure (e.g., 2.5s), compared to waiting multiple seconds while the sensor stabilizes.

Here’s the final setup I’m using, including live metadata logging to analyze frame timings:

```
from picamera2 import Picamera2
import time
from PySide6.QtWidgets import QApplication, QWidget, QLabel, QGridLayout
from PySide6.QtGui import QImage, QPixmap
from PySide6.QtCore import Qt, QTimer
import numpy as np
import sys

class CameraManager:
    def __init__(self):
        self.picam2 = Picamera2()
        self.picam2.configure(self.picam2.create_still_configuration(
            main={"size": (1920, 1080), "format": "RGB888"},
            controls={
                "AeEnable": 0,
                "AfMode": 0,
                "LensPosition": 7.0,
            }
        ))
        self.picam2.start()

    def apply_settings(self, exposure_time, gain):
        frame_duration = max(exposure_time + 1000, 33333)
        
        # If exposure is long, restart the camera with settings
        if exposure_time >= 1000000:  # 1 second
            print("🔄 Restarting camera for long exposure")
            self.picam2.stop()
            self.picam2.configure(self.picam2.create_still_configuration(
                main={"size": (1920, 1080), "format": "RGB888"},
                controls={
                    "AeEnable": 0,
                    "AfMode": 0,
                    "LensPosition": 7.0,
                    "ExposureTime": exposure_time,
                    "AnalogueGain": gain,
                    "FrameDurationLimits": (frame_duration, frame_duration)
                }
            ))
            self.picam2.start()
        else:
            self.picam2.set_controls({
                "AeEnable": False,
                "ExposureTime": exposure_time,
                "AnalogueGain": gain,
                "FrameDurationLimits": (frame_duration, frame_duration)
            })

    def settings_stabilized(self, target_exposure, tolerance=0.05, delay=2.5):
        start_time = time.monotonic()
        stabilized = False

        while time.monotonic() - start_time < delay:
            metadata = self.picam2.capture_metadata()
            actual_exp = metadata.get("ExposureTime", 0)
            gain = metadata.get("AnalogueGain", 0)
            timestamp = metadata.get("SensorTimestamp", 0)

            print(f"[Metadata] {timestamp} : Exposure={actual_exp}, Gain={gain}")
[... 75 lines cut ...]
            self.capture_timer -= 100
            if self.capture_timer <= 0:
                frame = self.camera_manager.capture_frame()
                print(f"📸 Captured frame {self.current_index + 1}: {frame.shape}")
                qimage = convert_to_qimage(frame)
                self.labels[self.current_index].setPixmap(QPixmap.fromImage(qimage).scaled(
                    self.labels[self.current_index].size(), Qt.KeepAspectRatio))
                self.current_index = (self.current_index + 1) % len(self.settings_list)
                self.state = "apply"

def main():
    app = QApplication(sys.argv)
    camera_manager = CameraManager()
    window = PreviewWindow(camera_manager)
    window.show()

    sys.exit(app.exec())

if __name__ == "__main__":
    main()
```

Thanks again — your explanation really helped me improve the responsiveness of the system while handling long exposure frames reliably.

If you have any further suggestions after reviewing this code, I’d really appreciate it — I come from an Android development background and I’m still learning the camera pipeline and tuning on Raspberry Pi. Your feedback would be super helpful!

Akshar062, 2025-06-17:
i have wrote one test code

```
from picamera2 import Picamera2
import time

def method_1_wait_for_stabilization(picam2, exposure_time, gain):
    print("\n▶️ Method 1: Wait for stabilization")
    frame_duration = max(exposure_time + 1000, 33333)
    picam2.set_controls({
        "AeEnable": False,
        "ExposureTime": exposure_time,
        "AnalogueGain": gain,
        "FrameDurationLimits": (frame_duration, frame_duration)
    })

    start = time.monotonic()
    for _ in range(10):
        metadata = picam2.capture_metadata()
        print(f"⏱ {metadata['SensorTimestamp']} - Exposure: {metadata['ExposureTime']}, Gain: {metadata['AnalogueGain']}")
        if abs(metadata['ExposureTime'] - exposure_time) < exposure_time * 0.05:
            break
        time.sleep(0.2)
    elapsed = time.monotonic() - start
    print(f"✅ Stabilized in {elapsed:.2f} seconds")
    return elapsed

def method_2_restart_camera(picam2, exposure_time, gain):
    print("\n▶️ Method 2: Restart and apply settings")

    start = time.monotonic()
    picam2.stop()
    config = picam2.create_still_configuration(
        main={"size": (1920, 1080), "format": "RGB888"},
        controls={
            "AeEnable": 0,
            "AfMode": 0,
            "LensPosition": 7.0,
            "ExposureTime": exposure_time,
            "AnalogueGain": gain,
            "FrameDurationLimits": (exposure_time + 1000, exposure_time + 1000)
        }
    )
    picam2.configure(config)
    picam2.start()
    metadata = picam2.capture_metadata()
    elapsed = time.monotonic() - start
    print(f"⏱ {metadata['SensorTimestamp']} - Exposure: {metadata['ExposureTime']}, Gain: {metadata['AnalogueGain']}")
    print(f"✅ Restart and configured in {elapsed:.2f} seconds")
    return elapsed

def main():
    picam2 = Picamera2()
    picam2.configure(picam2.create_still_configuration(
        main={"size": (1920, 1080), "format": "RGB888"},
        controls={"AfMode": 0, "LensPosition": 7.0}
    ))
    picam2.start()
    time.sleep(1)

    exposure_time = 2_500_000  # 2.5 seconds
    gain = 32.0

    t1 = method_1_wait_for_stabilization(picam2, exposure_time, gain)
    t2 = method_2_restart_camera(picam2, exposure_time, gain)

    print(f"\n📊 Comparison:")
    print(f"Method 1 (wait): {t1:.2f} seconds")
    print(f"Method 2 (restart): {t2:.2f} seconds")

    picam2.stop()

if __name__ == "__main__":
    main()

```

```
~/Desktop/frame_Capture $ /usr/bin/python /home/mindrontm/Desktop/frame_Capture/debug.py
[4:06:21.280859058] [28812]  INFO Camera camera_manager.cpp:326 libcamera v0.5.0+84-79f583c3-dirty (2025-05-06T02:52:19BST)
[4:06:21.289473297] [28815]  INFO RPI pisp.cpp:720 libpisp version v1.2.1 981977ff21f3 29-04-2025 (14:13:50)
[4:06:21.301452420] [28815]  INFO RPI pisp.cpp:1180 Registered camera /base/axi/pcie@1000120000/rp1/i2c@88000/imx519@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0
[4:06:21.305541689] [28812]  INFO Camera camera.cpp:1205 configuring streams: (0) 1920x1080-RGB888 (1) 1920x1080-RGGB_PISP_COMP1
[4:06:21.306041834] [28815]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/imx519@1a - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected CFE format: 1920x1080-PC1R

▶️ Method 1: Wait for stabilization
⏱ 14782493205000 - Exposure: 119998, Gain: 8.0
[4:06:22.629707743] [28820]  WARN IPARPI ipa_base.cpp:1412 Ctrl AeEnable is not handled.
⏱ 14782733986000 - Exposure: 119998, Gain: 8.0
⏱ 14782974761000 - Exposure: 119998, Gain: 8.0
⏱ 14783456319000 - Exposure: 119998, Gain: 8.0
⏱ 14785956044000 - Exposure: 2499994, Gain: 16.0
✅ Stabilized in 3.47 seconds

▶️ Method 2: Restart and apply settings
[4:06:25.978708094] [28819]  INFO Camera camera.cpp:1205 configuring streams: (0) 1920x1080-RGB888 (1) 1920x1080-RGGB_PISP_COMP1
[4:06:25.983965799] [28815]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@1000120000/rp1/i2c@88000/imx519@1a - Selected sensor format: 1920x1080-SRGGB10_1X10 - Selected CFE format: 1920x1080-PC1R
[4:06:25.987502257] [28844]  WARN IPARPI ipa_base.cpp:1412 Ctrl AeEnable is not handled.
⏱ 14788669940000 - Exposure: 2499994, Gain: 16.0
✅ Restart and configured in 2.72 seconds

📊 Comparison:
Method 1 (wait): 3.47 seconds
Method 2 (restart): 2.72 seconds
```

davidplowman, 2025-06-17:
Hi, just a couple of suggestions on the code.

You're using `create_still_configuration`  which by default allocates only one buffer because it's worried about large still image captures taking up lots of memory. In your case I don't think this is a problem, so maybe try adding `buffer_count=3`. Adding more buffers reduces the number of frame drops. In fact, with just one buffer, you are guaranteed to be dropping every other frame.

When you stop and restart the camera, there should be no need to call `configure` again. You can just use `set_controls` to update the controls before the next `start()`. Avoiding `configure` should speed things up, and avoid lots of nasty memory management churn!

Other than that, looked good to me. Good luck!