raspberrypi/picamera2 #1287: [BUG] Camera settings seem to need many frames to take effect

zkelly1, 2025-07-11:
**Describe the bug**
It appears that changes in camera settings (exposure time and gain) seem to take on the order of 80 frames to take effect in the output signal captured from my camera. In the plot below, this would look like a *worse* fit at slow modulations, as the camera adjusting the AGC would flatten the observed sinusoid. 

[image]

**To Reproduce**
I am running a Sony IM219 in the 640x480 mode 200 FPS default mode, capped at 120 FPS via the frame limit feature in Picamera2. I've set the buffer count to 32 and the format to YUV420. Additionally, I am using a modified driver that does 4x binning instead of 2x binning on the camera to get a larger FOV in this mode. The core capture loop of my program is simple and looks something like this: 

```python
While(True):
   # Check for pause/stop , capture frame 
   ...
   
  # If 0.25 seconds have elapsed, calculate new settings, update settings 
  if(current_time - last_AGC_update > 0.25):
     new_gain, new_exposure = calculate_AGC_settings(np.mean(frame))

      world_cam.set_controls({'AnalogueGain': new_gain, 
                                                 'ExposureTime': new_exposure
                                                 }
                                              ) 

```

**Expected behaviour**
I expect AGC settings to take effect within a few frames, as found in previous discussions on the RPi forums. 

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

**Additional context**
Add any other context about the problem here.

davidplowman, 2025-07-14:
Hi, let me start by explaining how control updates (such as to exposure time and analogue gain) happen.

When you submit new control values, they end up getting submitted at the back of the request queue (which is the same size as the buffer count). This is libcamera behaviour, and although there's agreement (I think) that requests need splitting into separate queues of buffers and of controls, this has not yet happened.

So in your case, you're probably incurring a 32 frame delay right there. You might want to consider reducing that number and see how it affects things. Large buffer counts can help to avoid dropped frames, and at these high framerates that might be an issue, so maybe halve it to 16 and see what happens.

Even once the controls get to the front of the queue, they take "a few frames" to take effect, usually about 3 or 4. So that needs to be factored in to your expectations too.

The loop that you've reproduced above looks like it waits for "one round" of adjustment to complete before starting the second. Generally you can get nicer behaviour if you can pipeline the changes (i.e. don't wait for a round to complete before starting the next) - you would apply a smaller delta, but more often. Note that you can monitor the metadata on frames to see when your requested gain/exposure have been applied, rather than simply waiting a "safe" period of time.

Also note that when you do AGC in application code like this, you're normally operating on non-linear (post gamma transform) data (unless you've taken steps to avoid this). This may cause your adjustment calculations to undershoot, causing longer convergence times. Again, could be a good reason to try and pipeline the changes.

In your example code you seam to be doing some mean-level kind of metering. I just wanted to point out that if `frame` is in YUV420 format, then `np.mean(frame)` is going to skew the results towards 128 because of the UV values in there. Maybe use `frame[:height]` if that might be a problem and you only intend to include the Y samples.

Finally, I guess I need to ask why you aren't using the built-in AGC/AEC algorithm. That would almost certainly give you the most responsive behaviour, and bypass all the problems with that 32-buffer queue. The in-built algorithm is extremely flexible, can do simple mean-level target metering and much more, and is fully customisable in the camera tuning file. If you would like to pursue that, I'd be very happy to help you try and get the best results.

Hope some of that is useful!

zkelly1, 2025-07-14:
Thanks so much for your detailed reply/advice! This has greatly helped us build an understanding of how things fit together. Below I'll clarify a little bit our setup, goals, and our ideal state. 

Our system is part of a scientific apparatus for which we need control over the temporal (and spatial) response properties of signal from the IMX219. We are reading the RAW frames directly from the camera, though have initialized the video stream with YUV420 (something I forgot to mention, which means we simply take the mean of the RAW image for the AGC). We need to avoid any AEC behavior. We also need an AGC that changes exposure / gain slowly enough that we do not filter temporal variation in the camera input at and above 0.25 Hz. We also need to have our own additional dynamic digital gain procedure to increase brightness of the frames, both to measure response and also be able to visually see what is in the image, as we are operating at very high FPS ranges (more on that below). Further, we want to prioritize the adjustment of these settings (e.g. only raise exposure after gain, digital gain last, and vice versa). 

As aforementioned, we are running the camera between 120-180 FPS additionally with a 4x binning mode instead of a 2x binning mode (so really maxing out the camera here). We have recollection that when we started working on this, that faster updates using our AGC (as opposed to 0.25 seconds) hindered the FPS. We also have vague memory (would could be wrong, both in our memory and current understanding of the camera and Picamera2/libcamera, very open to your guidance here) that the built-in AGC somehow hindered the goal results (either in FPS or expected behavior, currently unsure). 

We would be very open to using the built-in AGC if you have any advice on configuring it to meet these goals?

closed 2025-07-17 by zkelly1