raspberrypi/picamera2 #1288: Picamera2 should not override LIBCAMERA_RPI_CONFIG_FILE if it exists and is valid

njhollinghurst, 2025-07-16:
When imported/initialized, Picamera2 changes the environment variable `LIBCAMERA_RPI_CONFIG_FILE`.

The user can override it, but only by setting it from Python, in between the import and the constructor. This has been alluded to in other issues but I'd like to call it out as an undesirable gotcha.

Better to check first if the environment variable already exists and points to a valid file?

The hard-coded [reference](https://github.com/raspberrypi/picamera2/blob/366a13ab865fba5b73a25dc7d50af3d1792a8509/picamera2/__init__.py#L25l)  to the user directory `~/libcamera/src/libcamera/pipeline/rpi/` is also questionable.

davidplowman, 2025-07-17:
Thanks, Nick!

Is the correct answer just to ignore that environment variable entirely?

njhollinghurst, 2025-07-17:
The requirement is that the user needs a way to disable the timeout (e.g. when using an externally triggered camera). The environment variable is currently the means to do that, and has been somewhat documented.

> Is the correct answer just to ignore that environment variable entirely?

I was planning to do the opposite, and respect the existing value if already set rather than clobbering it. That would align `picamera2` behaviour more closely with `rpicam-apps`.

Does that seem reasonable? If so I'll make a PR.

PR #1289 (2025-07-21): picamera2 init: Don't override LIBCAMERA_RPI_CONFIG_FILE if valid
If the environment variable already exists and points to a valid file, keep it. This allows the config file to be user-specified from outside the Python script.

To fix #1288